translationalneuromodeling/tapas #3: Question: HGF - how to set parameters within region that model assumptions aren't violated…?

meganEJcampbell, 2017-12-10:
Hi there TAPAS team, 

Please look kindly on my newbie status! 

I am completely new to implementing the HGF, with a psychology/cogneuro background and little mathematical know-how I'm finding a few points in confusing. Primarily, how to set the parameters to simulate responses given inputs.  It's my first go at running a simulation, so going with the demo:
```
sim = tapas_simModel(u,...
                     'tapas_hgf_binary',...
                     [f -2.5 -6],...
                     'tapas_unitsq_sgm',...
                     5);
```
but just swapping in the inputs from my task. This throws an error: **Error using tapas_hgf_binary (line 225) Variational approximation invalid. Parameters are in a region where model assumptions are
violated.**

The tips in the comments of tapas/HGF/tapas_hgf_ar1_binary_config.m (quoted below) hint at the procedure but I'm afraid I have no idea what 'wide and loose' actually means with respect to the values for these parameters, and every number I try returns the 'model assumptions are violated' error... 

Cheers, 
Megan
(PhD Student)

```
% - When analyzing a new dataset, take your inputs u and use
%
%   >> est = tapas_fitModel([], u, 'tapas_hgf_binary_config', 'tapas_bayes_optimal_binary_config');
%
%   to determine the Bayes optimal perceptual parameters (given your current priors as defined in
%   this file here, so choose them wide and loose to let the inputs influence the result). You can
%   then use the optimal parameters as your new prior means for the perceptual parameters.
%
% - If you get an error saying that the prior means are in a region where model assumptions are
%   violated, lower the prior means of the omegas, starting with the highest level and proceeding
%   downwards.
```

from the hgf_demo.m
```
% To do that, we simply choose values for $\omega$. Here, we take $\omega_2=-2.5$ and 
% $\omega_3=-6$. But in addition to the perceptual model _hgf_binary_, we now 
% need a response model. Here, we take the unit square sigmoid model, _unitsq_sgm_, 
% with parameter $\zeta=5$.
%%
```

BPaton, 2017-12-10:
Hi Megan,

If you lookin the model config files, like 'tapas_hgf_binary_config' you
can see the full set of parameters or that model. Say you wanted to change
one of the kappa parameters you can set the mean under something like:

omega_x = [-3, - 6]

The first number is the mean for that value for the first level it occurs
in, some parameters are present at every level, some determine coupling
between levels and some are undefined like the binary_HGF level 1 x
parameter. Anyway, this value determines the mean of the Gaussian which
describes the possible parameter values. To set the other vital Gaussian
parameter, the variance (the inverse of which is the precision) you will
see right after the parameters describing the mean the ones describing the
variance, something like:

omega_sa = [0,0]

The variance as you know describes the dispersion of the distribution and
so a larger number here means a "wide" prior. This means many more
different values of the parameter are plausible. A variance parameter that
is set to zero has effectively infinite precision, in other words it is an
infinitely tall stick (delta) centred on the value of the parameter for the
mean. These 0 variance priors are in effect fixed and will not change and
the model will not estimate them but take them as given. Sometimes this is
helpful.

But why not just make every parameter have a prior variance which is
non-zero (it has to be zero or more positive)? Sometimes there is
insufficient data to correctly estimate the parameter, sometimes the
parameter is already well characterised. The other major reason is that if
the model has to estimate more parameters the model has more free
parameters and is thus more complex. In model fitting we always want to
trade off model complexity versus model accuracy. The AIC and BIC
parameters that the fitting outputs are model evidence estimates more
commonly used that have a penalty term in them related to the number of
parameters and/or data points. The LME or log model evidence  (Free Energy)
that the model fitting outputs is a 'better' estimate of model evidence
that generally performs better and gives you a closer approximation to the
true model evidence.

One other thing, the way simModel works and any other similar function is
that the list of parameters in the square brackets is the same as the list
of parameters in the  model config file (in order). Hope that is clear.

So when using the Bayes optimal model to give you a good idea of a sensible
parameter space,  setting the prior variances to larger values will help
the model fitting find the right part of parameter space. However, some
parts of parameter space are bad, in that certain model fitting assumptions
might be violated or you might never converge. The HGF does a pretty good
job of warning you when you might be disappearing over a nasty cliff or
similar in parameter space as it has done for you here. Find a good set of
priors can take a little work and looking at how your parameters evolve
over fitting iterations will help to see what values best work. If you open
the tapas quasi-newton config file there is a parameter for model fitting
outputs to be verbose (or not). Setting it to true (if I remember right)
will show you the model parameters on the command line over iterations.

So hope this helps and sorry I cannot be more specific as I am on my phone.
No doubt when Chris awakens on the other side of the world he will have
some excellent advice and let me gently know all the bits I rememberwd
incorrectly. :)

Good luck.

Thanks.

Bryan.

<https://github.com/translationalneuromodeling/tapas/issues/3>, or mute the
thread

meganEJcampbell, 2017-12-10:
@BPaton you're a legend! 
Thank you so much for your detailed and so very quick reply, on a Sunday no less! I didn't expect anyone to get back to me until sometime tomorrow (out time)...
Ok, I don't feel so bewildered now. I'll crack on and see how I go with setting verbose to true in tapas/HGF/tapas_quasinewton_optim_config.m, and see where a few iterations gets me! 😄

meganEJcampbell, 2017-12-11:
@BPaton firstly, apologies for spamming this forum with my newbie issues... the pressure is on to get this working so I can finish writing up (finally!). I'm eternally grateful to you for any help! 

so I got a step further configuring bayes optimal binary with:
```
[c.ommu = [NaN,  -10,  -10];    
c.omsa = [NaN, 0, 0]; 

(tapas_hgf_binary_congif) and get:
Regularizations exhausted - resetting algorithm. 
Initial argument: 
Initial value: 278.4352 
```
but then get an error for the next bit when it's trying to calculate (LME)...
```
Error using tapas_bayes_optimal_binary
Too many output arguments.

Error in tapas_fitModel (line 202)
[dummy, r.optim.yhat, r.optim.res] = r.c_obs.obs_fun(r, infStates, r.p_obs.ptrans);

```

chmathys, 2017-12-12:
Hi Megan,

Could you please post your whole config file and the exact command you're using? Also, it looks like you're fixing the omegas to a value of -10 and not estimating them. Is that intended?

Chris

meganEJcampbell, 2017-12-13:
@chmathys Hi Chris, Thank you for responding to my query! I'm trying to estimate the parameters - I thought I was following the advice in the Tips in tapas_hgf_binary_config.m (quoted below) but I clearly have little to no idea what I'm doing... Any help you can offer is greatly appreciated.
Cheers, 
Megan

%When analyzing a new dataset, take your inputs u and use
%
%   >> est = tapas_fitModel([], u, 'tapas_hgf_binary_config', 'tapas_bayes_optimal_binary_config');
%
%   to determine the Bayes optimal perceptual parameters (given your current priors as defined in
%   this file here, so choose them wide and loose to let the inputs influence the result). You can
%   then use the optimal parameters as your new prior means for the perceptual parameters.
%
% - If you get an error saying that the prior means are in a region where model assumptions are
%   violated, lower the prior means of the omegas, starting with the highest level and proceeding
%   downwards.

My config file is:
[tapas_hgf_binary_config.m.zip](https://github.com/translationalneuromodeling/tapas/files/1553860/tapas_hgf_binary_config.m.zip)

and I intended to then simply adapt the commands from the demo tutorial, as below. 

```
  nn=1  % add for loop per subject later
    filename = ['/Users/uqmcamp6/Documents/MATLAB/Exp3_PredIm_MC/HGF_firstgo/', char(data.ID(nn)), '_1matchV_0mismatch'];
    u = data.inputs(:,1,nn); % col 1: inputs for match/mismatch trials (1=match)
    
    scrsz = get(0,'ScreenSize');
    outerpos = [0.2*scrsz(3),0.7*scrsz(4),0.8*scrsz(3),0.3*scrsz(4)];
    figure('OuterPosition', outerpos)
    plot(u, '.', 'Color', [0 0.8 1], 'MarkerSize', 11) %%% notes:  blue [0 0 1]; magenta [1 0 1] ; cyan [0 1 1]; purple[0.8 0 0.5]
    xlabel('Trial number')
    ylabel('Inputs "u" ') %%%
    axis([1, 400, -0.1, 1.1])
    set(gca,'Xtick',0:40:400) %%% MEJC: changed tick-marks from 50's to 40's = blocks (note 'gca' = get current axis handle)
    set(gca,'XtickLabel',0:40:400)
    title([char(data.ID(nn)),  ' match 1 vs mismatch 0'])
    
% % % % cd /Users/uqmcamp6/Documents/MATLAB/tapas-2.7.3.0/HGF
% % % %     sim = tapas_simModel(u, 'tapas_hgf_binary', [NaN 0 1 NaN 1 1 NaN 0 0 NaN 1 NaN -2.5 -6], 'tapas_unitsq_sgm', 5);
% % % %     tapas_hgf_binary_plotTraj(sim)
% % % % 
% % % %     savefig('filename')
    
end % subject loop    
and command as per demo:
bopars = tapas_fitModel([],...
                         u,...
                         'tapas_hgf_binary_config',...      % where you can define values for omegas...               
                         'tapas_bayes_optimal_binary_config',...  % 'tapas_bayes_optimal_binary_config' for binary inputs = observation model. 
                         'tapas_quasinewton_optim_config'); % changed to have verbose = true to get more details
```

My inputs are binary, for a task with 400 trials with 10 hidden blocks of changing likelihood for a stimulus-response match ( 1) or mismatch(1) - as per an 'automatic imitation' task but manipulating uncertainty/ expectations of SR congruence.
 Running on MATLAB 2016b.

chmathys, 2017-12-14:
Thanks for posting your config file. When you set the prior variance (ie what’s in the vectors ending in *sa) to 0, then the corresponding parameter’s value is fixed to its prior mean (ie what’s in the vectors ending in *mu). The problem with your config file is that you’ve fixed ALL of your parameters, so estimation fails necessarily because there’s nothing to estimate.

As a start, try:

c.ommu = [NaN,  -10,  -10];    
c.omsa = [NaN, 1, 1]; 

If that throws up an error, try to find entries for c.ommu that work. If you can’t find any, just write again.

As a general remark, this process of finding priors that work is a normal part of model development. It’s the same kind of thing as when a histologist adjusts his or her microscope until it has the right focus. What matters is the patterns you see once your model is correctly focussed.

meganEJcampbell, 2017-12-14:
Thank you very much for this explanation Chris - I think I get it now! :-)

meganEJcampbell, 2018-02-02:
Hi Chris, 
Thanks again for responding to my query. I thought I had it, but actually, I don't.

Could you please expand a little on how I should find entries for c.ommu that work? What range of values for mu would be reasonable? I've entered values [nan, -10, -10] as suggested and had no luck, and then tried a wide range of values between -1 and -100 ... still getting the same error each time.

Cheers, 
Megan

chmathys, 2018-02-03:
Hi Megan,

Could you send me your input sequence and perhaps two or three examples of
response sequences you have (chmathys AT gmail DOT com)?

Best,
Chris

alex-lebedev, 2018-02-15:
Hi Dear Both,
I think I'm having a related issue.
I'm currently going through HGF Toolbox Demo for "example_binary_input.txt" (TAPAS Version 2.7.4.1), and when attempting to run 

`sim = tapas_simModel(u,...
        'tapas_hgf_binary',...
        [NaN 0 1 NaN 1 1 NaN 0 0 NaN 1 NaN -2.5 -6],...
        'tapas_unitsq_sgm',...
        5);`

I get the following error:
> Error using tapas_hgf_binary (line 215)
> Variational approximation invalid. Parameters are in a region where model assumptions are
> violated.
> 
> Error in tapas_simModel (line 164)
> [r.traj, infStates] = prc_fun(r, r.p_prc.p);

Tried different omega values, but it didn't seem to solve the problem. Is there anything I'm missing?

chmathys, 2018-02-15:
For the record, my response to Megan after she sent me input sequences
and responses:

Hi Megan,

Thanks for sending this. I’ve had a quick look at it, and the inputs
look fine. The problem seems to be that you weren’t using an
appropriate model for your responses.

Since you have logRT as your response measure, logrt_linear_binary
would be an appropriate response model. The estimation works with
priors out of the box, with the exception of setting the omega priors
in tapas_hgf_binary_config.m to

c.ommu = [NaN,  -4,  -2];
c.omsa = [NaN,   1,   1];

To understand how this response model works, you can look at the
documentation in the code of tapas_logrt_linear_binary.m. This
response model is similar to the ones used in Marshall, Mathys et al.
PLOS Biol, 2016 and Lawson et al, Nature Neurosci, 2017.

I hope this helps.

Best wishes,
Chris

chmathys, 2018-02-15:
Hi Alex,

The command as posted will not work because the ellipses (three full
stops) are only needed when breaking a line in Matlab code. Try
removing all ellipses from the command and running it again.

Also, make sure you are using a freshly unpacked version of TAPAS and
an unchanged version of the Live Script.

Chris

alex-lebedev, 2018-02-15:
Hi Chris,
I don't the problem is in line breaks, but thanks for your response.
Now, after going through hgf_demo.m I see that it uses different perceptual model parameter inputs (than the ones specified in hgf_demo.pdf). Putting the ones from the livescript works fine.
Best,
Alex