translationalneuromodeling/tapas #6: Using PhysIO with HCP data

mrikasper, 2018-01-27:
From a user:

> Hello,
> 
> I'm interested in using PhysIO for physiological data from the human connectome project. The physiological data is in HCP available as a single text file, containing 3 columns (trigger, physio, cardio (not sure about the order right now)). Does PhysIO support reading such text files, or is it necessary to read the raw physiological data?
> 
> - Anders

mrikasper, 2018-01-27:
Dear Anders,

following your input, we have decided to support the converted 3-column HCP single-file physiological data (`*_Physio_log.txt`) natively in PhysIO, starting with PhysIO R2017.3. Just set `logfiles.vendor = 'Siemens_HCP'`.

Note that this data is not in the format specified in the [HCP Wiki](https://wiki.humanconnectome.org/display/PublicData/Understanding+Timing+Information+in+HCP+Physiological+Monitoring+Files) - which rather describes the native Siemens VB logfile format (`*.puls, *.resp, *.ecg`).

If you use PhysIO and have the `*_Physio_log.txt` files at hand, choosing `'Siemens_HCP'` seems to deliver reasonable results:

Here we see an example of RETROICOR modeling for subject **178748**, session `tfMRI_MOTOR_LR`. The contrast is an F-contrast for a model including just the RETROICOR regressors from PhysIO (3rd order cardiac, 4th order respiratory, 1st order interaction regressors).

[image]

I attached the 3 batches to create these results( you have to rename them from `.txt` to `.m`
 to use them in the SPM batch editor):
- Preprocessing included realignment and smoothing with a 2mm kernel in SPM: [batch_preproc.txt](https://github.com/translationalneuromodeling/tapas/files/1670352/batch_preproc.txt)
- PhysIO was used to create the RETROICOR regressors: [batch_physio.txt](https://github.com/translationalneuromodeling/tapas/files/1670353/batch_physio.txt)
- GLM was computed with SPM using those regressors: [batch_glm_nomove.txt](https://github.com/translationalneuromodeling/tapas/files/1670355/batch_glm_nomove.txt)

Finally, to create the contrasts and overlays conveniently, I used another PhysIO function:

`tapas_physio_report_contrasts('fileSpm', 'GLMNoMove/SPM.mat', 'filePhysIO', 'OutPhysio/physio.mat', 'fileStructural', 'meantfMRI_MOTOR_LR.nii')`

I hope this step-by-step example will prove useful for you and other HCP data analysts.

All the best,
Lars

wangyulinatUGent, 2018-01-28:
Thanks for this manual!  When i applied the PhysIO to my data analysis, in the last step, i only used tapas_physio_report_contrasts without the contents in the brackets, will this matter a lot? 
Anyhow i can still get the results for the RETROICOR regressors

mrikasper, 2018-01-28:
Dear Yulin,

this depends on what the working directory was when you ran `tapas_physio_report_contrasts`. Per default, the function takes both `SPM.mat` and `physio.mat` from the current working directory. This means, if you ran the GLM, you usually end up in the GLM folder with the `SPM.mat` anyway, and if you didn't specify another directory for PhysIO, the toolbox will write its output structure `physio.mat` in there as well. So all good in your case. However, if you run batches or return to a subject later, it might be helpful to specify these paths explicitly to avoid unexpected behavior.

For the underlay, on the other hand, the default is just the `avg152T1.nii` template from SPM's canonical folder, which might not match the geometry of your subjects. I would always go for the mean functional image of the corresponding subject. But just for a quick check, your procedure is working as well.

Note: If you type `help tapas_physio_report_contrasts` in the command window, you will see that the function is very versatile, and you can specify more parameters, e.g., the reporting threshold or which contrasts to select (e.g., only cardiac) for the ouput PostScript file.

All the best,
Lars

wangyulinatUGent, 2018-01-29:
Dear Lars,

Thanks a lot for your detailed explanations! Now i am much more clear!

Have a nice day!
Yulin

annikalinke, 2018-02-09:
Dear Lars, 

how does RETROICOR as implemented in PhysIO deal with EPI sequences that acquire multiple slices at the same time (such as the HCP protocols)? I see that in the HCP batch_physio script you provided above you specify 91 slices but these were acquired 8 slices at a time. Is there a way to specify slice the order for multiband acquisitions, or - given the short TR - is it fine to treat them as having been acquired separately?

Thank you!

mrikasper, 2018-02-12:
Dear Annika,

well spotted, and sorry for any confusion!

At the moment, the assumptions on resampling to the specified onset slice in PhysIO are pretty simple: the temporal spacing between slices is assumed to be `TR/nSlices`, and the sampling time of the slice with index `iSlice` within the volume is assumed to be `(iSlice-1)/nSlices*TR`. This would refer to classical ascending slice order, acquired one-slice-at-at-time.

We will provide a proper multiband (and non-ascending slice order) support for `scan_timing.sqpar.onset_slice` with the next major PhysIO release R2018.1, probably in June.

Until then, I feel that for the short TRs of the HCP Multiband sequences (<1s), the actual time of the individual slice will not matter too much for physiological noise correction.  RETROICOR in particular takes care of constant physiological phase offsets between slices, which will be fitted by the GLM, if the sine and cosine of the phase are included (and SNR is high enough...), since

`cos(phi +dphi) = cos(dphi)*cos(phi) - sin(dphi)*sin(phi) = beta1*cos(phi) + beta2*sin(phi)`

...with beta being the parameter estimates in the GLM for these columns. So if you put sine and cosine of the phase in your GLM (which PhysIO does), you should be fine.

We also did not feel the issue to be top priority, since SPM does not provide slice-wise GLMs, so you have to decide on a reference slice anyway. Having said that, if you want confound regressor matrices resampled for individual slices, just specify an array for the onset_slice, e.g., `scan_timing.sqpar.onset_slice = [1 5 9 23]` (which again is assuming classical ascending slice order as explained above) to create all `multiple_regessors_slice*.txt` files at once for these slices.

I hope this helps and we will keep you posted on the next release!

All the best,
Lars

closed 2018-11-01 by mrikasper