translationalneuromodeling/tapas #8: HGF in one-condition tasks with a task-learning component

Leone-G, 2018-02-06:
Dear Dr Mathys, 

I would kindly ask you for an elucidation regarding the HGF toolbox. 

Our data consist in a series of 0 and 1 (y) which correspond to 
self-reported awareness judgment. There are no variations in condition 
but there is a learning component to the task (participants should 
report less and less « yes » response through the course of the task). 

We are interested in modeling the participants' internal beliefs about 
their response to the upcoming trial (a priori expectation that he/she 
will response yes or no) given the previous trial history. 

We are just a bit confused about how this should be done with the HGF 
toolbox. 

We are using the following command line: 

est = tapas_fitModel(y, y, 'tapas_hgf_binary_config', 
'tapas_beta_obs_config'); 

Could you please confirm that using y as both response and input is the 
correct implementation of our intention? Otherwise, is there another 
specific approach to apply HGF modeling to one-condition tasks with a 
task-learning component ? 

Finally, we are currently only using the HGF perceptual model and 
comparing a Beta to a Softmax observational model. Would you have any 
suggestion concerning other types of perceptual or observational models 
we could use with this kind of data? 

Thanks a lot for for your attention, 
Best regards, 

Giovanni

chmathys, 2018-02-07:
Dear Giovanni,

If your assumption is that subjects use their own decisions as learning
signals, then your implementation is correct. I assume you’re writing
because of a slight discomfort at how circular the logic looks here.
However, as long as you’re using decision models based on predictions
(muhat, etc.) as opposed to posteriors (mu, etc.), there is no circularity
because every decision will only be based on the observation of previous
decisions.

In addition to the softmax and beta_obs decision models, you could also try
unitsq_sgm (the unit square sigmoid). This is logically somewhat more
consistent than softmax based on mu1hat because it enforces zero
probability of making a choice corresponding to outcomes believed
impossible, but because of this enforced logical consistency, it often has
lower evidence than softmax.

I hope this helps.

Christoph

<https://github.com/translationalneuromodeling/tapas/issues/8>, or mute the
thread

Leone-G, 2018-02-08:
Dear Dr Mathys,

Thanks a lot for your extremely clear answer and suggestions.
Actually, I was a bit worried about possible circularity in the model.
It is much more clear to me now. 

Best regards,

Giovanni

closed 2018-02-08 by tnutapas