translationalneuromodeling/tapas #9: MATLAB Code for rDCM

sbajaj1, 2018-02-11:
Hi,

I am interested in using rDCM to calculate whole brain effective connectivity using DCM for controls vs patients.

I was wondering if I can get access to MATLAB code and basic instructions on this topic. I am familiar with basic DCM steps but not sure how to perform whole brain DCM.

I would really appreciate any directions on this topic.

Thanks a lot !
Sahil

StefanFraessle, 2018-02-11:
Dear Sahil,

Thank you very much for your interest in using rDCM for your project. We will soon make the MATLAB code available as part of the TAPAS toolbox. The reason why we haven't published rDCM yet is because the code needs a bit of polishing to turn it into an easy-to-use toolbox for people interested in performing whole-brain effective connectivity analyses (including proper documentation and tutorials on how to use the method).

My apologies for this delay - but we expect to have rDCM published in TAPAS within the next 2-3 months. I hope this is helpful.

With very best wishes,
Stefan

sbajaj1, 2018-02-11:
Dear Dr. Fraessle,

First of all, thank you so much for your quick response.

I am really looking forward to use rDCM on our data.

Would you suggest me to to finish SPM preprocessing steps by the time rDCM codes are available?
Or could you tell me **up-to which step** I can perform the basic analysis so that I can make my data ready to be used for rDCM when code is available?

Thanks a lo once again,
Sahil

StefanFraessle, 2018-02-12:
Dear Sahil,

You could prepare classical SPM analysis up to and including the specification of the first-level GLMs. At that point, classical DCM and rDCM analyses will differ in that for rDCM you would like to use a whole-brain parcellation atlas to extract VoI time series from the entire cortex, whereas in classical DCM, you would select some regions of interest based on the observed BOLD activation patterns.

I hope this helps.

With very best wishes,
Stefan

sbajaj1, 2018-02-12:
Dear Dr. Fraessle,

Great !
Thanks a lot. I will finish my classical SPM analysis soon and will use rDCM once its available.

Thank you so much once again,
Sahil

closed 2018-02-13 by tnutapas