translationalneuromodeling/tapas #10: PhysIO Data Processing

DanielHoffmannAyala, 2018-02-13:
Hello,

we are using PhysIO with a respiratory belt and PPU data from a Biopac system.
Our data was not processed in any way before going into PhysIO, so low frequency trends and scanner-related high frequency artifacts are still present. _(see image attached)_

[image]

The following two issues emerged:

1) Upon inspection of the graphical outputs, we noticed that the 1st & 2nd pass peak detection steps failed to recognize every peak where the _entire_ signal was below the zero line. Contrary to **respiratory** data, which is indeed bandpass filtered, **cardiac** data is **not** filtered in PhysIO before going into processing.

2) On a sidenote, we were not quite sure whether the high frequency (scanner-related) noise would meddle with the _exact_ positioning of the peak?

**Our pragmatic remedy** to both issues was High-Pass filtering the cardiac signal with a **2nd order butterworth** filter (**0.5** lower, **5** Hz upper cutoff, Matlab) before introducing it into PhysIO which seemed to fully resolve the issues with peak detection.

We wonder whether our simple bandpass filtering is a valid approach.
_If it is_, would the exact parameters just be determined empirically by trying out settings and evaluating their effects on the signal or are there more specific guidelines?

(On another sidenote, we wonder whether the bandpass filter could distort the cardiac signal's phase information, which filters are known to do in the vicinity of their cutoff frequencies)

_If it's not_, what approach would you recommend?

Thank you very much for your effort!

Regards,
Daniel

mrikasper, 2018-02-14:
Dear Daniel,

thank you very much for your detailed description of the problem. This is very helpful.

You are right, the initial two passes of the peak detection rely on a very simple threshold approach (Matlab's `findpeaks`), which will not work if data is completely below the assumed threshold. One very simple approach around that is to change this threshold. It is not a GUI parameter, but you can call the cardiac-pulse wave-template-generating function with the optional parameter `thresh_min`

```
    tapas_physio_get_cardiac_pulse_template(t, c, verbose, ...
    'thresh_min', newNegativeTreshold)
```

or directly change the default threshold in the function `defaults.thresh_min` (which is 0.4 at the moment).

I am not sure, though, that this will resolve the issue, since the global drift will induce varying thresholds over the time course of your recording.

So, in my opinion, some kind of filtering is indeed required for your data. I am not sure about the bandpass options you chose, though. I would say their filtering parameters should follow from what you know about the system you are studying, i.e., the human heart in a gradient-switching MRI enviroment, i.e.:

- 0.5 Hz as lower boundary seems too tight, since there are sporty peoply with heart rates of about 30-40 bpm (~0.5Hz). I would go for something more like 0.1-0.2 Hz.
- 5 Hz upper cutoff will be OK for peak detection, but could still impact on the phase. For typical fMRI sequences, the slowest gradient switching of the slice excitation gradient will occur at about 20 Hz (assuming one slice per 50 ms), so 10 Hz should still be a very conservative upper bound.
- What is your transition band for the filter? That will impact on the phase information a lot. You could also use the filtering only for the initial peak detection, but apart from the template generation, use the unfiltered time series.

It might be that a simple linear regression (*detrending*) might be sufficient for peak detection and not impact on the phase issue so much. One way to use it in Matlab is

`detrendedC = detrend(c)`

This would leave you with the high frequency gradient noise, but its peak shifts of max 50 ms might be negligible. To understand that better: What is the time axis of your plot?

Finally, I would be very interested what you come up with and integrate it in the toolbox later on. So, if you are willing to share your code, or at least some anonymized test data that reproduces the problem, we could benchmark our implementation against that and provide a general solution for the community.

Thank you very much in advance and all the best,
Lars

mrikasper, 2019-09-01:
Dear Daniel,

I know it has been a long time, but we finally included bandpass filtering of the cardiac recordings into the PhysIO Toolbox, R2019b (v7.2.0). You can test it with the latest [TAPAS release v3.2.0](https://github.com/translationalneuromodeling/tapas/releases/tag/v3.2.0).

All options are in 
`physio.preproc.cardiac.filter`, described in a section of `tapas_physio_new`, and accessible via the SPM Batch Editor as well. 

We support Chebychev-Type-II filters as the recommended default, but Butterworth filters are also implemented.

All the best,
Lars

closed 2020-07-10 by mrikasper