translationalneuromodeling/tapas #13: Scan Timing from gradient in Philip scanner

mrikasper, 2018-03-18:
From a user:

> Dear Dr. Kasper, 
> 
> I am using  Tapas for my analysis and I would like to get the scanner timing information. I had total 355 volumes, (34 slices ), although the spacing in volumes seems correct I am only able to get 330 volumes. 
> 
> I have attached the Matlab figure and the screenshot of the figure, It would be great if you can suggest if the thresholding and spacing are required to change or should I proceed with the nominal setting? 
> 

[image]

mrikasper, 2018-03-18:
Dear Tara,

some observations from your figure:

- The detection of volumes seems reasonable, TRs of about 2.5s
    - Therefore, from this plot, I would conclude that you have indeed only 330 volumes, or that the logfile ended prematurely. 
    - Could you please check in your accompanying image file (dicom/par/rec/nifti) that you indeed have 355 volumes acquired?
- The spacing between the volumes, however, seems rather large (800 ms, the red peaks in the lowest subplot) 
    - Did you use slice spacing *minimal*?
    - Was there a reason to do so? Typically, *equidistant* creates more homogeneous auditory noise by the gradients; or you could shorten your TR significantly.
- The gradient time courses itself within each volume (the green lines in the 2nd subplot) look atypical for a 2D EPI, and consequently, the slice detection does not work correctly (usally only about twenty-something peaks per volume)
    - Is that a different sequence, e.g., a 3D EPI, a spiral, or a multiband-sequence?
    - If not, you could try using a different gradient, e.g., 
    `grad_direction = 'z'` with thresholds `sync.zero = 1000;` and `sync.slice = 1200`
    - If yes, and this is, for example, a spiral, then you have the rare case that the volume detection works fine, but the slice detection does not. There is no inbuilt functionality in PhysIO yet for this case, but, as you suggest, you could use the nominal timing information.
        - Be careful, though, because of the non-equidistant spacing for your volumes. It is very important to set `sqpar.time_slice_to_slice = TA/nSlices`, where `TA` (the time of acquisition) equals the length of this green gradient block, which is about 1.7 s long, as estimated from the plot.

I hope this helps and look forward to you response.

All the best,
Lars

tarachandtaak, 2018-03-27:
Hi, 

Thanks for your valuable comments, I checked the dicom files and found that I was taking a wrong file, in which we have multiband sequencing but when I choose the right one, I got 355 volumes. 

Thanks, 

Tara Chand

closed 2018-10-08 by mrikasper