translationalneuromodeling/tapas #14: PhysIO toolbox error (gausswin/find_peaks)

mrikasper, 2018-03-18:
From a User:

> Dear Dr. Kasper,
>  
> I’m using the PhysIO toolbox to obtain regressors files that i use during my spm analysis. When i run the batch (that worked previously with other files), i obtain two errors and i don’t understand why they appear now given that the batch has worked very well with other similar data.
>  
> Here are the errors :
> Failed  'TAPAS PhysIO Toolbox'
> Undefined function 'gausswin' for input arguments of type 'double'.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_findpeaks_template_correlation.m" (v787), function "tapas_physio_findpeaks_template_correlation" at line 180.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_get_cardiac_pulses_auto_matched.m" (???), function "tapas_physio_get_cardiac_pulses_auto_matched" at line 72.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_get_cardiac_pulses.m" (???), function "tapas_physio_get_cardiac_pulses" at line 88.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 159.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1463.
>  
>  
> Failed  'TAPAS PhysIO Toolbox'
> Undefined function 'generatemsgid' for input arguments of type 'char'.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_findpeaks.m" (???), function "getPeaksAboveMinPeakHeight" at line 127.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_findpeaks.m" (???), function "tapas_physio_findpeaks" at line 54.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_rescale_gradient_gain_fluctuations.m" (???), function "tapas_physio_rescale_gradient_gain_fluctuations" at line 66.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_create_scan_timing_from_gradients_philips.m" (???), function "tapas_physio_create_scan_timing_from_gradients_philips" at line 177.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_create_scan_timing.m" (???), function "tapas_physio_create_scan_timing" at line 85.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 139.
> In file "C:\Users\nabila brihmat\Documents\programs\spm12\toolbox\PhysIO\tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1463.
>  
>  
> would you have any idea of the cause of these errors ?
>  
> Thanks in advance for your response,
> Best regards,
>  
> Nabila

mrikasper, 2018-03-18:
Dear Nabila,

- Which version of the toolbox are you using? 
    - In the current [release](https://github.com/translationalneuromodeling/tapas/releases/download/v2.7.4.1/tapas.zip), `gausswin` is replaced by `tapas_physio_gausswin` to avoid the use of the signal processing toolbox for this part of PhysIO.
- Consequently, your 1st error could relate to a recent change in your matlab environment, for example, you changed the Matlab installation, (de-)installed some toolboxes or removed them from the path
    - The 2nd error also points at that. `generatemsgid` is an auxiliary function used by Matlab in a lot of toolboxes, and the error indicates it is not on your path.
    - Could you type `which generatemsgid` in the command window of Matlab to see whether it finds it?
        - For me, the output is `/Applications/MATLAB_R2016b.app/toolbox/shared/spcuilib/generatemsgid.m`
    - If the response is `generatemsgid not found`,   
try to type `restoredefaultpath`, add the SPM path again, and rerun the PhysIO batch via the SPM batch editor.

I hope this helps.

All the best,
Lars

Nabila1812, 2018-04-06:
Dear Lars, 

In fact, the solutions you propose work. 

Thanks a lot for your help, 
All the best, 

Nabila

closed 2018-04-27 by mrikasper