translationalneuromodeling/tapas #15: Question: HGF Demo binary input semantics

psharp1289, 2018-04-01:
Hi all,

I'm a novice in computational approaches to cognition and am going through the HGF toolbox demo. I had a bit of confusion in regards to what the binary input mapped onto in the Iglesias et al piece. My guess is that they don't have any intrinsic relation to the stimuli, but rather a "1" is either a face or house depending on which cue (high/low tone) is presented on that trial of the task. This is the only way for me to make sense of the transition of contingency rates noted in Dr. Mathys' video tutorial on the HGF demo (unless I'm assuming something incorrectly about the frequency of outcomes -- e.g., if "1s" are faces, is seems the first 50 or so trials seems are overwhelmingly faces). 

Thanks in advance for your time & assistance,
Paul

chmathys, 2018-04-03:
Hi Paul,

You’re right, the outcomes are coded in what you might call ‘contingency
space’: a high tone followed by a face as well as a low tone  followed by a
house are 1, while a high tone followed by a house as well as a low tone
followed by a face are 0.

Christoph

<https://github.com/translationalneuromodeling/tapas/issues/15>, or mute
the thread

psharp1289, 2018-04-03:
Great -- thanks so much, Christoph. 

I'm also wondering how precisions on likelihoods (at the outcome level) are arrived at. Is this another parameter that's optimized to minimize surprise? (I also assume, too, that other precision values in the model are initialized with either Bayes-optimal priors or at 0?)

chmathys, 2018-04-05:
In this example, the precision at the outcome level is infinite (ie, no
perceptual uncertainty). This means that u = x_1 = mu_1 for all trials.
When this isn’t the case, you can use the hgf_binary_pu model in the
toolbox (for constant perceptual uncertainty) or hgf_binary_pu_tbt (for
perceptual uncertainty that varies trial-by-trial). The equation for mu_1
in these cases is Eq. 47 in Mathys et al. (2011), Front. Hum. Neurosci. 5,
39.

The initial precision values are specified (as variances) in the
configuration file tapas_hgf_binary_config.m. Priors on the initial mean
and variance of the belief trajectories are specified in c.mu_0{mu,sa} and
c.logsa_0{mu,sa}, respectively.

closed 2021-09-09 by chmathys