translationalneuromodeling/tapas #16: PhysIO installation into SPM toolbox

RyanJM41, 2018-04-19:
Hello,

I am having trouble installing the PhysIO component of the TAPAS package into the SPM toolbox. 

I copied the "physio/code" folder to the "spm12/toolbox/physio" folder, resulting in the following directory: "spm12/toolbox/physio/code". However, the SPM toolbox menu did not seem to recognize the physio toolbox. So, I then copied all contents of the "code" folder into the "toolbox/physio" folder, with the resulting directory: "spm12/toolbox/physio". The physio toolbox was then recognized by the SPM toolbox menu.

However, when I clicked on "physio" in the SPM toolbox menu, I received the following error message.

----
"Not enough input arguments.

Error in tapas_physio_job2physio (line 32)
physio = tapas_physio_update_from_job(physio, job, ...

Error in spm (line 1009)
    evalin('base',xTB(i).prog);
 
Error while evaluating UIControl Callback"
----

Any ideas as to how I could resolve this?

Thank you very much in advance for your time!

Kind regards,
Ryan

mrikasper, 2018-04-19:
Dear Ryan,

thank you for trying out the toolbox!

The second procedure you describe for making PhysIO visible as SPM toolbox is correct: 
- only the `code`-folder has to be a direct subfolder of `spm12/toolbox` and you can give it any name. - More precisely, SPM needs to find the `tapas_physio_cfg_matlabbatch.m` file within `spm12/toolbox/anyFolderName`.

With respect to the error, I am a bit at a loss still. Some shots in the dark:
1. I presume you are using the latest release of TAPAS (and PhysIO) downloaded here on GitHub?
2. If yes, did you restart SPM (or at least the batch editor via `spm_jobman('initcfg')`) before selecting PhysIO in the Batch editor?
3. Could you go into the function `tapas_physio_job2physio`, set a breakpoint just before line 32 and save the matlab workspace (by typing `save myWorkspace.mat`) and send it to me?
4. I understand this occurs as soon as you select the TAPAS PhysIO Toolbox from the Batch Editor `Tools` Menu, so there is no batch file you already loaded, is there?

Let's see how fast we figure this out, thank you for your input!

All the best,
Lars

RyanJM41, 2018-04-20:
Dear Lars,

Thank you very much for your response and for your assistance.

Here are the answers to your questions:

1. Yes, I am using the latest release of TAPAS and PhysIO.
2. Yes, I did restart SPM, using both spm_jobman and restarting Matlab entirely.
3. I have attached the updated function with the breakpoint here in this email.
4. Correct, there is no batch file I already loaded.

Thank you very much again for your time and help!

Kind regards,
Ryan

mrikasper, 2018-04-20:
Dear Ryan,

I see, thank you for your answers, that rules out the obvious candidates!

The attachment does not seem to have been attached (at least I don't see it here). If GitHub does not accept `.mat` files, please send it to my e-mail address, which you can find here: 

https://www.tnu.ethz.ch/en/team/faculty-and-scientific-staff/kasper.html

Thank you and all the best,
Lars

mrikasper, 2018-04-25:
Dear Ryan,

the `.mat` file you sent is not actually containing the workspace variables, but ist identical to the .m-file (`tapas_physio_job2physio.m`). Could you please try again to 

1. put a breakpoint (red dot) in front of line 32 of  `tapas_physio_job2physio.m` in the *Matlab Editor* (by clicking next to the line number).
2. Then type in the command window: `spm fmri`
3. Open the Batch Editor, Choose SPM->Tools->TAPAS PhysIO Toolbox

…This is what you did, when the error occurred, right?

If so, there should be a green arrow in front of the red dot in line 32 now, since the execution of the function has reached this point. Then click into the Matlab *Command Window* and type

`save myWorkSpace`

Sorry for these detailed instructions, this is also to double-check whether I understood everything correctly. On my machine, `job2physio` is not called until the green triangle (play button) in the Batch Editor is pressed, so the break point above is not reached by just choosing TAPAS PhysIO Toolbox from the Batch Editor menu.

All the best,
Lars

RyanJM41, 2018-04-27:
Dear Lars,

Thank you very much for your response and these detailed instructions. They were very helpful.

Upon following the instructions, I was successful in opening the TAPAS Physio Toolbox batch. I believe the error stems from me mistakenly trying to open the PhysIO toolbox via the dropdown toolbox menu in the SPM menu GUI, thinking it was via a GUI and not the batch editor that one uses the PhysIO toolbox.

When I performed the steps you recommended, I was able to launch the TAPAS PhysIO Toolbox via the batch editor (I have attached an image file of the batch as it appears on my PC). Does this seem to be correct?

Thank you very much again for your help.

Kind regards,
Ryan

mrikasper, 2018-06-07:
Dear Ryan,

this seems to be correct, even though I cannot see the image you attached.

Sorry for the confusion about the dropdown menu of the main SPM GUI (instead of the batch editor). We will clarify this in the instructions.

All the best,
Lars

closed 2018-06-07 by mrikasper