translationalneuromodeling/tapas #17: PhysIO on CMRR multiband data from Siemens Prisma

Yuqiabcd, 2018-04-20:
[Physio_20170803_120306_85d97388-a318-4a32-b3a9-bb10c2109660_Info.log](https://github.com/translationalneuromodeling/tapas/files/1933069/Physio_20170803_120306_85d97388-a318-4a32-b3a9-bb10c2109660_Info.log)
[Physio_20170803_120306_85d97388-a318-4a32-b3a9-bb10c2109660_PULS.log](https://github.com/translationalneuromodeling/tapas/files/1933070/Physio_20170803_120306_85d97388-a318-4a32-b3a9-bb10c2109660_PULS.log)
[Physio_20170803_120306_85d97388-a318-4a32-b3a9-bb10c2109660_RESP.log](https://github.com/translationalneuromodeling/tapas/files/1933071/Physio_20170803_120306_85d97388-a318-4a32-b3a9-bb10c2109660_RESP.log)

Hi tapas group,

  I am currently implementing the PhysIO toolbox on a dataset acquired using CMRR multiband sequences (factor=6). I am using Matlab command line by modifying the script comes with examples. I got Figures 1-5 (with verbose.level=2), but meanwhile getting glitches (no output generated). I am attaching the .log files I have and the script I am using. Here are messages I got from Matlab: 
-------------------------------------------------------------------------------------
>> PhysIO
Warning: First volume with missing slice events (144: 0 instead of 66 found)
 
> In tapas_physio_log (line 50)
  In tapas_physio_get_onsets_from_locs (line 73)
  In tapas_physio_create_scan_timing (line 114)
  In tapas_physio_main_create_regressors (line 141)
  In PhysIO (line 322) 
    maxscan (incl. dummies) = 135 
    tmin (1st scan start (1st dummy))=  20.14 s
    tmin (1st scan start (after dummies))=  20.14 s
    tmax = 154.05 s 
    mean TR =   0.99 s

Warning: No figures found to save to file 
> In tapas_physio_log (line 50)
  In tapas_physio_print_figs_to_file (line 36)
  In tapas_physio_main_create_regressors (line 401)
  In PhysIO (line 322) 
Output argument "R" (and maybe others) not assigned during call to "tapas_physio_main_create_regressors".

Error in PhysIO (line 322)
[physio_out, R, ons_secs] = tapas_physio_main_create_regressors(physio);
------------------------------------------------------------------------------------------------------

Any help is appreciated!

Thank you,
Yuqi

Yuqiabcd, 2018-04-20:
Just realized I couldn't attach .m file. Here I am attaching my script as .zip file.
[PhysIO.zip](https://github.com/translationalneuromodeling/tapas/files/1933107/PhysIO.zip)

mrikasper, 2018-04-26:
Dear Yuqi,

I found two issues when studying your script:

1. You have to specify an onset slice in order for the model to run. I therefore changed
`sqpar.onset_slice       = []; -> sqpar.onset_slice       = 33;`
    - There was no warning about that, I will add this to the next PhysIO release.
2. You specified `log_files.relative_start_acquisition = 10.07;` which means you assume a shift between the 2 logfile timings of about 10 seconds.
However, for the `Siemens_Tics` format, you usually don't have to change the default (`0`) here, since physiological recordings (`.RESP`, `.PULS`) use the same time scale as the scan timing (`.INFO`) files.
    - I will add a clearer description about the parameter in the next release.
    - In your script, I just commented out the respective line of code.

I adapted your script accordingly (see attached m-file, you have to rename the `.txt` extension), and it runs through on my machine.
[PhysIO_LK.txt](https://github.com/translationalneuromodeling/tapas/files/1951990/PhysIO_LK.txt)

Three further suggestions and comments:
1. Often it helps to switch the verbosity to a debug level (>=3) to understand problems, i.e., set `verbose.level = 3`
2. If you have SPM installed anyway, using the Batch Editor and creating the jobfile with the GUI is typically less error prone. The interactive help is more verbose, and these examples are typically checked more frequently than the matlab scripts.
3. The syntax in the Siemens example for specifying the models (RETROICOR) is deprecated, sorry for not adapting the example. It should read 
```
model.retroicor.include = 1;
model.order = struct('c',3,'r',4,'cr',1, 'orthogonalise', 'none');
```
instead of 
```
model.type = 'RETROICOR';
model.order = struct('c',3,'r',4,'cr',1, 'orthogonalise', 'none');
```
This did not create a problem in your case, since these lines specify the default RETROICOR settings anyway (set by `tapas_physio_new`), but also no warning was issued. Had you changed, e.g. the model order, in the old syntax, it would not have created a new model. The attached script is corrected.

All the best,
Lars

mrikasper, 2018-05-02:
Dear Yuqi,

- `sqpar.onset_slice` determines for which slice the regressors are sampled. 
    - Most neuroimaging analysis packages (for example, SPM), use the same design matrix to model the time series of all voxels in all slices. Therefore you will have to decide for which slice your physiological noise regressors will have the exact timing. 
    - This `onset_slice` is typically the one you are most interested in, for example, because your region of interest is in there, or a region that is subject to a lot of physiological noise.
    - If you have no a priori preference, the middle slice is typically a good compromise, since the timing of any slice will be off by less than half a TR.
    - For RETROICOR, since you include both `cos` and `sin` terms, these constant, slice-timing related delays can be corrected by the GLM fit itself, given your data has enough SNR, because a constant physiological phase shift can be expressed by the [trigonometric addition formula](http://mathworld.wolfram.com/TrigonometricAdditionFormulas.html) for `'cos(\phi_{phys} + \delta \phi_slice)'`
    - If you use slice timing correction, you should use the same slice here as for the slice timing correction (but note that PhysIO always uses the temporal index of the slice, i.e., 33 would mean the slice that was acquired as the 33rd...which is not necessarily the spatially central slice, if you have differing slice orders. For ascending slice order and no interleaving, both are equal).
- `sync.method` should be set to `'scan_timing_log'` for the `Siemens_Tics` format. I will add this information into the help, you are right. Basically, you always have the option to choose `'nominal'` to create your own timing, that is why I did not make it a default, and other vendors handle it differently. Once we have all the examples updated, I think this should become much clearer as well.
- Data-driven noise correction is possible via the `noise_rois` model. 
    - This is very similar to the aCompCor model of *Behzadi et al, 2007*, but we chose to use the mean time series and principal components of noise ROIs without any adaptation with respect to task regressors. In a way, PhysIO's implementation might be more conservative but less sensitive to task effects.
    - In order to use it, you will have to set `model.noise_rois.include = 1` and specify both the preprocessed fmri file (usually the smoothed warped functional 4D image) and the rois from which to extract the noise time series for PCA. Since you can use tissue probability maps (TPM) directly here, just choose the warped white matter and CSF (`wc2<anat>.nii, wc3<anat>.nii`) TPMs from SPM's Unified Segmentation here. Any binary masks or intensity images that can be thresholded are also fine as long as they are in the same standard space as the functional image you provide.
    - There will be more documentation once the examples are updated here as well. I have pasted the section of `tapas_physio_new` below, which gives a minimum explanation of the parameter usage and meaning:
```
%% noise_rois (Model): Anatomical Component Correction, Behzadi et al, 2007
    % Principal Components of time series of all voxels in given regions 
    % of localized noise, e.g. CSF, vessels, white matter
    %
    % e.g. CompCor: Behzadi, Y., Restom, K., Liau, J., Liu, 
    % T.T., 2007. A component based noise correction method (CompCor) 
    % for BOLD and perfusion based fMRI. NeuroImage 37, 90-101. 
    % doi:10.1016/j.neuroimage.2007.04.042
     
    model.noise_rois.include = 0;
    % cell of preprocessed fmri nifti/analyze files, from which time series
    % shall be extracted
    model.noise_rois.fmri_files = {}; 
    % cell of Masks/tissue probability maps characterizing where noise resides
    model.noise_rois.roi_files = {};
    % Single threshold or vector [1, nRois] of thresholds to be applied to mask files to decide
    % which voxels to include (e.g. a probability like 0.99, if roi_files
    % are tissue probability maps)
    model.noise_rois.thresholds = 0.9;
    
    % Single number or vector [1, nRois] of number of voxels to crop per ROI
    % default: 0
    model.noise_rois.n_voxel_crop = 0;
    
    % Single number or vector [1, nRois] of numbers
    % integer >=1:      number of principal components to be extracted
    %                   from all voxel time series within each ROI
    % float in [0,1[    choose as many components as needed to explain this
    %                   relative share of total variance, e.g. 0.99 =
    %                   add more components, until 99 % of variance explained
    % NOTE: Additionally, the mean time series of the region is also
    % extracted
    model.noise_rois.n_components = 1;
    ```

All the best,
Lars

amr-farahat, 2018-08-02:
Hello Lars,

I also have a question regarding the multiband acquisition sequence.
How should we choose these parameters 
**scan_timing.sqpar.time_slice_to_slice** since there is simulatenous slice aquisition (Factor = 3).
and
**scan_timing.sync.method** 
Can I manually add the slice aquisition times here?

amr-farahat, 2018-08-07:
Hello Lars,

I found somehow a hacky solution to the problem, but I want to make sure an assumption I have is correct for the solution to be valid. Is the model agnostic to the locations of the slices recorded simultaneously? Going through the code, I realized that the variables VLLOCs and LOCS contain the timepoints of the volumes and slices acquisition events and for the LOCS, it just has the information that at a specific timepoint 6 slices (for example) are acquired, but not which slices in particular. 
If this assumption is correct then I could overwrite the LOCS variable generated by the **'tapas_physio_create_scan_timing_nominal'** function afterwards with a manually constructed variable in the same style.
Thanks in advance.

mrikasper, 2018-08-10:
Dear Amr,

sorry for the delayed response and thank you for digging a bit into the code in the meantime! 

Concerning the use of PhysIO for Multiband imaging, also have a look at [this comment](https://github.com/translationalneuromodeling/tapas/issues/6#issuecomment-364979595) in a different thread.

Basically, as you stated correctly, PhysIO is (at the moment) agnostic to spatial slice location. It just counts slice excitation events, such that the index you specify, e.g., in `onset_slice`, refers to the temporal order, i.e., at which time during `TR` did you excite this slice. So, for example, if you have 36 slices and MB-factor 3 in a TR of 1 second, and within TR, you excite spatial slice `[1 13 25]`, then `[2 14 26]`, then `[3 15 27]`, then `[4 16 28]` etc., you would specify - if you are interested in modeling physiological noise for slice 16 (or 4 or 28)

- `TR = 1s`
- `nSlices = 36/3 = 12`
- `onset_slice = 4`

I believe this does not need any additional coding on your side. If you wanted, you could indeed include a custom logfile for the scan time events. We have implemented something like that for Siemens VD/VE logfiles, where you specify a file holding slice excitation times:

- `log_files.scan_timing = 'Physio_20180503_100106_85d97388..._Info.log'`
- `log_files.vendor = 'Siemens_Tics'`
- `scan_timing.sync.method='scan_timing_log'`

See `tapas_physio_create_scan_timing_from_tics_siemens()` for the implementation. If you add your own custom format, you could also extend the interface of the wrapper function `tapas_physio_create_scan_timing()`, and the GUI in `tapas_physio_cfg_matlabbatch()`.

Concerning your first question:
- `time_slice_to_slice` is only important, if TR is not minimal for the number of slices you want to acquire, i.e., if there is a greater temporal gap between the last slice of volume `n` and the first slice of volume `n+1` the between two consecutive slices of volume `n`.

I hope that helps.

All the best,
Lars

amr-farahat, 2018-08-12:
Dear Lars,

Thank you very much for your response. It is very helpful. Just one follow-up question regarding the onset_slice. As you mentioned earlier in this thread, this should be my slice of interest (the slice containing the seed for FC analysis for example) or the reference slice if I did slice time correction. 
I just want to make sure that it is acceptable to do physiological noise regression after slice time correction and the other steps in the preprocessing pipeline (motion correction, normalization, and smoothing).

Kind regards,
Amr

mrikasper, 2018-08-17:
Dear Amr,

this comes down to a philosophical question. 

- For task data, PhysIO is set up to model the physiological noise as part of the same GLM that also models your task (as nuisance regressors, just as for e.g. motion parameters). So it will be performed *after* all preprocessing. I think this is reasonable for all preprocessing steps, for the following reasons:
    1. *slice timing correction*: In SPM, you only model one GLM for all voxels, so via slice timing correction, you select a common timepoint within each volume TR that both your model and your data is aligned to. 
        - To a certain degree, modeling of derivatives (for RETROICOR, the `sin` and `cos` of the physiological phases are both modeled) can take care of this as well, but the Sladky 2011 paper showed convincingly that prior knowledge on slice timing and its correction is superior to derivative-modeling for these kind of timing issues.
    2. *motion correction*: You model the time series of each individual voxel. So for each volume, a voxel should cover the same part of the brain, otherwise you have intensity differences due to motion and mislocation of physiological effects (imagine oscillating between gray matter and a ventricle due to movement)
    3. *normalization*: Since all volumes are warped the same way, the modeled time series per voxel will not be altered. There might be some mixing of adjacent time series due to the interpolation involved, but I would consider it a minor effect.
    4. *smoothing*: In a way, this improves your fit of the physiological noise, since physiological noise is often correlated in adjacent voxels (e.g., in the CSF) - the same way the BOLD effect is increased by this local averaging. 
        - So, if you do physiological noise correction, you probably do it better after smoothing (unless anticorrelated physiological effects mix)
        - On the other hand, this means that smoothing *without* physiological noise correction is a bad idea, because you will often amplify effects and even project them into adjacent regions previously unaffected by physiological noise (e.g. large vessel signal into gray matter).
        - People often mask out typical physiological noise regions to avoid this effect, in particular CSF, before smoothing. This works, as long as you can identify them easily anatomically, and the effect of physiological noise correction on such data is typically much more modest. I don't think it is easy to mask out everything, though, in particular vessel structure.
- On a more general note, I like the approach of having only one GLM after preprocessing (instead of projecting out noise in a separate noise GLM), because keeping it in the same model with your task allows you to be very conservative, while not introducing anti-correlations: only the unique variance of the time series that can be explained by your task exclusively - and not by physiological noise - will end up as significant effect in your statistical inference. 
    - Of course, you could have false negatives, if your task induces physiological changes (breathing volume or rate, heart rate), but the truth there is that you can just not distinguish them from physiological noise.
- For resting state data, these considerations are not so important, and a nuisance regression can be performed at any stage (you just want to get rid of voxel-to-voxel time series correlation driven by physiology). There are the usual caveats with this (e.g., the Murphy and Bright paper), i.e., don't have a too complex model (10x or more datapoints than regressors should be fine) and preprocess your regressors the same way as your data (especially filtering).
    - mind you, this only goes for resting state and a nuisance-only regression. If you model your task with the nuisance factors, this is taken care of automatically, at least in SPM.

I hope this helps,
Lars