translationalneuromodeling/tapas #21: F-contrasts in dummy GLM for PhysIO performance assessment

mrikasper, 2018-06-28:
From a user:

> Dear Lars,
> 
> Hope my email finds you well. 
> 
> Recently, i have finished the first step suggested by you, that is 
> 1) create “dummy-glms” just containing RETROICOR/motion regressors in the design matrix, for each session individually (setting multiple regressors, but leaving multiple conditions empty), and then run tapas_physio_report_contrasts on them. tapas_physio_compute_tsnr_gains will work the same way.
> 
> But then i am confused about the second step. According to you, in the first step i shall create the F-contrast spanning the physiological noise regressors of all eight sessions manually. Can you elucidate this more precisely? Now for each session for each participant, i have a folder containing the 6 F contrasts for the physiology noise(namely all phys, all phys+move, caidiac, respiratory, card*Resp interaction, move), how shall i create F-contrast spanning the physiological noise regressors of all the sessions? Do you mean in the second level for each physiological noise? 
> 
> Hope my questions are not bothering you too much. I will appreciate if you can give me further information. 
> 
> 
> 
> Sincerely yours, 
> Yulin

mrikasper, 2018-06-28:
Dear Yulin,

in order to assess the quality of the physiological noise, correction, I would advise to check it for each session of each subject individually. This is because each session has their own physiological logfiles, which could vary in quality.

So, in essence, you should create a loop over subjects and sessions and call `tapas_physio_report_contrasts` within that. You can even specify to write the output to a postcript file, which you can evaluate conveniently at the end. 

BTW: Once you have specified the dummy GLMs, you don’t have to create the contrasts manually. `tapas_physio_report_contrasts` will do this automatically. Have a look at the header of this function (pasted below) for a set of all options.

Hope that helps,
Lars

```
function varargout = tapas_physio_report_contrasts(varargin)
% This function reports all relevant F-contrast-maps for physIO-created regressors
% in a given GLM
%
%  - Section plots on an anatomical overlay are created and saved as
%    post-script file with one page for each contrasts
%
%  - Input parameters are specified via name/value pairs, e.g.
%
%
%   args = tapas_physio_report_contrasts(...
%                   'fileReport', 'physio.ps', ...
%                   'fileSpm', 'analysisFolder/SPM.mat', ...
%                   'filePhysIO', 'analysisFolder/physio.mat', ...
%                   'fileStructural', 'anatomyFolder/warpedAnatomy.nii')
%
% IN
%
%   Required parameters:
%
%                  fileReport: sets the path and file name for the postscript 
%                              file your contrast report is printed to. 
%                              A reasonable (and default) location is the 
%                              single subject analysis folder so the file 
%                              resides with the GLM (fileSPM)
%                              default: 'pathSpm/physio_report_contrasts.ps';
%              fileStructural: structural underlay for results,
%                              e.g. 'mean.nii'
%                     fileSpm: SPM.mat holding physiological regressors,
%                              e.g.'SPM.mat'
%                  filePhysIO:  mat-file where PhysIO-object was saved
%                               e.g. physio.mat
%
%   Optional Parameters:
%
%          namesPhysContrasts: cell Array of contrast names in design matrix
%                              e.g. {'All Phys', 'Cardiac', 'Respiratory',
%                               'Card X Resp Interation',
%                               'HeartRateVariability',
%                               'RespiratoryVolumePerTime', 'Movement'}
%
%      indReportPhysContrasts: vector of contrast indices to be reported
%                               e.g. [1:7] for all physiological contrasts
%     reportContrastThreshold: 0.001
%    reportContrastCorrection: 'none' or 'FWE'
%    reportContrastMax:        maximum value of contrast colormap
%                              to scale different contrasts with equal
%                              F-values (default: Inf, scales to max F of
%                              map)
%      reportContrastPosition: 'max' or [1,3] vector of crosshair position
%                               (in mm)
%               fovMillimeter: field of view in mm; set to 0 for full FOV
%         doPlotSliceParallel: if true, slices are plotted parallel to
%                              their slice acquisition direction
%                       model: physio.model-substructure holding
%                              model orders, i.e. .c .r .cr
%                              See also tapas_physio_new
%         titleGraphicsWindow: additional title prepended to contrast name
%                              in each plot
%       saveTable              true of false (default)
%                              prints screenshot of results table to
%                              ps-file as well
%
% OUT
%   args    structure of default and updated arguments used in this
%           function; the fields of args hold all possible options of this
%           function
```

wangyulinatUGent, 2018-07-02:
Dear Lars, 

Thanks a lot for your reply! 

I just tried one participant's data as an example. Indeed i do not have to create the contrasts myself. 

What if the quality of the physiological noise correction is not good after checking per session? Shall i leave that session, so no regressors from that session will be entered for the GLM analysis? 

Best regards, 
Yulin

mrikasper, 2018-11-01:
Dear Yulin,

indeed, if there is no significant amount of variance explained by the PhysIO regressors (i.e. the F-map over regressors is sparse or empty), it makes sense to leave out these regressors for that session, because this increases your degrees of freedom in the model, and therefore sensitivity for effects of interest.

I hope that helps.

All the best,
Lars

closed 2018-11-01 by mrikasper