translationalneuromodeling/tapas #22: cannot include motion regressors when running PhysIO from matlab script?

switt4, 2018-07-19:
Hi,

I have been playing around with the PhysIO toolbox both through the SPM GUI and edited versions of the provided example Matlab scripts.  

When I input the rp*.txt file as another multiple regressor through the SPM batch system, it is added to the end of the RETROICOR regressors in the output multiple_regressors.txt file.  I am then able to use the super-helpful tapas_physio_report_contrasts and tapas_physio_compute_tsnr_gains commands with no issue (for some reason, the report_contrasts command assumes that my GLM includes the movement parameters, otherwise I get an error that the contrast vectors are too long for my specified model).

However, when I went to adapt one of the example scripts to match my input, the output multiple_regressors.txt file does not include the movement parameters as the last 6 regressors, despite the rp*.txt file being specified under model.input_other_multiple_regressors.  

I tried comparing the script generated by the SPM batch system with the provided example scripts, but it looks as if the physio structure is built differently depending on whether I use the SPM batch system or the matlab scripts.

Am I just missing something really obvious in specifying the structure via the scripts?

Thanks in advance,

suzanne

switt4, 2018-08-16:
Hi,

Thanks for the reply.  I figured out a similar solution in the interim using a combination of the spm batch script output and the manual.

The example script I was originally working from was included in the PhysIO_R2016 zip file I downloaded I believe from this link (https://tnurepository.ethz.ch/physio/physio-examples).  The specific file was "example_main_ECG3T.m" from the Philips subfolder, as this appeared to be the only example that included any sort of code for including motion regressors outside of Karl Friston's motion model.

One other question, is the toolbox supported as an interface in Nipype?  If so, how would I import it?

Thanks again,

suzanne

mrikasper, 2018-08-18:
Dear Suzanne,

thank you for the reference to the example download. This is indeed a version of the examples that is incompatible with the most recent PhysIO releases, e.g.,  R2017.3, which is included in TAPAS 2.7.4. There will be a new release of TAPAS soon, which will include the compatible examples right away.

With respect to Nipype, I have not worked with it myself, and do not know how they integrate SPM in general.
- Do they need a compiled SPM version or can they interface an installed Matlab version that contains SPM in its path?
- If you can execute jobs via the batch editor or `spm_jobman` via an installed Matlab version, then I think
- If, on the other hand, Nipype needs a compiled SPM version (or comes with one), then this compiled version would have to be compiled with PhysIO.
    - Here are some details about that from Guillaume Flandin (the SPM software manager), which I go a while back.
    - Basically, the answer is to re-compile SPM yourself, with the toolbox being in the right subfolder of SPM:
    >>>
The standalone version of SPM12 we distribute does not contain toolboxes
and you cannot add any as they would have had to be compiled at the same
time.

If you want to compile SPM12+PhysIO yourself, just copy PhysIO somewhere
in your SPM12 installation (eg the `toolbox` folder) and run
`spm12/config/spm_make_standalone.m` (provided you installed the MATLAB
Compiler).
    >>>
   - More information on the SPM standalone version can be found here:
https://en.wikibooks.org/wiki/SPM/Standalone

I hope that helps!

All the best,
Lars

--
Lars Kasper, PhD
Translational Neuromodeling Unit /
MR Technology and Methods Group

Institute for Biomedical Engineering
University of Zurich & ETH Zurich

Wilfriedstrasse 6 / Gloriastrasse 35
8032 Zurich / 8092 Zurich

phone: + 41 44 634 91 09 / +41 44 632 09 68
e-mail: kasper@biomed.ee.ethz.ch<mailto:kasper@biomed.ee.ethz.ch>

mrikasper, 2018-08-18:
Dear Suzanne,

thank you for the reference to the example download. This is indeed a version of the examples that is incompatible with the most recent PhysIO releases, e.g.,  R2017.3, which is included in TAPAS 2.7.4. There will be a new release of TAPAS soon, which will include the compatible examples right away.

With respect to Nipype, I have not worked with it myself, and do not know how they integrate SPM in general.
- Do they need a compiled SPM version or can they interface an installed Matlab version that contains SPM in its path?
- If you can execute jobs via the batch editor or `spm_jobman` via an installed Matlab version, then I think
- If, on the other hand, Nipype needs a compiled SPM version (or comes with one), then this compiled version would have to be compiled with PhysIO.
    - Here are some details about that from Guillaume Flandin (the SPM software manager), which I go a while back.
    - Basically, the answer is to re-compile SPM yourself, with the toolbox being in the right subfolder of SPM:
 

    > The standalone version of SPM12 we distribute does not contain toolboxes
    > and you cannot add any as they would have had to be compiled at the same
    > time.
    > 
    > If you want to compile SPM12+PhysIO yourself, just copy PhysIO somewhere
    > in your SPM12 installation (eg the `toolbox` folder) and run
    > `spm12/config/spm_make_standalone.m` (provided you installed the MATLAB
    > Compiler).
    - More information on the SPM standalone version can be found here:
https://en.wikibooks.org/wiki/SPM/Standalone

I hope that helps!

All the best,
Lars

closed 2018-10-08 by mrikasper