translationalneuromodeling/tapas #23: Cannot download HGF Toolbox from TNU website

ImcaH, 2018-08-08:
I am trying to download the HGF toolbox from the TNU website but the downloads page of TAPAS seems to be unavailable. Is there any other way that I can download the toolbox? Thank you in advance! Best, Imca

tnutapas, 2018-08-08:
Dear Imca,

Could you please indicate us where the broken link is? In any case, you can use the following link

https://github.com/translationalneuromodeling/tapas/releases/download/v2.7.4.1/tapas.ziphttps://github.com/translationalneuromodeling/tapas/releases/download/v2.7.4.1/tapas.zip

Regards,

E.

ImcaH, 2018-08-08:
Hi E, 

This is the link that I tried to follow: https://www.tnu.ethz.ch/en/software/tapas/download.html

Thank you for the other link!

Best,
Imca

mrikasper, 2018-08-08:
Dear Imca,

thank you for letting us know - the download link accidentally pointed to the TNU archive with older TAPAS versions, which is currently not online. However, I fixed the link and the download page should now redirect to the new GitHub page with all recent TAPAS versions.

All the best,
Lars

closed 2018-08-17 by mrikasper