translationalneuromodeling/tapas #24: PhysIO - errors when including Noise ROIs module

switt4, 2018-08-29:
Hi,

I was playing around with adding the aCompCor (Noise ROIs) module in addition to the RETROICOR in the SPM batch GUI.  I am using the realigned files as the fMRI input and a single WM+CSF segmentation file coregistered to the fMRI files as the Noise ROI image input (per the Behzadi 2007 paper).   

When I go to run the batch, I keep getting one of three error messages when the toolbox reaches the Noise ROI step:
1. If I keep the number of principal components set to the default of 1, then I get an error using horzcat, "Dimensions of matrices being concatenated are not consistent."
2. If I set the number of components >1, 'Index exceeds matrix dimensions' error.  This error seems to occur for any number of components >1 (iterated up to 10).
3. I finally, tried specifying a float (0.95) to let the toolbox determine the number of components.  This resulted in the error message, "In an assignment A(:)=B, the number of elements in A and B must be the same".

I was able to get a bit more information about where the error is occurring by running this as a script directly from matlab:

Error in tapas_physio_create_noise_rois_regressors (line 171)
    R = [R, COEFF(:,1:nComponents)];

Error in tapas_physio_main_create_regressors (line 307)
        [noise_rois_R, model.noise_rois, verbose] =
        tapas_physio_create_noise_rois_regressors(...

There is not a lot of documentation about adding any of the other modules other than RETROICOR, so I am not sure how best to proceed.  

Thanks in advance,

suzanne

mrikasper, 2018-08-29:
Dear Suzanne,

it seems as if the matrix dimensions do not match for the PCA. A likely candidate is the input 4D functional image file. You will have to specify the same number of volumes for the fMRI input as for the realignment / GLM.

If that doesn't help, maybe you could send me your batch as .m-file.

All the best,
Lars

switt4, 2018-08-29:
Hi,

Thanks for the quick reply.

I checked both the input in the script and the SPM batch GUI, and I have 601 input files for the fMRI and 601 rows in the rp.txt file.  I'm assuming that I don't need to specify the ROI file 601 times to match?

I've attached the batch .m file and the matlab script I built in case I am missing something quite obvious.  I had to save both as a .txt files, as github does not support .m as a file type for attachments.

[example_3T_PPU_Siemens_RETROICOR_aCompCor.txt](https://github.com/translationalneuromodeling/tapas/files/2333234/example_3T_PPU_Siemens_RETROICOR_aCompCor.txt)
[tapas_batch_acompcor_job.txt](https://github.com/translationalneuromodeling/tapas/files/2333235/tapas_batch_acompcor_job.txt)

Thanks again,

suzanne