translationalneuromodeling/tapas #28: physIO extract breath duration or max inhalation/exhalation

jaetzel, 2018-10-03:
I would like to extract the time of each breath (max exhalation and/or max inhalation) detected by physIO. Is there something similar to ons_secs.cpulse for respiration? The "interpolated breath duration (secs)" timecourse from the attached figure would also work well. 

I have been unable to locate these derived respiration measures in the created physio.mat object. If they are present, please point the field out to me, and apologies. If not, is there a simple way to extract them, perhaps writing an additional csv when physIO is run?

Thanks!
Jo Etzel

[image]

mrikasper, 2018-10-07:
Dear Jo,

the function you are looking for is `tapas_physio_rvt()`. It already has the max exhalation/inhalation time points as 2nd and 3rd output parameter, though they are unused in the current PhysIO version:

```
[rvt, timeRpulseMax, timeRpulseMin, verbose] = ...
    tapas_physio_rvt(fr, t, sample_points, verbose)
```

You could just save these two vectors to a text file within that function, but it might be a bit of a hack.

Alternatively, you can pass it up to the `physio.ons_secs` structure as follows:

`tapas_physio_rvt()` is called by `tapas_physio_create_rvt_regressors` in line 55. So if you add a 2nd and 3rd return argument to that call (changes in capital letters):

```
[rvt, RPULSEINHALE, RPULSEEXHALE] = tapas_physio_rvt(ons_secs.fr, ons_secs.t, sample_points, verbose);
```

and return it from `tapas_physio_create_rvt_regressors` to `tapas_physio_main_create_regressors()`
in line 318 

```
    [convRVT, ons_secs.rvt, ons_secs.RPULSEIN, ons_secs.RPULSEOUT, verbose] = tapas_physio_create_rvt_regressors(...
                ons_secs, sqpar, model.rvt, verbose);
```
it should augment the physio-structure in the `.mat` file as desired. I put it down as a feature request for the next release.

All the best,
Lars

jaetzel, 2018-10-09:
Thanks! We implemented your second suggestion (adding the fields to ons_sec) and it's just what we needed.

closed 2018-10-09 by jaetzel