translationalneuromodeling/tapas #30: Undefined function or variable 'VOLLOCS'.

d3christ, 2018-10-31:
Hey,

I am trying to incorporate PhysIO into our labs pipeline. I ran the Siemens_VD example script, switching out variables for my participants data, and came across this error

`>> testing_physio_11053_1
Undefined function or variable 'VOLLOCS'.

Error in tapas_physio_create_scan_timing (line 115)
    ons_secs.t, VOLLOCS, LOCS, sqpar, verbose);

Error in tapas_physio_main_create_regressors (line 149)
                log_files, scan_timing, ons_secs, verbose);

Error in testing_physio_11053_1 (line 57)
physio = tapas_physio_main_create_regressors(physio); ` 

The script does print the Raw Physio Logfile Data figure but none of the regressors.
[Here is a folder with my dicom, 3 log files and Matlab script.](https://github.com/d3christ/Physio_Help)

Any help is much appreciated, 

Thanks!

mrikasper, 2018-11-01:
Dear d3christ,

thank you for your interest in the toolbox.

I have updated the script so that it now runs through in [this fork](https://github.com/mrikasper/Physio_Help) of your repository.

Basically, I changed two things:

1. `physio.scan_timing.sync.method = 'scan_timing_log';`
    - You had specified the dicom file, but this parameter here should be a string defining the method. The file related to that should be defined in `physio.log_files.scan_timing`, as you did correctly. And for the Siemens VD, using the `*_Info.log` file is recommended, rather than the DICOM-file.

2. `physio.preproc.cardiac.initial_cpulse_select.method = 'auto_matched';`
    - I changed that back from `'load_from_logfile'`, because the vendor peak detection failed to detect a substantial proportion of the cardiac pulse events. In general, I always recommend the 'auto_matched' variant, since our peak detection algorithm in PhysIO is rather robust and takes information from the whole time series into account.

One last general remark about help on valid parameter sets (quoted from the PhysIO README):

- Within SPM: All toolbox parameters and their settings are explained in the 
   Help Window of the SPM Batch Editor
- Within Matlab: Extensive header at the start of each `tapas_physio_*` function and commenting
    - accessible via `help` and `doc` commands from Matlab command line
    - **starting point for all parameters (comments within file)**: `edit tapas_physio_new` 

I hope that helps!

All the best,
Lars

d3christ, 2018-11-01:
Thank you soo much for the thorough reply!!

I was trouble shooting it this morning and I was luckily able to figure out the scan_timing_log parameter fix, however I would not have been able to fix out the auto_matched cpulse.

All that aside I think there is a bigger problem here. I'm not sure if you noticed,(looking at the "Raw Physiological Logfile Data" output figure), but there seems to be trigger drop outs in our physio data systematically over time. (this would be the blue line going up to .5 in blocks where as it should consistently show a peak at each TR, right?) We noticed this was happening in all of our subjects data.

We contacted our scanner admin and they came back to us with

"Basically there is no external trigger hooked up to the PMU.  HCP originally required this and we got it all set up, but then they decided not to use external triggers.  Some sites were having issues splitting the TTL pulse.  It seemed to be working fine at our site though, but we disconnected it just in case.  This is what they said regarding it. 

"“The decision to back out was because the payout didn’t seem worth the effort and potential risk. The method of logging physio data to DICOM provides an additional timing source that it appears we can use instead of the EXT triggers. So, after discussing with PIs, we’re not going to ask other sites to risk splitting the EXT signals, and we’re going to rely on the DICOM timing instead.”"

So technically maybe there shouldn’t be any triggers in the graph?

That being said I think we need to ignore the trigger info in the log files and just rely on the dicom timing. Is that is possible for us to do in PhysIO?

Sorry for the long post, been a long day of trouble shooting and trying to understand these trigger dropouts!!

Thanks again,
Dylan

mrikasper, 2018-11-02:
Dear Dylan,

in short: I think everything is fine with the data and you can go on using the `*_Info.log` file for scan timing, rather than the DICOM.

I attached a screenshot to make sure I understand your interpretation correctly:

[image]

Do you mean the cyan peaks labelled as scan trigger events, that are missing in a lot of places? As you can see in the zoomed version, they always coincide with peaks of the PPU pulse wave:

[image]

The label "scan trigger events" might be a bit misleading here, but I used it to harmonize nomenclature between vendors. In principle, these are all codes that *could* trigger a scan event by the scanner, i.e., cardiac peaks, respiratory peaks, or external triggers. So indeed, none of the trigger events in the plot are external triggers, as your scanner admin pointed out, because your have no TTL device connected.

What it rather shows is that the cardiac pulse events detected by the scanner are incomplete, as I mentioned in my previous post, and rather severely so. Therefore I recommended the `auto_matched` method for cardiac pulse detection.

You don't have to worry for the Siemens VD physiological logging ("*automatic logging*" of the CMRR sequence) about receiving trigger events, since the  `*_Info.log` file registers the acquisition time for each individual slice in the same time scale (the *Tics* format) as the `*_Puls.log` and `*_Resp.log` file. So the alignment is already good given these three files. More details can be found in our [PhysIO Wiki Page on File Read-In](https://gitlab.ethz.ch/physio/physio-doc/wikis/MANUAL_PART_READIN).

I hope that helps!

All the best,
Lars

d3christ, 2018-11-02:
This was very helpful. Thank you so much. My PI and I are very appreciative of you taking time out of your day to help us out! 

Thanks again!

closed 2018-11-22 by mrikasper