translationalneuromodeling/tapas #37: PhysIO, error in `tapas_physio_main_create_regressors` if heart beat present in the first sample

hstojic, 2018-12-17:
Hello,

PhysIO produces the following error if I run it with cardiac data where heart pulse is present already in the first sample. The problem goes away if I simply eliminate these 1st sample pulses (which should not cause a big issue in any case).
```
No scan trigger events provided
    maxscan (incl. dummies) = 420 
    tmin (1st scan start (1st dummy))=   0.00 s
    tmin (1st scan start (after dummies))=   0.00 s
    tmax = 608.90 s 
    mean TR =   1.45 s

In an assignment  A(:) = B, the number of elements in A and B must be the same.

Error in tapas_physio_get_cardiac_phase (line 76)
    scannertpriorpulse(i) = pulset(n);

Error in tapas_physio_create_retroicor_regressors (line 90)
            tapas_physio_get_cardiac_phase(cpulse, spulse, verbose, svolpulse);

Error in tapas_physio_main_create_regressors (line 287)
                tapas_physio_create_retroicor_regressors(ons_secs, sqpar, ...
```

Here are the files and a script I ran:
[pulse_issue.zip](https://github.com/translationalneuromodeling/tapas/files/2686465/pulse_issue.zip)

hstojic, 2019-01-24:
Apologies for spamming the issues these days, is this solution of simply eliminating a pulse in the 1st observation OK? 

If it is perhaps we can close the issue. In the long run though its probably good to take care of such corner case within the toolbox.

mrikasper, 2019-01-24:
Dear Hrvoje,

this sounds reasonable. I will have a look at your data and see whether it can be easily fixed within the toolbox.

All the best,
Lars

mrikasper, 2019-01-24:
Dear Hrvoje,

it might be an equivalent solution, but i fixed it now in the [last commit](https://github.com/translationalneuromodeling/tapas/commit/37d1927822bc5e1a9fa98ae50ef4ec96e0c3c33c) to the development branch by using an already existing functionality of guessing additional pulses before the scan start, using the estimated heart rate at the time.

closed 2019-01-29 by hstojic