translationalneuromodeling/tapas #38: PhysIO, error in `tapas_physio_findpeaks_template_correlation (line 210)`

hstojic, 2018-12-17:
Hello, 

I encountered an error that I cannot easily find a fix. Here is the error output:
```matlab
No scan trigger events provided
    maxscan (incl. dummies) = 426 
    tmin (1st scan start (1st dummy))=   0.00 s
    tmin (1st scan start (after dummies))=   0.00 s
    tmax = 617.60 s 
    mean TR =   1.45 s

Warning: Guessed additional cardiac pulse at time series start for phase estimation 
> In tapas_physio_log (line 55)
  In tapas_physio_get_cardiac_phase (line 50)
  In tapas_physio_create_retroicor_regressors (line 90)
  In tapas_physio_main_create_regressors (line 287) 
Warning: Ignoring extra legend entries. 
> In legend>set_children_and_strings (line 658)
  In legend>make_legend (line 334)
  In legend (line 282)
  In tapas_physio_get_cardiac_pulse_template (line 210)
  In tapas_physio_get_cardiac_pulses_auto_matched (line 56)
  In tapas_physio_get_cardiac_pulses (line 88)
  In tapas_physio_rvt (line 68)
  In tapas_physio_create_rvt_regressors (line 55)
  In tapas_physio_main_create_regressors (line 318) 
Subscript indices must either be real positive integers or logicals.

Error in tapas_physio_findpeaks_template_correlation (line 210)
        amplitudeWeight = abs(cPadded(n+searchPosition+1));

Error in tapas_physio_get_cardiac_pulses_auto_matched (line 72)
            c, pulseCleanedTemplate, cpulseSecondGuess, averageHeartRateInSamples, ...

Error in tapas_physio_get_cardiac_pulses (line 88)
                    tapas_physio_get_cardiac_pulses_auto_matched( ...

Error in tapas_physio_rvt (line 68)
timeRpulseMax = tapas_physio_get_cardiac_pulses(t, fr, ...

Error in tapas_physio_create_rvt_regressors (line 55)
rvt = tapas_physio_rvt(ons_secs.fr, ons_secs.t, sample_points, verbose);

Error in tapas_physio_main_create_regressors (line 318)
            [convRVT, ons_secs.rvt, verbose] = tapas_physio_create_rvt_regressors(...
```

It seems like tapas_physio_findpeaks_template_correlation is producing an error, due to searchPosition (half of the averageHeartRateINSamples) being absolutely larger than n variable (bestPosition), resulting in a negative index. Perhaps an unusually high `averageHeartRateInSamples` ?

hstojic, 2019-01-22:
Any suggestion how to correct the issue?

mrikasper, 2019-01-24:
Dear Hrvoje,

I pushed a hotfix to the development branch: here is the [commit diff](https://github.com/translationalneuromodeling/tapas/commit/c34d94c81de403dd918ea6b035d17e4dcc08865a)

It restricts the valid `searchPosition` for early (and late) detected peaks. The detected peak itself still looks not trustworthy (see screenshot from start of your example timeseries), but the algorithm is robust enough to get it right for the later peaks.

Anyway, I labeled it as a hotfix, maybe there is a more principled approach.

All the best,
[image]

Lars

hstojic, 2019-01-24:
Thanks a lot for this, I will finally be able to include this subject in my fMRI analysis :)

At some point I will have to bite the bullet and go over each run manually to check them and make corrections where necessary. I did learn the lesson though, always get the analog data!

closed 2019-01-24 by hstojic