translationalneuromodeling/tapas #39: Matlab compatibility

CarolineLndl, 2018-12-18:
Dear Lars,
I was not able to use the PhysIO Toolbox because it would appear that a function (strel.m) is associated with  the Image Processing Matlab Toolbox. Is it possible to avoid the use of the Image Processing Toolbox ?

Error message :
Done    'Coregister: Reslice'
Done
Failed  'TAPAS PhysIO Toolbox'
Undefined function 'strel' for input arguments of type 'char'.
In file "/home/anne/MATLAB/tapas/PhysIO/code/tapas_physio_create_noise_rois_regressors.m" (???), function "tapas_physio_create_noise_rois_regressors" at line 112.
In file "/home/anne/MATLAB/tapas/PhysIO/code/tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 335.
In file "/home/anne/MATLAB/tapas/PhysIO/code/tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1520.

The following modules did not run:
Failed: TAPAS PhysIO Toolbox

Thanks in advance,

Caroline

mrikasper, 2018-12-18:
Dear Caroline,

as it happens, another user, Benoît Beranger, just added an improved version to the noise rois extraction relying on `spm_erode` instead of the image processing toolbox (see the [pull request](https://github.com/translationalneuromodeling/tapas/pull/34) with the changes). It is still not fully tested and not released yet, but you can clone/download it with the [TAPAS development branch](https://github.com/translationalneuromodeling/tapas/tree/development).

I hope that helps!

All the best,
Lars

CarolineLndl, 2019-01-04:
Dear Lars,
Thanks a lot for your quick answer and the new versions of the scripts.
Physio seems to work well with this changes.

Best wishes for the new year.
Caroline

closed 2019-01-23 by mrikasper