translationalneuromodeling/tapas #43: BUG: PhysIO, Error in tapas_physio_crop_scanphysevents_to_acq_window

hstojic, 2019-01-24:
I was trying out the new BIDS support in the development branch. I have a remaining issue with one run. The following error pops up:

```matlab
TR = 1.450 +/- 0.010 s (Estimated mean +/- std time of repetition for one volume)
Index exceeds array bounds.

Error in tapas_physio_crop_scanphysevents_to_acq_window (line 87)
if ~isempty(r),  r      = r(acqwindow); end;

Error in tapas_physio_main_create_regressors (line 215)
            [ons_secs, scan_timing.sqpar, verbose] = tapas_physio_crop_scanphysevents_to_acq_window(...

```

Here is the data and script I used:
[physio_s038_r06.zip](https://github.com/translationalneuromodeling/tapas/files/2791536/physio_s038_r06.zip)

I think this is because first trigger actually occurs before physio recordings began (as indicated by positive value in the json file). With this subject we accidentally didnt start recording before starting the run. I worked out manually exact time when it should have occured. The new development version cannot handle this for some reason.

hstojic, 2019-01-25:
Hi Lars,

thanks for a quick fix of the issue.

I'm not sure I understand where the jump comes from. Is the problem in discontinuous jump between 0s and signal? if so, perhaps prepending with some sine function data that would be filtered away?

Hrvoje

mrikasper, 2019-01-25:
Dear Hrvoje,

the jump comes from the way I fill up the missing timepoints - without any assumptions, just by using zeros...this of course poses a problem of the filtering step that happens for the respiratory time series....maybe prepending it with the mean of that time series is already better, or some clever guessing, e.g,, with the identified templates of a breathing/cardiac cycle. Similarly, using a sine wave (with the principal breathing frequency?) would need some processing....or are you suggesting a frequency that will be filtered out by the filter? 

In the workflow at the moment, the finding of peaks/template generation usually happens much later, that's why it needs some thinking.

All the best,
Lars

hstojic, 2019-01-25:
Hi Lars,

thanks for additional clarifications. Its a bit out of my depth, so I'll take it as it is right now.

Best,
Hrvoje

closed 2019-01-25 by hstojic