translationalneuromodeling/tapas #46: PhysIO CMRR Multiband Pulse only

alexsayal, 2019-02-07:
Hello,

I am acquiring physiological data on a Siemens Prisma with the proprietary PPU and respiratory cushion, using a CMRR multiband sequence.
Since the respiratory signal fails more than what I am willing to accept, I decided to use the pulse data only. I am exporting *_Info.log, *_RESP.log, and _PULS.log files.

If I provide the PULS and Info files only, I get the following error:
```
Error in the scan trigger data.
Warning: No respiratory time series provided 
Warning: First volume with missing slice events (374: 0 instead of 42 found)
07-Feb-2019 16:54:46 - Failed  'TAPAS PhysIO Toolbox'
Index exceeds matrix dimensions.
In file "C:\Users\alexandresayal\Documents\MATLAB\spm12\toolbox\PhysIO\tapas_physio_crop_scanphysevents_to_acq_window.m" (???), function "tapas_physio_crop_scanphysevents_to_acq_window" at line 70.
In file "C:\Users\alexandresayal\Documents\MATLAB\spm12\toolbox\PhysIO\tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 215.
In file "C:\Users\alexandresayal\Documents\MATLAB\spm12\toolbox\PhysIO\tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1520.
```

Could you provide some help on this?
I attach the log files, error data and batch used.

Thank you in advance,

[batchfilesiemens_cardiac_only.zip](https://github.com/translationalneuromodeling/tapas/files/2841904/batchfilesiemens_cardiac_only.zip)
[Physio_20181203_151116_9b6683d7-1925-4bdb-ae00-3365fff615ef_Info.log](https://github.com/translationalneuromodeling/tapas/files/2841899/Physio_20181203_151116_9b6683d7-1925-4bdb-ae00-3365fff615ef_Info.log)
[Physio_20181203_151116_9b6683d7-1925-4bdb-ae00-3365fff615ef_PULS.log](https://github.com/translationalneuromodeling/tapas/files/2841900/Physio_20181203_151116_9b6683d7-1925-4bdb-ae00-3365fff615ef_PULS.log)
[Physio_20181203_151116_9b6683d7-1925-4bdb-ae00-3365fff615ef_RESP.log](https://github.com/translationalneuromodeling/tapas/files/2841901/Physio_20181203_151116_9b6683d7-1925-4bdb-ae00-3365fff615ef_RESP.log)

mrikasper, 2019-02-08:
Dear Alex,

thank your for your detailed error description and, of course, for using the toolbox.

I was able to identify the problem, and resolve it in the following [commit](https://github.com/translationalneuromodeling/tapas/commit/7e98426bbd746d5477d433f907a9defd0c710b38).

In short, the new logfile version of Siemens you are using is able to skip certain 'tics' (i.e. time units of 2.5 ms), and only write out every `SampleTime  = 2` (or 8 for Resp) sample into the logfile, as indicated in its header. I wasn't aware of this feature, and the new commit takes care of that. 

Note besides that the `log_files.sampling interval` is then `2*2.5 ms = 5 ms`, not 2, as you put in the batch. I also changed the default behavior of PhysIO here now, ignoring user input for this parameter, since the sampling times are completely defined by the logfile, which is less error prone than user input.

You can now just clone or download the [TAPAS development branch](https://github.com/translationalneuromodeling/tapas/tree/development) and it should work.

Let me know if you are still having problems!

All the best,
Lars

alexsayal, 2019-02-11:
Dear Lars,

I can confirm that it is now working as expected.

Thank you for your fast response!

mrikasper, 2019-02-11:
You are welcome!

I am happy this solved the issue!

All the best,
Lars

closed 2019-02-11 by mrikasper