translationalneuromodeling/tapas #47: Filtering out white matter and CSF noise time series with PhysIO

mrikasper, 2019-02-13:
Dear TAPAS-users,

I recently answered a user request on NITRC of how to perform nuisance regression when you don't have physiological logfiles.

In short, PhysIO can do that, extracting the relevant nuisance regressors, and SPM plus the batch editor help to do the rest, i.e., ending up with a filtered fMRI time series Nifti file that you may use, for example, for a resting state connectivity analysis. This way, you can setup your whole preprocessing pipeline in SPM.

Here are the details, including an example batch of the whole procedure: 

https://www.nitrc.org/forum/forum.php?thread_id=9959&forum_id=2

pbaue, 2019-07-11:
Dear Lars, 

Thank you for posting this extra information on implementing PhysIO in SPM. 
This being my first fMRI project, I followed the steps described here: https://www.nitrc.org/forum/forum.php?thread_id=9959&forum_id=2. After realignment, slicetiming,coregistration, segmentation and normalisation,  I used PhysIO, using just the model and no real cardiorespiratory data.
Then I specified the GLM with the output of PhysIO (5 components for ROI1 and 5 for ROI2 + the six movement parameters from SPM), leaving the masking threshold at 0.05, then combining the residuals into a 4D file,  and smoothing the resulting .nii file (FWHM [10 10 10]).

I then proceeded to do a level 1 analysis with the smoothed noise-filtered time series, but got the following error: 
Chunk  14/14                            :                  ...processing
Failed  'Model estimation'
Error using spm_est_non_sphericity (line 208)
Please check your data: There are no significant voxels.

Displaying the smoothed-noise-filtered timeseries and comparing them with the unsmoothed-noise-filtered and the smoothed-timeseries with noise shows that the noise-filtered series look rather different, with the skull being visible (see attached images). The noise-filtered time series (2.5Gb) are much larger  than the not noise-filtered ones (640Mb) . 

I am now wondering whether my understanding of the procedure is correct: are the residuals that are combined into the 4D file in the procedure above the new time series that contain the task-related activation but no noise, or do they contain only the noise? i.e. is the 4D file the one that is used for subsequent level 1 analysis?

This is probably a naive question: why is it necessary to go through the procedure of writing the residuals and combining them into a new 4D file, instead of including the nuisance regressors that come out of the PhysIO module as regressors in the Level 1 analysis, just as one would do with the motion regressors of SPM?

Thank you in advance for your help!

best wishes, 
Prisca

[image]
[image]
[image]

mrikasper, 2019-07-17:
Dear Prisca,

thank you for trying out the PhysIO Toolbox. I have a suspicion concerning
your SPM error, but just to clarify one thing beforehand: Are you trying to
model resting state or task-based fMRI data?

The procedure I outlined in this issue/on NITRC was specific to "filtering out"
physiological noise in order to perform a resting state analysis afterwards,
e.g., assessing seed-based correlations, within the realms of SPM.

If you are using a task-based design, then indeed you can put the PhysIO output
multiple regressors `.txt` or `.mat` file into the same GLM as your task model,
just as with the realignment parameters (there are also options within PhysIO
    to estimate some other motion regressors, e.g., for motion censoring).

We have outlined the reasons of co-modeling task and nuisance regresssors within
one GLM in our paper on the PhysIO Toolbox (see section 2.7 there), which is a
   good start in general, if you want to learn about the methods behind the
   toolbox:

https://dx.doi.org/10.1016/j.jneumeth.2016.10.019

If you indeed try to model resting state data (and your second 1st level GLM
    contained, e.g., time series from individual regions), I presume there is an
issue with the mask threshold, which you should probably set to 0 (or at least
    much smaller than the default 0.8), since the filtered data don't contain
any mean any more, only residuals. You can check your `mask.img` visually via
SPM CheckReg to see whether it has any big holes.

Let me know if any of this helps.

All the best,
Lars

pbaue, 2019-07-18:
Dear Lars, 

Thank you for your reply, this definitely helps! 
I am indeed working on task-based data, so I will include the calculated regressors in the GLM with the task regressors. 

best wishes,  
Prisca

closed 2022-02-20 by mrikasper