translationalneuromodeling/tapas #51: PhysIO warnings "Invalid MinPeakHeight" and "Guessed additional cardiac pulse" when using the gradient for time synchronization

carolfs, 2019-03-15:
We are running the physio toolbox on data sets from several different studies, all performed in the same lab using a Phillips 3T scanner, and frequently getting the following warnings:

```
Warning: Invalid MinPeakHeight. There are no data points greater than MinPeakHeight. 
> In tapas_physio_findpeaks>getPeaksAboveMinPeakHeight (line 128)
  In tapas_physio_findpeaks (line 54)
  In tapas_physio_rescale_gradient_gain_fluctuations (line 66)
  In tapas_physio_create_scan_timing_from_gradients_philips (line 177)
  In tapas_physio_create_scan_timing (line 85)
  In tapas_physio_main_create_regressors (line 149)
  In tapas_physio_cfg_matlabbatch>run_physio (line 1520)
  In cfg_run_cm (line 29)
  In cfg_util>local_runcj (line 1688)
  In cfg_util (line 959)
  In cfg_ui>MenuFileRun_Callback (line 703)
  In gui_mainfcn (line 95)
  In cfg_ui (line 53)
```

```
Warning: Guessed additional cardiac pulse at time series end for phase estimation 
> In tapas_physio_log (line 55)
  In tapas_physio_get_cardiac_phase (line 61)
  In tapas_physio_create_retroicor_regressors (line 90)
  In tapas_physio_main_create_regressors (line 287)
  In tapas_physio_cfg_matlabbatch>run_physio (line 1520)
  In cfg_run_cm (line 29)
  In cfg_util>local_runcj (line 1688)
  In cfg_util (line 959)
  In cfg_ui>MenuFileRun_Callback (line 703)
  In gui_mainfcn (line 95)
  In cfg_ui (line 53)
```

For some participants and tasks we might get either warning or both. What do they mean? Should we be worried about them?

We have noticed that the gradient data is missing (the values are zero) in certain sections of the physio log or sometimes completely missing, and we will get one or both of the above warnings. When the gradient is missing, what does the PhysIO toolbox do? Does it fall back to the nominal method, and are the results reliable in this case?

Some of these warnings could be eliminated by changing the gradient_log.zero and gradient_log.slice parameters. We have also noticed that changing these parameters will yield completely uncorrelated regressors for the cardiac parameters, and the same is true for using the nominal method versus the gradient data. How should we set the gradient_log.zero and gradient_log.slice parameters for the best results? They are mentioned in the FAQ, but there's no explanation on how to set them optimally.

mrikasper, 2019-03-19:
Dear Carolina,

thank you for using the toolbox. The two warnings you get are of a very different nature

1. 
> ```
> Warning: Invalid MinPeakHeight. There are no data points greater than MinPeakHeight. 
> ...
>   In tapas_physio_create_scan_timing_from_gradients_philips (line 177)
>   In tapas_physio_create_scan_timing (line 85)
> ```
This warning indeed stems from invalid thresholds set in the `sync` module for  `sync.method = 'gradient_log'`, and should not be ignored, because it can lead to wrong or no synchronization between physiological recordings and fMRI data. I will give some suggestions below.

2.
> ```Warning: Guessed additional cardiac pulse at time series end for phase estimation 
>   In tapas_physio_get_cardiac_phase (line 61)
>   In tapas_physio_create_retroicor_regressors (line 90)
> ...
> ```
This warning can be happily ignored most of the time. It usually just means that the physiological recordings ended before a final cardiac pulse (after the last slice of the last fMRI volume) was recorded. In this case, the toolbox just guesses one more heartbeat based on the current heart rate, in order to be able to compute the cardiac phases for that last volume as well. However, if warning (1) occurred, and if many more than one pulse was guessed, it could also hint at a synchronization issue, because the cardiac pulses were shifted wrongly relative to the fMRI volume onsets.

> We have noticed that the gradient data is missing (the values are zero) in certain sections of the physio log or sometimes completely missing, and we will get one or both of the above warnings. When [...]
There is no automatic fallback. The toolbox will try to work with the gradients, no matter what their values are. It is normal for the gradient logging to be zero at the beginning of the logfile, but after, say 50s, until the end of the logfile, there should be a repetitive time course logged for all three gradients. If this is not the case, you should resort to the `nominal` method, because the gradient logging will not be reliable. Maybe you could provide a screen shot of the Gradient thresholding figure for a time course you find suspicious?

> Some of these warnings could be eliminated by changing the gradient_log.zero and gradient_log.slice parameters. We have also noticed that changing these parameters will yield completely uncorrelated [...]

I have updated the [FAQ on that question](https://gitlab.ethz.ch/physio/physio-doc/wikis/FAQ#12-philips-i-would-like-to-use-the-gradient-log-for-timing-synchronization-but-how-do-i-set-the-thresholds) to give some more specific pointers on how to set the thresholds and paste them below for your convenience. The fact that you don't get any meaningful regressors from the nominal method is a bit worrysome. By "uncorrelated regressors" you mean that the F-contrasts on the cardiac regressors don't show any significant "activation" in typically affected areas like CSF, brainstem, insula? What settings did you choose for the nominal timing? For Philips, `log_files.align_scan = 'last';` is usually the more robust option.

> 
> The following heuristics might help with the threshold settings in the `sync`
> structure:
> 
> 1. Note that these thresholds have to be set correctly only once for each
> functional sequence, i.e., usually once per study. Even small changes to scan
> geometry (e.g. slice tilt) between subjects shouldn't affect them significantly.
> 2. Setting the thresholds is an iterative procedure. You might start with the
> defaults, probably running into an error or warning (`Warning: Invalid
>   MinPeakHeight. There are no data points greater than MinPeakHeight.` or `Not
>   enough volume/slice scan events found`). Then you inspect the figure output
>   resembling the one above and adjust (usually lower) the thresholds in the
>   order mentioned below.
> 3. There are three time courses in the upper of the three subplots shown in the
> figure. These time courses show the traces of the three gradient directions
> `x`,`y`,`z`. Choose the one as `sync.grad_direction` parameter that has the
> highest peaks and most regular features reflecting slice and/or volume scan
> events.
> 4. `sync.zero` has to be smaller than `sync.slice` and `sync.vol`. It should
> be about 4/5 of the typical peak height in the gradient trace. Note that you can
> set this thresholds (and all other) either in absolute values or relative to the
> maximum peak height. Set a value below 1, if you prefer the latter.
> 5. `sync.slice` should be about 9/10 of the typical peak height of a slice scan
> event.
> 6. `sync.vol`, if you set it, should be larger than `sync.slice`. It should be
> 9/10 of the peak height that stands out at the beginning of a volume, and is
> followed by some dozens of smaller peaks (for the slices) typically. It might
> be, however, that there is no such peak marking the start of a volume. If so,
> you might try `sync.vol_spacing` or leave it empty and rely on the slice
> thresholds exclusively
> 6. `sync.vol_spacing`, if set, should reflect the temporal spacing (in seconds),
> between the end of the previous volume and the start of the next one. The figure
> above gives some idea how to do that based on the bottom subplot that shows peak
> onset differences. If once every few seconds (your `TR`) you find an exposed
> peak, its height will give you the value for `sync.vol_spacing` (maybe reduce
> it by about 5-10ms to allow for timing inaccuracies).

I these suggestions don't do the trick, maybe you can send me one of your logfiles and the corresponding PhysIO batch for a problematic subject (and a good one for comparison).

All the best,
Lars

carolfs, 2019-03-19:
By "uncorrelated regressors" I mean that I calculated the Pearson correlation for the cardiac regressors using the nominal versus the gradient log method and they were sometimes even anti-correlated. About the settings used for the nominal timing, I'm not sure which settings this method uses, but I set the sampling interval, the TR, the number of slices, scans, and dummies, and the onset slice.

I have tried your heuristics for setting the `sync.slice` and the `sync.zero` parameters, but the warnings are still there. The `log_files.align_scan` parameter was already set to `'last'`. I have also tried setting `sync.vol` or `sync.vol_spacing`, but then PhysIO finds fewer volumes in the data, which is wrong.

The gradient logging is zero at the beginning and end of the file, but even though it is zero at the end, I have a log file from a previous study that also has a few zeros at the end and the analysis runs fine on it.

I would like to send you one of my logfiles and the corresponding PhysIO batch. The problematic subject would be from one study, but the good subject would have to be from a different study, but done in the same lab and scanner. How should I send you these files?

mrikasper, 2019-03-19:
Dear Carolina,

the small correlation between nominal and gradient-log-based timing synchronization points to an issue with one of the methods, and taking together all your other observations, it might be that the gradient time course is indeed compromised. One other thing you could try is to evaluate the F-contrasts I mentioned before for models with both regressor sets on your imaging data, and see which one explains more variance.

However, I think it might still be good to send the files and batches. You can send them to me via e-mail (see the PhysIO README) or by one of our preferred secure file sharing options.

All the best,
Lars

mrikasper, 2019-03-21:
Dear Carolina,

thank you for sending the two example datasets. From what I see, both warnings you get are irrelevant for the processing success. 

1. The invalid peakheight warning came from a gradient preprocessing function related to a bug in the Philips logfile recordings that sometime rescale the logged gradient amplitude. Since this rescaling did not happen in your logfiles, there were no such events found, which triggered the warning. I removed that warning now for the next release, since it is confusing.

2. As expected, there was only one heart beat missing after the last scan, which PhysIO added for phase estimation. I replaced the warning now by a simple note and mention the number of pulses added (1 in this case).

So in principle, the warnings were unrelated to the logfile quality. However, I found a reason why nominal and gradient-log based scan timing synchronization might differ.

For comparison, have  a look at a screenshot from the good example (full view and then zoom): The bottom subplot has regular peaks of about the same height (2ms difference = log file sampling rate) that reflect additional spacing between last slice of volume `n` and first slice of volume `n+1`. Hence, you could use a `sync.vol_spacing = 0.072` (s) here to detect all volume starts, which gives sensible timings (TR 2.37 s etc.)

[image]
[image]

In contrast, if you try the same for the "bad" logfile, the following happens: You still see a regularity in spacing as above (e.g., `sync.vol_spacing = 0.068` s should work), however, there is a large outlier peak at about 820 s. If you zoom into it, it seems that the logfile indeed had at least a hiccup there, because the peak height of roughly 90 ms does not match two slice spacings (2*60 = 120 ms) as well. 

[image]
[image]
 
I have seen this before, and it usually points to the Philips logging cache being filled up, so that some data is not written into the logfile. The problem, however, is, that we don't know how much is missing. Some milliseconds, several seconds? From the mechanism of the recordings, one could assume that the logfile *before* this peak is rather reliable, so if you choose `log_files.align_scan = 'first';` and know exactly the number of prepulses (at least 2) and dummies in your sequence, you could count slices and volumes from the beginning of the logfile, and only remain uncertain about the last 80s of your run. It's best to take these `Nprep` and `Ndummies` values from a good dataset and seeing with which setting `last` and `first` alignment becomes equivalent.

If one looks more closely at the high peak, it seems indeed that only one peak for one slice is missing, so if you leave everything as is (no `vol_spacing`, alignment `last`), you might only shift the phases by the duration of one slice (60 ms). This seems acceptable. Of course you could try to manually change the logfile to artificially include this extra slice gradient trace.

So while this explains some differences and the failure of `vol_spacing` for the bad dataset, I still don't see why the nominal method would perform vastly differently from the gradient logging to deliver uncorrelated regressors. One thing to consider, however, is that you take into account that the logfile is a bit longer than the acquisition (see screenshot below). Therefore, in the `nominal` syncing, you would have to set the `log_files.relative_start_acquisition = -2.75 s` to reflect that the scanning ended before the log file. 

[image]

I hope this helps. And finally, looking at the F-contrasts (e.g. via `tapas_physio_report_contrasts()`) is a robust independent way to find out whether the logfile timing makes sense.

All the best,
Lars

carolfs, 2019-03-22:
Dear Lars,

Thank you for reviewing our log files. Your remarks were super helpful and instructive, and now I feel I can understand much better what's going on. We have already contacted the fMRI staff. Let's hope they can increase the buffer size and make this problem go away in future studies.

In the sample log file I sent you, there's a bad peak at the end, but I've checked other files and a bad peak appears from time to time in random places, sometimes at the very beginning. So I don't think changing the alignment to `first` is a solution.

Since in the bad log file I sent you there is only a few miliseconds missing, I think this will not be a problem. For the other files, I was planning on estimating the amount of time that was not recorded by calculating the number of lines between the 0010 mark that signals the beginning of each run and the 0020 mark that signals the end of each run. If for every file only a few miliseconds are missing, we will use the gradient method, and otherwise switch to the ordinal method. Do you think this should work?

Best,
Carolina

mrikasper, 2019-03-27:
Dear Carolina,

I agree that if missing logfile parts appear at random, `log_files.align_scan = 'last';` seems to be the better option. Since your sequence seems to have some `vol_spacing` regularity in general, you could set that option and see how many large peaks you get, and if the overall number of volume starts matches the expected number. If not, or if there are outlier peaks, this will tell indicate a deviation, and the height of these outlier peaks might indicate how much time (or how many slices) you are losing. 

So basically, use the same procedure as I showed above for the "bad" logfile, and zoom in where you find the suspicious peaks.

I hope that helps.

All the best,
Lars

carolfs, 2019-03-28:
Setting the `vol_spacing` parameter always makes PhysIO crash. Visually I can see the outlier peaks. I was just hoping there was an algorithm to calculate the lost time. Thanks again for your assistance!

closed 2019-06-28 by mrikasper