translationalneuromodeling/tapas #53: creating RETROICOR regressors with PhysIO without batch

wiktorolszowy, 2019-03-29:
I have just downloaded the PhysIO package and try to create the RETROICOR regressors without the use of batch or SPM GUI. I am looking at the documentation, checked the examples (/PhysIO/examples) and feel a bit lost. The examples are only batches. I have cardiac and respiration measurements saved as a .mat file and I try to figure out how to create the log files needed by PhysIO (now I am not able to create the RETROICOR regressors even with the batch...). Any help will be appreciated! Sorry for bothering you!

mrikasper, 2019-03-29:
Dear Wiktor,

thank you for trying out the toolbox!

Did you download the example datasets as described in the [Getting Started](https://github.com/translationalneuromodeling/tapas/tree/master/PhysIO#getting-started) section of the README?

Therein, you will find the examples as matlab scripts as well. More info on the examples is in the [wiki](https://gitlab.ethz.ch/physio/physio-doc/wikis/EXAMPLES), and, for PhysIO's documentation in the general, please see the [documentation section](https://github.com/translationalneuromodeling/tapas/tree/master/PhysIO#documentation) in the README.

In brief: The `examples` subfolder in `PhysIO` is more of a template folder for the spm batches, and maybe a misnomer. The downloaded examples via `tapas_download_example_data()` will reside in `tapas/examples/PhysIO` and contain the data to run the examples as well as both spm jobs and plain matlab scripts. 

One more thing: The `matlabbatch` spm job structure is not equivalent to the internal `physio` structure used by the plain matlab scripts that is created by `tapas_physio_new()`. Use the latter only for running PhysIO w/o the batch editor. The technical documentation of each parameter is within that function as well.

Concerning your second question: What is the structure of your `.mat` files? Maybe you can create ascii files (tab separated columns) and then use `Custom` as manufacturer in the `log_files` sub-structure of `physio`.

BTW: Let me know, if I can improve anything concerning documentation to make the access to the examples easier!

I hope that helps.

Thank you and all the best,
Lars

wiktorolszowy, 2019-04-05:
Thanks a lot for your response! I downloaded the examples as suggested in the Getting Started section, but I didn't understand that all the physio.* lines just define the structure/set of parameters used in function tapas_physio_main_create_regressors.m which is run at the very end of the example *_matlab_script.m. Anyway, I am still not sure how to generate the logs, as they differ for the different examples. I work with scans from Siemens Prisma. I do not know how the physiological recordings were initially saved, but now I have .mat files with arrays ~500,000 x 6. The frequency at which the recordings were acquired was 1000Hz. The six labels are "Cardiac", "Respiration", "Scanner trigger", "Pulse Rate", "Control limits", "Alarm". I also have .acq files, but don't even know how to open them... well, I see there is some additional matlab function ACQREAD. I would be extremely grateful for some minimum guidance :)

mrikasper, 2019-04-11:
Dear Wiktor,

this required some detective work. I believe that your physiological data is in fact not from the vendor Siemens devices, but a BioPac module, because the `ACQREAD` function you mention can be found on Matlab file exchange (https://ch.mathworks.com/matlabcentral/fileexchange/16023-acqread) and is designed to read BioPac AcqKnowledge `.acq` files.

This also fits to the description of your output as being `.mat` files with different variables. I have implemented a reader for BioPac `.mat` files (see [Wiki](https://gitlab.ethz.ch/physio/physio-doc/wikis/MANUAL_PART_READIN#mat-file-export-mat)) and its main source of documentation is the file [`tapas_physio_read_physlogfiles_biopac_mat.m`](https://github.com/translationalneuromodeling/tapas/blob/master/PhysIO/code/readin/tapas_physio_read_physlogfiles_biopac_mat.m)

Since I wasn't able to release the data on which this interface was built, you won't find any example for this format, but you can start, e.g., from the `example_biopac_txt_ppu3t_matlab_script.m` and adapt:

`physio.log_files.vendor = 'Biopac_Mat'`

Looking at the code, I believe your column labels might be different from the ones used there and the reader could need some adaptation. If you run into errors, it would probably be easiest to send me one of your example datasets (both `.mat` and `.acq` files as well as the `ACQREAD` function you used).

The current implementation also ignores the scanner trigger column, assuming a `start_sample` variable in the `.mat` file instead. This is also fixed most easily if I can work with some example data.

All the best,
Lars

wiktorolszowy, 2019-06-08:
Once again thanks for support! At the end I found it easiest to export the data from .mat files to .tsv files and to use one of the templates you prepared for BIDS data. If someone was interested, BIDS specification for physiological data is described here:

https://bids-specification.readthedocs.io/en/stable/04-modality-specific-files/06-physiological-and-other-continous-recordings.html

closed 2019-06-08 by wiktorolszowy