translationalneuromodeling/tapas #54: tapas_fitModel breaks with longer sequences

paulsowman, 2019-04-02:
Hi thanks for making these awesome tools available.

I have just started trying the model and following the example in the HGF_demo.

It works as per the demo example using the example binary data. However, if I generate a sequence any longer than about 375 elements it breaks.

e.g.

```
for i=370:400
u = rand(i,1)<0.25;

bopars = tapas_fitModel([],...
u,...
'tapas_hgf_binary_config',...
'tapas_bayes_optimal_binary_config',...
'tapas_quasinewton_optim_config');

fprintf(['failed at length(u) = ',num2str(i+1),])
end
```

Breaks when u reaches ~ 376 elements.

I'm sure I'm missing a simple step?

Many thanks, Paul

BPaton, 2019-04-02:
I have been having the exact same issue with the same max sequence length.

Bryan.

ianthe00, 2019-04-02:
Hi everybody,

We had this problem too but fixed it a few months back based on the HGF instructions (in MATLAB scripts):

% - If you get an error saying that the prior means are in a region where model assumptions are

% violated, lower the prior means of the omegas, starting with the highest level and proceeding downwards.

%

% - Alternatives are lowering the prior means of the kappas, if they are not fixed, or adjusting
% the values of the kappas or omegas, if any of them are fixed.

So, e.g. instead of using as priors

c.ommu = [NaN,  -3 -6];

for 370 trials, 

You can run it all again using (we originally changed first omega3 only but nothing, so proceeded to change also omega2)

c.ommu = [NaN,  -4 -7];

Best,

Ianthe

BPaton, 2019-04-02:
I was not able to get it work as I had to change omega2 so far it would not work as I have sequences around 2000-3000 trials long.

Thanks.

Bryan.

ianthe00, 2019-04-02:
Did you simultaneously change just a bit omega2 and omega3? That worked good for us.
Otherwise try kappa next.

BPaton, 2019-04-02:
I had thought I had tried all possible combinations of parameter changes including kappa but I might go back to it again and check.

Bryan.

paulsowman, 2019-04-03:
Many thanks for pointing that out Ianthe. With 

`c.ommu = [NaN,  -6,  -9];`

Sequences of 2000 elements are ok.

Paul

ianthe00, 2019-04-03:
Glad to read!

closed 2021-09-09 by chmathys