translationalneuromodeling/tapas #56: HUGE_analysis_resting_state

krisbg95, 2019-04-10:
Hi, 
I'm trying to perform the HUGE analysis on DCMs created from resting-state data. However, it comes up with an error when estimating the priors:  
TAPAS:HUGE: Unsupported format. Use cell array of DCM in SPM format as first input.
Reference to non-existent field 'dt'.

Error in tapas_huge_import_spm (line 86)
DcmInfo.hemParam.scaleC = DcmSpm{1}.Y.dt/DcmSpm{1}.U.dt;

Error in tapas_huge_build_prior (line 44)
        DcmInfo = tapas_huge_import_spm(DcmInfo);

Error in Insula_HUGE_analysis (line 2)
priors = tapas_huge_build_prior(DCM_insula_cluster); 

The issue is that the DCM files do not have a U.dt field which from what I understand is the input time resolution. However, as this is  resting-state data there are no exogenous inputs. Thanks very much for your help. 

Best wishes, 
Kristian

TNU-yaoy, 2019-04-13:
Dear Kristian,

currently, the HUGE toolbox does not support resting state DCM. If you want to perform unsupervised generative embedding, you must use a task-based DCM analysis.

Best regards,
Yu

krisbg95, 2019-04-13:
Thank you for your response Yu. I hope the toolbox will be developed to support resting-state data in the near future. 

Best regards, 
Kristian

TNU-yaoy, 2019-04-14:
Dear Kristian,

while it should be (in theory) possible to extend the HUGE framework to include unconstrained cognition, we do not have concrete plans to pursue this line of research, at the moment.

If you would like to perform unsupervised learning on resting-state data, my current recommendation is to try out the classical generative embedding framework as described in Brodersen et al. (2014) NeuroImage. I.e.: extract MAP estimates from your resting-state DCM and feed those into a regular Gaussian mixture model (for example, the one implemented in Matlab's machine learning toolbox).

With very best wishes,
Yu

krisbg95, 2019-04-15:
Thank you for the recommendation Yu. I’ll use that approach instead in that case.

All the best,
Kristian

closed 2019-10-16 by TNU-yaoy