translationalneuromodeling/tapas #58: Interpretation of the output

marinaferalv, 2019-04-30:
Hi,

I want to use the Physio Toolbox of TAPAS with the SCANPHYSLOG files of the acquisitions (with Philips), but I have the following doubts.

First, It is very important for me to understand what is the meaning of the 24 columns in the multiple_regressors.txt file that this toolbox creates. As for the rows, this file has one line for each volume, and I also would like to understand how, based on the information of a volume, the hemodynamic activity can be modulated in different areas of the brain.

Finally, I would like to understand the following issue: in the SCANPHYSLOG file, 500 lines represent a second of the acquisition, and therefore if the TR = 2, a volume equals 1000 lines. I noticed that the total number of lines exceeds the corresponding number, which would be 250,000 (1000 lines x 250 volumes) and by doing some tests I have seen that there are many lines at the beginning that do not correspond to the first volumes. Does PhysIO-Toolbox know how to interpret this correctly? Or should I manually delete lines that are not corresponded with volumes in the SCANPHYSLOG-file before using Physio Toolbox? 

Thank you very much in advance.

Best regards,
Marina Fernández Álvarez, PhD-student.

Laboratory of Functional Neuroscience
Pablo de Olavide University
Ctra. de Utrera, Km.1
41013 - Seville  (Spain)

SaskiaBollmann, 2019-05-04:
Dear Marina,

I hope we can answer some of your questions:

1) what is the meaning of each column of the multiple_regressor file that this toolbox creates. the toolbox creates 24 columns in the multiple_regressors.txt file, and I would like to know what each one is.

Every column represents a regressor for your GLM. RETROICOR creates a Fourier expansion of the measured physiolocal signals that allow the model to explain the impact of cardiac and respiratory activity on the fMRI signals. Using the default settings of the physio toolbox, the first 6 columns (3x2) represent the sine and cosine (2) of the cardiac phase up to the 3rd order of Fourier terms (this is the fourier expansion part). The next 8 columns are respiratory related regressors up to the order of 4 (4x2). The next 4 regressors model the interaction between cardiac and respiratory activity. The last 6 columns are the realignment parameters that were added to the design matrix to allow a user friendly specification of the full model in SPM.

2) This file has one line for each volume, and I would like to understand how, based on the information of a volume, the hemodynamic activity can be modulated in different areas of the brain.

This goes back to how SPM uses a mass univariate model, where the signal in each voxel is modelled with the same design matrix. During inference we can then asses where in the brain specific regressors explain the signal well. The physio regressors are in this case confound regressors that help to reduce the residual variance of your total model fit and therefore help to improve inference for your task regressors.

3) I noticed that the total number of lines exceeds the corresponding number, which would be 250,000 (1000 lines x 250 volumes) and by doing some tests I have seen that there are many lines at the beginning that do not correspond to the first volumes. Does PhysIO-Toolbox know how to interpret this correctly? Or should I manually delete lines that are not volumes?
You do not need to delete lines at the beginning - physio will handle this for you :). However, always make sure to look at the diagnostic plots to check if everything worked out.

All the best
Steffen and Saskia Bollmann

mrikasper, 2019-05-05:
Dear Marina,

thank you for using the toolbox and sorry for the late reply. I just wanted to add that - in general - please always have a look at the [documentation of PhysIO](https://github.com/translationalneuromodeling/tapas/tree/master/PhysIO#documentation) before posting questions.

I mention this not to discourage you asking questions, but to have more time left for us developers to answer more specific questions and improve the toolbox further.

In your case, [FAQ No. 10](https://gitlab.ethz.ch/physio/physio-doc/wikis/FAQ#10-what-is-the-order-of-the-regressor-columns-in-the-multiple-regressors-file) is equivalent to your question (1) (as well as figure 7A in the [paper](http://dx.doi.org/10.1016/j.jneumeth.2016.10.019)) and both question (2) and (3) are addressed within the [paper](http://dx.doi.org/10.1016/j.jneumeth.2016.10.019) as well, in the respective methods sections about noise correction (2.7) and Read-In (2.3).

The only addition to that is that, while SPM is using one GLM for the entire volume, if you want slice-wise regressors for other analysis packages, PhysIO is able to create them and puts them in multiple files, if you specify an index array for the onset_slice, e.g., `scan_timing.sqpar.onset_slice = [1 25 34 40]`. 

Thank you for your understanding!

@SaskiaBollmann Thank you for helping out!

All the best,
Lars

closed 2019-05-05 by mrikasper