translationalneuromodeling/tapas #59: rDCM example for empirical data

pfdamasceno, 2019-06-05:
Hi, I was wondering if you guys could provide a pointer on how to run rDCM given an empirical fMRI. 

This would be similar to the`tapas_rdcm_tutorial.m` tutorial but without the step of reading a pre-defined DCM file.

Do I first need to create a DCM struct? And, if so, is there a helper function that assists in this step?

Thanks.
Pablo

StefanFraessle, 2019-08-23:
Hi Pablo,

My apologies for the terribly delayed response - your question must have somehow skipped my attention. 

In case this is still relevant (and you haven't figured it out by yourself in the meantime anyway): Yes, your intuition is correct that estimating rDCM for empirical data is essentially the same as for synthetic data, as shown in the `tapas_rdcm_tutorial.m` function. Obviously, the synthetic data generation step would fall away - instead, you would directly load the DCM.mat that contains the model and the empirical data.

We do not provide a helper function to generate the DCM structures because rDCM is designed to read in the standard DCM structures from [SPM](https://www.fil.ion.ucl.ac.uk/spm/software/spm12/). Hence, you can use SPM to create your DCMs and then feed it directly into the estimation routine of rDCM: `tapas_rdcm_estimate.m`

I hope this helps. Let me know if you have any other questions or problems.

Cheers
Stefan

zenkavi, 2021-02-22:
Reviving this old thread instead of opening a new issue:

Is there any documentation on the `DCM_LargeScaleSmith_mode1.mat` file used in `tapas_rdcm_tutorial.m`. Looking at the model I understand it's a 50 node network with 25 inputs but I can't trace where/how/why it was generated.

BPaton, 2021-02-22:
The dataset (if I a remembering it correctly) is described here:

https://www.fmrib.ox.ac.uk/datasets/netsim/

including a link to the NeuroImage paper that goes into parameter recovery
for various network discovery algorithms.

Thanks.

Bryan.

StefanFraessle, 2021-02-23:
Hi Zeynep, hi Bryan,

Thanks for the question and for the quick help - great to see the TAPAS community helping each other out :)

Maybe just to add some further details (in the hope that this is helpful): As Bryan already pointed out, the network structure resembles the one introduced by Smith and colleagues in their NeuroImage paper from 2011. The idea of the model is to - very loosely and superficially - account for the small-world architecture of the human brain. That is, the model comprises clusters (or cliques) of strongly connected regions which are then linked via a few long-range connections. This yields a network that combines both high clustering and short path length, which is the defining property of a small-world network.

Based on the proposal by Smith and colleagues, we had used the model in one of our initial publications on rDCM ([here](https://www.sciencedirect.com/science/article/pii/S1053811918304762?via%3Dihub)). This is why the model is the exemplary network architecture used in `tapas_rdcm_tutorial.m` of the rDCM toolbox.

I hope this helps. If you have any further questions, please let me know.

All the best,
Stefan

zenkavi, 2021-02-24:
Thank you both for your prompt and informative responses!