translationalneuromodeling/tapas #61: Signal Processing Toolbox - needed?

wiktorolszowy, 2019-06-08:
Function `tapas/PhysIO/code/utils/tapas_physio_findpeaks.m` employs the `findpeaks` function from the Signal Processing Toolbox in MATLAB. Is it really needed? I see the script has the comment:

```
% Finds local peaks in the data. Wrapper for Matlab's findpeaks, 
% with compatibility alternative, if no signal processing toolbox available
```

but I could not find this alternative, `tapas_physio_findpeaks_compatible.m`, in the archive:

https://www.tnu.ethz.ch/en/software/tapas/archive.html

Did I miss it?

mrikasper, 2019-06-28:
Dear Wiktor,

we decided to make the dependence on the signal processing toolbox more explicit, that's why you only find the compatible old version of findpeaks in older commits of the development branch of tapas.

In the long run, we want to reduce the dependence on Matlab toolboxes by custom implementations of core functions used in PhysIO, but it's a tedious effort. If you have troubles with Matlab licensing, you may try whether you can get the code to run in octave.

For completeness, or as a starting point if you want to improve the PhysIO compatibility, here is a list of Matlab toolbox-dependent function in PhysIO:

| function physio | Matlab toolbox | toolbox function |
| --------------- | -------------- | ---------------- |
| tapas_physio_findpeaks | signal | findpeaks.m |
| tapas_physio_filter_respiratory | signal | butter.m |
| tapas_physio_create_noise_rois | stats | pca |
| tapas_physio_create_scan_timing_from_tics_siemens | stats | knnsearch |

In conclusion, only the signal and stats toolbox are really needed at the moment, and for some functionality (e.g. `imerode` in the PCA time series extraction), you either need SPM or a Matlab toolbox.

All the best,
Lars

closed 2019-06-28 by mrikasper

dlq, 2022-11-25:
Are there any updates on dependence on Matlab toolboxes? Has the situation changed since these comments?

mrikasper, 2022-11-25:
Dear Darcy,

In terms of using PhysIO in Matlab, things are similar, with the exception of the `stats` and `image` toolbox, which are no longer needed for `tapas_physio_create_noise_rois`, thanks to @benoitberanger's work (I still have to merge a pull request there, sorry!). Especially for the filtering operations via the Signal Processing toolbox, I don't see an easy workaround other than manually implementing these filters.

From the next release onwards, you can find the updated list of Matlab toolbox dependencies now also within TAPAS via `[infos, auxillary_toolboxes] = tapas_get_toolbox_infos()` in the `infos.physio.dependencies` field.

Would you mind explaining the background of your question a bit? If you are concerned about using PhysIO without having access to the Matlab Toolbox licenses, you could also consider using our stand-alone compiled version of SPM+PhysIO on [NeuroDesk](https://www.neurodesk.org/tutorials/functional_imaging/physio/).

All the best,
Lars

dlq, 2022-11-25:
We've integrated a PhysIO pipeline into NeuroHub/CBRAIN and we're just a little hamstrung by the MATLAB requirement. 

https://portal.neurohub.ca/available

We've made our own compiled version with some restrictions but will try the NeuroDesk route.