translationalneuromodeling/tapas #62: Uncertainty about observations in HGF

apeirophilia, 2019-06-21:
I have a quick question about your HGF framework,

I'm looking at Eq. 46 in your 2011 paper. In this equation, mu1(k) is updated after receiving sensory input u(k), where there is no uncertainty about the value u(k) (rather, the uncertainty is about mu1(k) after observing u(k)). I'm interested in a case where there is uncertainty about the sensory input itself, so that u(k) is Gaussian with a mean (what I think I observed) and precision (how certain I am about my observation). In this case, how would Eq. 46 be modified to generate the update for mu1(k)? Or, do you have any tips for deriving the update myself?

Again, any help at all would be greatly appreciated. Thank you so much for your time.

chmathys, 2019-06-24:
I think Eq. 47 already gives you what you need if you slightly change the
way you interpret it. Instead of framing the situation as certain u(k),
constant alpha, and uncertain x1(k) (represented by Bernoulli(mu1(k))), you
can see the uncertain input as parameterized by u(k) and alpha _together_,
giving you an input in the form of a distribution. Note that there is no
need for alpha to be constant. Eq. 47 also works with a time-varying
alpha(k).

apeirophilia, 2019-06-24:
Professor Mathys,

Thank you so much for your reply — I really appreciate it. I think I see what you're saying, but I suspect that the type of uncertainty I'm talking about can't really be described in terms of alpha. To illustrate exactly what I mean I've provided an example below. But if I'm just wrong about this let me know and I'll ponder further :)

Suppose I have two aunts, Vicki and Wendy. Every year for my birthday, Vicki gives me gold coins. The number of coins that Vicki gives me varies from year to year, and is normally distributed with mu = 5 and sigma^2 = 2.  Wendy also gives me gold coins every year for my birthday, but hers are normally distributed with mu = 10 and sigma^2 = 2. 

Fast forward to my next birthday. I receive a jar of gold coins in the mail but there's no letter or return address, so I need to infer which aunt sent me the coins.

So far this situation seems to map on to eq. 45:
**u** = the number of coins I received
**x1** = the prior probability that the coins are from Vicki 
**eta_a** = 5 (the expected number of coins if Vicki sent them, i.e., x1 = 1)
**eta_b** = 10 (the expected number of coins if Wendy sent them, i.e., x1 = 0)
**alpha** = 2 (the variance over my two expectations)

So, assuming I know exactly how many coins I received, I have all the information I need to do the inference using eq. 47. But what if _don't_ know exactly how many coins I received? Suppose I don't want to open the jar before I know who sent it, so I only _guessed_ how many coins are inside (e.g., by shaking it). My best guess is that there are 4 coins, but I'm not certain: my overall belief about the jar's contents is described by a gaussian with mu = 4 and sigma^2 = 3. 

In this situation it seems odd to think of my uncertainty about the input as u(k) and alpha together, since alpha = 2 whereas my uncertainty about u(k) is 3. To put a finer point on it, eq. 47 assumes that I'm uncertain about _one_ thing (the # of coins I'd receive from Vicki or Wendy) but I'm actually uncertain about _two_ things: (1) the # of coins I'd receive from Vicki or Wendy, and (2) the # of coins in this particular jar. 

I hope this makes sense. Thank you again for your help!

chmathys, 2019-07-16:
Thanks for this clarification. You should be able to deal with this using a
two-step procedure. First, you could apply standard (static) Bayesian
inference. Using this, and your assumptions, you would calculate the
posterior probability of the coins being from Vicki or Wendy. In a second
step, you could then do inference on the dynamics of auntly coin-giving by
using this posterior as the mu_1 of Eq. 25 in Mathys et al. 2011. This
gives you a way around Eq. 47 where you effectively replace this by the
posterior suited to your specific case.

closed 2021-09-09 by chmathys