translationalneuromodeling/tapas #63: PhysIO: No scan trigger events provided

Da5idN, 2019-07-22:
Dear Lars,

Thank you for providing the toolbox!

I have conducted a resting-state fMRI study using a Siemens Trio scanner with respiratory belt and PPU and now want to use the PhysIO toolbox to regress out physiological noise in SPM. 

As input, I used .puls, .resp and the first DICOM image of my time series (see attached files plus the SPM batch file). I noticed the warning below and also found that the result from the toolbox didn't change when omitting to input the DICOM file. Hence, I doubt the physiological data is properly synchronized with my imaging data when running PhysIO.

No scan trigger events provided
    maxscan (incl. dummies) = 204 
    tmin (1st scan start (1st dummy))=  16.12 s
    tmin (1st scan start (after dummies))=  26.36 s
    tmax = 538.30 s 
    mean TR =   2.56 s

I wonder whether this is something I do wrong during data acquisition. Do you have any ideas how to solve this issue? I have also attached a .ext file from the scanner which should contain the timing information.

Thanks a lot!
David

[Files.zip](https://github.com/translationalneuromodeling/tapas/files/3416261/Files.zip)

mrikasper, 2019-08-20:
Dear David,

my apologies for the late reply. I now had the time to run your example data and the good news first: I think the data acquisition is fine.

However, in the batch file I noticed that you set 

`physio.log_files.align_scan = 'last'`

But, as you mention, the dicom file you provide is the *first* of the time series. Therefore, you should set 

`physio.log_files.align_scan = 'first'`

If you do that, the estimated delay of 11.3 s between start of logging and start of first scan (that you will see in on the time axis of the *Read-In: Raw Siemens physlog data plot* below, t=0 being the start of the scan in the dicom file) seems much more realistic than the -530 s when setting `'last'`. And the regressors also change. I would always recommend to run an F-test on the regressors in the GLM to see whether they explain significant variance.

The warning *no scan trigger events provided* is not directly related to any of this and rather refers to the fact that there were no triggers logged in the `.pulse/.resp/.IMA` files you provided. That is fine, because you rely on the time stamps from the scanner here rather than an external signal.

The use of `.ext` files is not supported at the moment, but I put it as a feature request.

If you want to know more about timing synchronization in SIEMENS logfiles, have a look at the [read-in wiki page of PhysIO](https://gitlab.ethz.ch/physio/physio-doc/wikis/MANUAL_PART_READIN#siemens)

All the best,
Lars

[image]

closed 2020-02-03 by mrikasper

mrikasper, 2020-02-03:
Dear David, 

after resolving a similar bug in issue #55 (see my recent [post](https://github.com/translationalneuromodeling/tapas/issues/55#issuecomment-581651096) in that thread), I came to the conclusion that your reported behavior might be related to the same problem.

Have a look whether **v7.2.2** of PhysIO resolves your issue that is included in the current  [development branch](https://github.com/translationalneuromodeling/tapas/tree/development) of TAPAS.

My apologies for not spotting this behavior immediately.

All the best,
Lars