translationalneuromodeling/tapas #76: cardiac data modelling - missing sections

JJCbr, 2019-11-11:
Dear PhysiO,

Thank you for providing such a useful and intuitive toolbox for the regression of physiological data from fMRI. I am currently modelling cardiac and respiratory signals recorded on a Siemans Terra with a CMMR MB sequence and extracting physiological data from the .dcm to .log using the CMMR's extractCMMRPhysio function.

This seems to work well but on inspecting the output from PhysiO the cardian modelling appears to cut sections out of the data (please see attached images), and HRF modelling does not look similar to that shown in the PhysiO paper. This is also  mirrored in the RETROICOR regressors which seem to model the cut-out data.

I'm not super experienced with regression of physiological signals and am a little unsure on where to start, as such any advice would be much appreciated.

Many Thanks

[image]
[image]
[image]
[image]
[image]

JJCbr, 2019-12-12:
Changing the option 'Initial Detection of Heartbeats' from 'load_from_logfile' to 'auto_matched' seems to have helped. though not quite sure why. Nonetheless the regressors and heart rate look much more sensible now. Would anyone be able to have a quick glance to confirm these look reasonable> (botton two panesl). Thanks for your help.

mrikasper, 2020-01-29:
Dear JJCbr,

my apologies for the late reply. I am very happy you like the toolbox user experience!

It seems you already found the solution to your initial problem: the "cutout" appearance of cardiac signal in the first plots stems from a detection failure of the heart beat pulses on the SIEMENS system. Consequently, the regressors were all zero for these phases. `auto_matched` detection is our default recommendation to circumvent such issues, as I also noted in a recent [thread](https://github.com/translationalneuromodeling/tapas/issues/79#issuecomment-579994942). The reason why it outperforms prospective detection by the system is explained in our [paper](https://doi.org/10.1016/j.jneumeth.2016.10.019) (Figs. 4/5).

Concerning your second question whether the regressors look reasonable - this is really hard to tell from these figures, but there are two other things you should check:

1. What does the diagnostics figure look like that I mentioned in this [thread](https://github.com/translationalneuromodeling/tapas/issues/79#issuecomment-579994942)? Is the green line (peak-to-peak interval time course) rather smooth without outliers?
2. Compute an F-contrast in a GLM that contains the physiological regressors (e.g., the first 6 columns of the multiple regressors for the cardiac RETROICOR effects). The spatial noise distribution should look similar to [Fig. 9](https://ars.els-cdn.com/content/image/1-s2.0-S016502701630259X-gr9_lrg.jpg) of the PhysIO paper.

I hope this helps!

All the best,
Lars

closed 2022-02-20 by mrikasper