translationalneuromodeling/tapas #79: PhysIO_peak detection

Alego13, 2020-01-28:
Dear experts, 
I am working on a project with fMRI Siemens and recording physio data with the pulse oximeter of the scanner.
I would like to use your nice toolbox for the analysis of physio data but I have some questions:

- For my project I need to extract the number of peaks occurred during a task for every subject, so is there a way to have as output a mat. file with the number of peak detected and to save it for further analysis?

- It's important that I can check if peaks are detected correcly, so is there an automatic peak rejection option BUT ALSO an option to change manually if a peak is rejected or not?

Thank you in advance for your availability

mrikasper, 2020-01-29:
Dear Alego,

thank you for considering our toolbox!

>* For my project I need to extract the number of peaks occurred during a task for every subject, so  is there a way to have as output a mat. file with the number of peak detected and to save it for  further analysis?

If you specify an output `.mat` file in parameter `physio.model.output_physio`, a lot of extra data is saved in a structured variable `physio`, computed during the run of the main function.

For example, `physio.ons_secs.cpulse` will contain all indices of cardiac pulse events. Therefore, `numel(cpulse)` should give you the number of heartbeats during the task.

More information can be found in the comments in `tapas_physio_new` or in the [FAQ](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/FAQ#9-i-want-to-access-subjects-physiological-measures-eg-heart-rate-or-respiratory-volume-per-time-before-they-enter-the-regressors-where-can-i-do-that).

>* It's important that I can check if peaks are detected correcly, so is there an automatic peak rejection option BUT ALSO an option to change manually if a peak is rejected or not?

We spent a lot of care designing the peak detection algorithm, see the section in our [paper, Fig. 4 and 5](https://doi.org/10.1016/j.jneumeth.2016.10.019). So in general, you can be quite confident about the quality of the detection, especially with the pulse oximeter, which is more robust than the ECG.

If you want to check, there is a diagnostics figure showing you peak-to-peak interval time courses. The green line there should not have to many spikes and averages around 1s (60bpm)

[image]

If you want to manually change outliers in that time course, we provide the option, as explained in [Fig. 3](https://ars.els-cdn.com/content/image/1-s2.0-S016502701630259X-gr3_lrg.jpg) of the paper. Just set `physio.preproc.cardiac.posthoc_cpulse_select = 'manual'`.

Note that all mentioned parameters are from the matlab input structure. They have slightly different names in the SPM Batch Editor GUI.

I hope that helps.

All the best,
Lars

Alego13, 2020-02-15:
Dear Dr. Kasper,

Thank you very much for your exhaustive and precise answer, I am now definitely using your toolbox and for the moment everything is very clear!

Best,
Alessio

closed 2020-02-18 by Alego13