translationalneuromodeling/tapas #81: Asking for help about the wrong used of  "tapas_hgf_binary (line 225) "

liumeng87, 2020-02-05:
Dear Professor,

Thank you very much for your HGF toolbox! Recently, I have some difficulty in fitting time series models using this toolbox, and I hope to consult you.

Specifically, my time series contain 640 binary numbers. When I fitting the data, the error message of MATLAB is “wrong used tapas_hgf_binary (line 225) Variational approximation invalid. Parameters are in a region where model assumptions are violated”. I found that there is a parameter (jumpTol=16) in the tapas_hgf_binary.mat. I do not know if I can increase this parameter? If I have 640 binary numbers for fitting, which value is appropriate for this parameter?

Thank you very much!

chmathys, 2020-05-20:
Dear liumeng87,

Sorry for the slow response. Your question came just as I was moving to a new university in a new country and just before the virus struck.

At the top of the file tapas_hgf_binary_config.m you can find additional documentation on that particular model. I’ll copy the relevant tips here:

% - If you get an error saying that the prior means are in a region where model assumptions are
%   violated, lower the prior means of the omegas, starting with the highest level and proceeding
%   downwards.
%
% - Alternatives are lowering the prior means of the kappas, if they are not fixed, or adjusting
%   the values of the kappas or omegas, if any of them are fixed.

I wouldn’t play around with the jumpTol setting unless you have a principled reason for that. Normally, you will be able to fit your time series using the tips above.

Best wishes,
Christoph

liumeng87, 2020-05-20:
Thank you very much！
On 05/20/2020 17:44, chmathys wrote:

Dear liumeng87,

Sorry for the slow response. Your question came just as I was moving to a new university in a new country and just before the virus struck.

At the top of the file tapas_hgf_binary_config.m you can find additional documentation on that particular model. I’ll copy the relevant tips here:

% - If you get an error saying that the prior means are in a region where model assumptions are
%   violated, lower the prior means of the omegas, starting with the highest level and proceeding
%   downwards.
%
% - Alternatives are lowering the prior means of the kappas, if they are not fixed, or adjusting
%   the values of the kappas or omegas, if any of them are fixed.

I wouldn’t play around with the jumpTol setting unless you have a principled reason for that. Normally, you will be able to fit your time series using the tips above.

Best wishes,
Christoph

q610898680, 2021-01-08:
I've got the same problem and lower the value of omegas does not help. How can i solve it?

chmathys, 2021-01-11:
Dear q610898680,

Would you care to be a bit more specific?

Thanks,
Christoph

q610898680, 2021-01-13:
Dear Christoph,

Thank you for responding. I have time series of 400 binary numbers. An error message came up when I fit the data using  
bopars = tapas_fitModel([],...
                        u,...
                        'tapas_hgf_binary_config',...
                        'tapas_bayes_optimal_binary_config',...
                        'tapas_quasinewton_optim_config');

"Error using tapas_hgf_binary (line 215)
Variational approximation invalid. Parameters are in a region where model assumptions are violated."

I tried to adjust the values of the kappas and omegas, but the problem was not solved.

Best wishes

q610898680, 2021-01-13:
Is there a way that I can try a number of Parameter values in a batch method until I find the value that does not pose an error? say omegas=-3:-1:-100

chmathys, 2021-01-18:
Dear q610898680,

You could try

• Estimating fewer parameters, ie fixing more of them by setting their prior variance to 0
• Then find prior settings for the estimated parameters that work
• Or use the enhanced HGF: ehgf_binary instead of hgf_binary

Best wishes,
Christoph

chmathys, 2021-01-18:
> Is there a way that I can try a number of Parameter values in a batch method until I find the value that does not pose an error? say omegas=-3👎-100

You can write a script that loops through all points on a grid and do that. However, omega values of less than about 8 or 10 don’t make sense in binary models because the inferred belief trajectory will just be flat.

Best wishes,
Christoph

chmathys, 2021-01-18:
That should be “-8 or -10” in the message below, I forgot the minus signs.

Christoph

closed 2021-09-09 by chmathys