translationalneuromodeling/tapas #86: rDCM: strange pattern in estimated matrix

davidevaleriani, 2020-03-16:
Hi,

I have used rDCM without sparsity constraints to estimate the posterior's connectivity matrix (Ep.A) for some finger tapping data we have. However, the resulting matrix looks very strange, with a sort of pattern represented by vertical lines (see below). I can't figure out what is wrong with the settings, has this ever happened to you?

Thanks,
Davide

[image]

StefanFraessle, 2020-03-17:
Hi Davide,

I agree that this looks very odd. In principle, such a strip-like pattern could occur due to the mean-field approximation across regions, which essentially means that all regions are estimated independently of each other. However, having said this, something seems to be off here. 

From the figure, I assume that you estimated a fully connected matrix (i.e., all reciprocal connections were assumed to be present)? This might be overly ambitious, given the number of data points that you have. In other words, I would suggest to reduce the complexity of the inference by: (i) constraining the A-matrix based on, for instance, a structural connectome (maybe you have DTI data available or can use a parcellation scheme that comes with a default structural connectome), or (ii) using sparsity constraints to prune the network to a more reasonable number of parameters given the number of data points you have. Would that sound sensible?

In terms of why exactly you find these odd results, it is very difficult to judge from here. You could send me one exemplary file if the above suggestions do not result in a more sensible pattern. My guess is that the inversion of the covariance matrix might be ill-defined - but I could be completely wrong.

Hope this helps for a start.

With very best wishes,
Stefan

davidevaleriani, 2020-03-17:
Hi Stefan,

Thanks for the quick reply. 

I have estimated a fully connected matrix, yes, as I don't have a structural connectome available for this parcellation. 
(i) I have tried to use only half or 1/4 of the ROIs in the parcellation scheme, which intrinsically reduce the number of parameters, but that yields the same results (see images below, different subject from above). 
(ii) I have also tried to use sparsity constraints, but since these are 3T data, I end up either having completely sparse matrices (often all zeros) or not pruning anything at all for larger p0 (as addressed in a previous [question](https://github.com/translationalneuromodeling/tapas/issues/82)).

212 ROIs
[image]

106 ROIs
[image]

53 ROIs
[image]

I can't understand why rDCM is behaving like that, as I am also analyzing other task's data, with similar number of data points and from the same subjects, and the matrix looks OK.

What file do you need me to send? The DCM output and/or the data/code?

Many thanks,
Davide

StefanFraessle, 2020-03-17:
Hi Davide,

Is this just for one participant, or do you see this behavior for every participant for the finger tapping task? 

I would need the DCM structure that enters the "tapas_rdcm_estimate" main routine. Then, I can try to reproduce the behavior and check what is causing these results.

With very best wishes,
Stefan

davidevaleriani, 2020-03-17:
Hi Stefan,

This is happening with almost all 35 participants I am using. I have just sent you a few DCM structures for taking a look.

All the best,
Davide

PhDqiwang, 2023-02-28:
I also encountered a similar situation. The EC estimates of all subjects have this strong pattern. Is there a problem with the input of the model? Thank you.