translationalneuromodeling/tapas #87: PhysIO : No cardiac R-peak (heartbeat) events provided

juufsoares, 2020-03-17:
Hello, 

I am acquiring physiological data on a Siemens Prisma with PPU and respiratory cushion, using a multiband sequence and I am exporting *_Info.log, *_RESP.log, and _PULS.log files. Attached is the batch with the defined parameters and the log files. However I am getting the following error: 

No cardiac R-peak (heartbeat) events provided
17-Mar-2020 22:46:36 - Failed  'TAPAS PhysIO Toolbox'
Array indices must be positive integers or logical values.
In file "C:\Users\juliasoares\Documents\MATLAB\spm12\toolbox\PhysIO\sync\tapas_physio_get_onsets_from_locs.m" (???), function "tapas_physio_get_onsets_from_locs" at line 103.
In file "C:\Users\juliasoares\Documents\MATLAB\spm12\toolbox\PhysIO\sync\tapas_physio_create_scan_timing.m" (???), function "tapas_physio_create_scan_timing" at line 117.
In file "C:\Users\juliasoares\Documents\MATLAB\spm12\toolbox\PhysIO\tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 149.
In file "C:\Users\juliasoares\Documents\MATLAB\spm12\toolbox\PhysIO\tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1596.

Could you provide some help on this?
[Physio_20200113_190112_b9331b77-938d-447e-ad8c-40f87cfbbfe5_Info.log](https://github.com/translationalneuromodeling/tapas/files/4346221/Physio_20200113_190112_b9331b77-938d-447e-ad8c-40f87cfbbfe5_Info.log)
[Physio_20200113_190112_b9331b77-938d-447e-ad8c-40f87cfbbfe5_PULS.log](https://github.com/translationalneuromodeling/tapas/files/4346222/Physio_20200113_190112_b9331b77-938d-447e-ad8c-40f87cfbbfe5_PULS.log)
[Physio_20200113_190112_b9331b77-938d-447e-ad8c-40f87cfbbfe5_RESP.log](https://github.com/translationalneuromodeling/tapas/files/4346223/Physio_20200113_190112_b9331b77-938d-447e-ad8c-40f87cfbbfe5_RESP.log)
[cnt-01-v01_physio_batch.zip](https://github.com/translationalneuromodeling/tapas/files/4346228/cnt-01-v01_physio_batch.zip)

mrikasper, 2020-04-07:
Dear Júlia,

thank you for your patience and for using the PhysIO Toolbox!

I had a look at your logfile and it seems untypically short (only about 28s, of which 18s of data are recorded). See the attached figure:

[image]

Since the logfile is so short, PhysIO cannot find the 240 volume events you specified in the scan timing parameters.

Could it be that this logfile does not belong to the fMRI scan or that logging was stopped prematurely?

All the best,
Lars

juufsoares, 2020-04-09:
Dear Lars, 

Thank you for your enlightening answer. In fact, I realized after you show me the figure that I wasn't usng the correct logfile. 

Best regards,

Júlia Soares

closed 2020-04-19 by mrikasper