translationalneuromodeling/tapas #88: Are HRV and PRV provided?

Alego13, 2020-03-21:
Dear experts,

I am already using your excellent toolbox for my project, in which I recorded physio data with the pulse oximeter of a fMRI Siemens scanner (vendor Siemens_Tics).

I have noticed that your toolbox can offer noise modeling of HRV and that is great, but at the moment I need to know the HRV value for each subject, in order to perform later some analyses.

I have explored the outputs and the guide but I cannot find that value, is it possible to implement it as a new output ?

In addition,  I discovered that HRV is properly calculated by ECG data (RR distance), while with PPG data is better called pulse rate variability (PRV). So my question is: is it possible to calculare PRV in your toolbox and obtain the value as a mat.file within the physio.mat output? 

Thank you in advance for your great availability!

Alessio

mrikasper, 2020-03-30:
Dear Alessio,

thank you for using the PhysIO Toolbox.

1. Regarding your first question, HRV and many other physiological parameters are available in the `physio.ons_secs` structure saved by PhysIO in the `physio.mat` output file. See the [FAQ](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/FAQ#9-i-want-to-access-subjects-physiological-measures-eg-heart-rate-or-respiratory-volume-per-time-before-they-enter-the-regressors-where-can-i-do-that) or comments in `tapas_physio_new` for details.

2. I am not sure I understand the difference between HRV and PRV in terms of calculation, could you clarify, please? PhysIO happily calculates HRV from pulse oximetry data as well (and you can use the same variable as in (1)) for that, if you switch on the HRV regressor calculation.

I hope that helps.

All the best and best of health,
Lars

Alego13, 2020-04-05:
Dear Lars,

Thank you for the detailed answer!

I have checked again the documentations but unfortunately I haven't found what I need. 

Basically I inserted the HRV regressor option as showed in the picture
[image]
and I noticed that in the output now the new HRV regressor is included and that in physio.ons_secs.hr the estimated heart rate at each scan is performed, and that is great.

However, I would like to calculate the HRV for each subject (one value for subject) by time-domain methods such as the 'root mean square of successive RR interval differences' (RMSSD) which is based on the ibi (interbeat interval or RR interval). Is there an option to include this calculation?

Regarding my second question about the difference between HRV and PRV, I have realised that HRV is also a valid measure with PPG data, so I apologize for this question. (Theorically HRV is based on RR interval while PRV on pulse interval, so I supposed there was a difference between the two measures, but it is not so true).

Thank you for the great work on this toolbox, I highly appreciated it.

All the best and I hope everything is fine from you

Alessio

mrikasper, 2020-04-07:
Dear Alessio,

I understand now that you would like to calculate one value per subject (or run), summarizing heart rate variability throughout.

The onset times (in seconds) of all heartbeat events within the scan window (volume 1 to volume N) is saved in `physio.ons_secs.cpulse`. Therefore, you can easily calculate the 'root mean square of successive RR interval differences' (RMSSD) in Matlab as follows:

```
RMSSD = rms(diff(physio.ons_secs.cpulse))
```

or, if you don't use the signal processing toolbox

```
RMSSD = sqrt(mean(diff(physio.ons_secs.cpulse).^2))
```

I hope this helps.

All the best,
Lars

closed 2020-05-11 by mrikasper