translationalneuromodeling/tapas #89: PhysIO: save batch process figure outputs (jpegs) to file without opening them in Matlab

jsheunis, 2020-03-26:
Hi there

Thanks for the great and well documented toolbox :) I'm using PhysIO in Matlab scripts to generate RETROICOR, HRV and RVT regressors from cardiac and respiratory traces in a BIDS tsv file.

I would like to create and save the output figures to file, however I do not want to have them plotted in Matlab while the script is running. I currently set the following values in the batch script:

```
matlabbatch{1}.spm.tools.physio.verbose.level = 2;
matlabbatch{1}.spm.tools.physio.verbose.fig_output_file = 'PhysIO_output_level2.jpg';
matlabbatch{1}.spm.tools.physio.verbose.use_tabs = false;
```

I checked the different options for `verbose.level` but couldn't see a way to do what I want to do. Is there a setting that I can use to save the figures to file, without plotting them in a Matlab window? Just checking before I start digging into the code.

mrikasper, 2020-03-30:
Dear Stephan,

thank you for using the PhysIO Toolbox and for reaching out. There is two suggestions I have:

1. There is a function `tapas_physio_review` that can recreate some of the relevant plots from `physio.mat` information. So you could run with `verbose.level=0` and then create the plots later on. See the [FAQ](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/FAQ#7-i-am-running-the-toolbox-for-a-lot-of-subjects-on-a-remote-server-without-graphics-can-i-somehow-reproduce-the-output-figures-relevant-to-assess-the-data-quality) for details.
2. People in my former lab have used the toolbox on a cluster with Matlab `-nodesktop` option (I think) and job submission systems (SGE, LSF), which still allowed to create and save the figures. NB: The next TAPAS release (probably within the next few weeks) will fix a small bug for such operation, I could push it to the development branch, if you are interested.

If you want to contribute and improve code, I see two pathways:

1. You could augment `tapas_physio_review` to re-plot more of the figures, in particular the ones you might be interested (HRV...)
2. If it is about the visibility of all these figures, you could alter `tapas_physio_get_default_fig_params`, which is called as a constructor for every figure created by physio. It normally sets font sizes etc., but you could also include figure visibility there. Ideally, this would be accompanied by a new parameter in the `verbose` structure.
3. If it’s not visibility, but the amount of figures in the Matlab workspace, and you want to close them immediately after creation (and saving), you could think about a “teardown”-function analogous to how `tapas_physio_get_default_fig_params` is a figure constructor. This would have to be called after each function-call in `tapas_physio_main_create_regressors` that outputs new figures. As all figure handles are stored in `verbose.fig_handles`, you could selectively save and close created figures during execution. Again, would be nice to have a `verbose.do_keep_figures_open` parameter to control this.

I hope this helps and I understood your question correctly.

All the best and best of health,
Lars

jsheunis, 2020-03-30:
Hi Lars.

Thanks for the detailed answer! I think I'll start off with `tapas_physio_review`and see if that solves my current challenge. Down the line, I think a `visible` 'on'/'off' setting on the figure handle, as well as a way to close figures that are open and invisible would be useful. I'll respond here if I have some time available to update the code and send a pull request.

jsheunis, 2020-04-14:
Hi Lars

I've adapted some of the code to suit my particular purposes. The adapted code is available on the [`fig-options`](https://github.com/jsheunis/tapas/tree/fig-options) branch of my fork. Here I'll explain what the changes entail and what we could do next:

### The need
I basically want to be able to control the behaviour of figure outputs when running `tapas_physio_main_create_regressors` and/or `tapas_physio_review`. To suit my pipeline and to save time, I opted for your suggestion of setting `verbose.level = 0` when running `tapas_physio_main_create_regressors` and then running `tapas_physio_review` at the end (with `verbose.level = 2`) to generate and save the figures.

The existing figure display behaviour (1) and the behaviour I would like to see (2), is as follows:
1. generate figures, display them, do not save them (current default)
2. generate figures, *do not* display them, save them, close the (hidden) figures
3. a user-specified combination of the above could also be useful

### The updates

To achieve behaviours 2 and 3 above, I introduced three new `physio.verbose` flags:

- `show_figs`: If true, all created figures will be visible (`figure('visible', 'on')`). If false, sets figure visibility to 'off', which leaves the possibility of saving the figure to file without displaying it. (`default = true`)
- `save_figs`: If true, all created figures will be saved using specified `fig_output_file` as filename (`default = false`)
- `close_figs`: If true, will close all open figs after saving them. Only used if `save_figs` is true. (`default = false`)

I have updated `tapas_physio_get_default_fig_params` to include a new function parameter `visibility` that makes the figure visibile or not depending on the  value of `verbose.show_figs`.

In order to get the above to work as expected, I also had to send the `verbose` structure to some functions that didn't previously include it as a parameter, e.g. `tapas_physio_plot_cropped_phys_to_acqwindow` and others. This should be backwards compatible since I made use of the existing `if nargin < x`-type logic that you used. I also made the required changes inside the subfunctions that are called from within `tapas_physio_review`.

For my pipeline, after running `tapas_physio_main_create_regressors`, I set the new verbose flag values before running `tapas_physio_review`.

**Important note:** I didn't make any changes to the plotting functions that are called (directly or indirectly) from `tapas_physio_main_create_regressors` unless they happen to also be called from `tapas_physio_review`. So setting the new flag values will not result in expected (new) figure displaying/saving behaviour when running `tapas_physio_main_create_regressors`, this functionality will remain unchanged (except where function use overlaps with `tapas_physio_review`). I think updating the remaining plotting functions to mirror the new behaviour could be a sensible next step.

### Next steps
Further updates that I think make sense:
- updating `tapas_physio_main_create_regressors` with similar changes to figure displaying/saving/closing
- including all missing plotting options in `tapas_physio_review`, so as to to create equivalent output to `tapas_physio_main_create_regressors`
- possibly restructuring plotting functions based on their being used from multiple pipelines

Could you let me know what you think, both of my specific update and the possible next steps? If you think my current changes are sensible and could be a self-sufficient update, I can send a pull request. If not, I can see if i can make the other changes as well, but I can't promise a quick turn-over.

mrikasper, 2020-04-15:
Dear Stephan,

thank you very much for this very impressive contribution. I do believe it makes sense to submit a pull request at this point. I quickly looked at your commits and noticed the following:

1. There are some hidden files (xml?) added that probably should not go into it.
2. Maybe it's better to have `verbose` (instead of `visibility`) as input to `tapas_physio_get_default_figure_params`, because this saves some switch statements in the rest of the code and moves them into that function.
3. Just food for thought and discussion to get your input: Indeed, it would be great to extend the functionality to all plotting functions. Probably the easiest way forward is to have a `tapas_physio_teardown_figure` function in each file that also calls `tapas_physio_get_default_figure_params`
    - This would be similar to your close figure function, but instead determine the saving and closing right away within each function. What do you think? Or is it OK to only do it once for every run of the main PhysIO function?
4. In order to acknowledge your contribution, but also to clarify licensing issues (I explain this [here](https://github.com/translationalneuromodeling/tapas/issues/11#issuecomment-435213894)), please include the `Contributor-License-Agreement.md` with your name into the pull request.

If we can finalize the pull request within the next two weeks, it could already be shipped with the next TAPAS release.

Thank you very much for your work again and have a great week,
Lars

jsheunis, 2020-04-15:
Hi Lars

Thanks for the feedback. 

> 1. There are some hidden files (xml?) added that probably should not go into it.

Sorry these are files created by my development environment, and I forgot to update the `.gitignore` file. Will do that.

> 2. Maybe it's better to have verbose (instead of visibility) as input to `tapas_physio_get_default_figure_params`, because this saves some switch statements in the rest of the code and moves them into that function.

That makes sense. I wasn't sure at first if you'd prefer to keep the `tapas_physio_get_default_figure_params` function general enough so as to not be dependent on receiving a verbose structure. But if that's fine with you, it will definitely make the code cleaner in many ways. (Coincidentally, I'm building lots of pipelines and helper functions lately and this seems to *always* take a lot of time, i.e. deciding whether to make a function general enough to be called from multiple sources, or to make it dependent on receiving specific inputs).

> 3. Just food for thought and discussion to get your input: Indeed, it would be great to extend the functionality to all plotting functions. Probably the easiest way forward is to have a [...]

I think your way would be better, especially if there will be scenarios where the plotting functions will be called independently of the main PhysIO scripts (`tapas_physio_main_create_regressors` and `tapas_physio_review`). In such cases one might still want to have control over the behaviour of the figures.

> 4. In order to acknowledge your contribution, but also to clarify licensing issues (I explain this here), please include the Contributor-License-Agreement.md with your name into the pull request.

Thanks. I had a look at the agreement and it looks like the University of Zurich wants to be able to use the code for whatever purposes suit them, which is fine with me. I have two questions that I am uncertain about, though:
- does this agreement give them the right to, in future, make the currently open-sourced code fully closed source?
- With regards to the item on patents:
   > With respect to any patents Contributor owns, or that Contributor can license without payment to any third party, Contributor hereby grants to University of Zurich a perpetual, irrevocable, non-exclusive, worldwide, no-charge, royalty-free license to...

  This is confusing since it mentions "any patents" and not purely patents that have sway over the code that is being contributed to the project. Is that the intention (i.e. any patents) or am I over-interpreting that?

mrikasper, 2020-04-16:
Dear Stephan,

thank you for your fast reply. I only comment on the Contributor License Agreement (CLA) here, because the other points seem resolved to me now.

As a disclaimer, I have no legal training, so my understanding of the CLA's implications is probably limited. However, I was involved from the TAPAS developers' side when we invoked the CLA together with the tech transfer office of University of Zurich (unitectra), so I hope my information is useful and accurate:

Basically, the first and foremost purpose of TAPAS is to provide useful free tools for the scientific and clinical research community. So with respect to the licensing for research purposes, I am rather certain that we will stick with the GPL. I think you can even interpret the last sentence of point 4 of the CLA that way:

> Any contribution University of Zurich makes available under any license will also be made available under a suitable FSF (Free Software Foundation) or OSI (Open Source Initiative) approved license.

Also, any TAPAS code that is already on GitHub now under this license is in the public domain and anybody can use it following the GPL principles. I don't think one can retrospectively revoke that.

The reason why the CLA might be useful in the future is that it makes the ownership of the code clear (University of Zurich). And the owner can in principle re-license the code for different purposes, for example commercialization. So, should any of our tools become clinically useful, then the only way to make it available to a large patient population would be to work with a company, who most likely does not want to publish all their product code because of using our GPL code. This scenario might never happen, but without the CLA, one would probably have to involve the institutions of all contributors, in case it did.

With respect to the patent question, I believe that the whole document has to be read relative to the first sentence of the CLA:

> ...relating to the Contribution (as defined below) that Contributor makes to the following Project:
> 
> TAPAS - Translational Algorithms for Psychiatry-Advancing Science
> (hereinafter referred to as "Project")

So - no, I don't think it relates to other work you are doing or will be doing unrelated to TAPAS.

I hope this helps - and I understand that these legal documents are a bit overwhelming and raise suspicion. Please let me know if you have any more questions or concerns!

All the best,
Lars

jsheunis, 2020-04-17:
Thanks for the feedback.

I have now updated the code to send the `verbose` structure to `tapas_physio_get_default_figure_params ` everywhere it is called from within `tapas_physio_review `.

I thought again about the teardown function and whether it makes sense to use it inside each function that plots figures or at the end of a script. I realised that the function to save the plotted figures is currently only located at the end of the `tapas_physio_review ` script. So if we move the teardown function to each sub-function, we will have to move the `tapas_physio_print_figs_to_file` function to each sub-function as well. This is of course fine, and perhaps the more sensible route in the long term, but I just wanted to get your input on that first as well. What are your high-level thoughts about how this should be done in the long term?

In the mean time, I will test the all the recent changes more thoroughly tonight and will send a pull-request after that.

mrikasper, 2020-05-11:
Dear Stephan,

I have now checked the integrations tests for all example datasets, and they run through fine. Thank you so much again for your input. There is still a couple of follow-up issues related to this work, which include:

1. Integration tests for all PhysIO visualization (in particular `tapas_physio_review`)
2. Expanding `tapas_physio_review` to include all desired visualization.
3. More fine-grained control which figures to show/close and when (`teardown`-functionality), particularly within main script.

Some of them you already mentioned. Let me know, if I forgot anything. I will compile proper issues for this at a later point.

All the best, and see you at OHBM, I guess :-)

closed 2020-05-11 by mrikasper

jsheunis, 2020-05-11:
Hey Lars. Thanks for the update, was nice to contribute :) You can ping me when you create those issues, then I'll see if I have time to dig in. Yeah, I'll see you at OHBM. I'll be hanging out in the open science room mostly, so hope to see you there!