translationalneuromodeling/tapas #92: rDCM : rs-fMRI data

MatteoCanini, 2020-04-20:
Hi there,
first we would like to thank you for the release of TAPAS and for sharing the rDCM code. 

We would like to address you 2 main questions concerning the application of rDCM to resting state data.

1.  With respect to your 2018 work:

“…in its current form, rDCM can be applied to the “resting state” (i.e., unconstrained cognition in the absence of external perturbations) by “switching off” driving inputs (i.e., setting pi0 to zero). ….

….However, the model does not explicitly account for endogenous fluctuations in neuronal activity; for example, it has no concept of stochastic “innovations” or similar ways how noise can drive activity intrinsically. We anticipate that including a mechanism to account for endogenous fluctuations will constitute an important future step to further increase the explanatory power of rDCM for “resting state” fMRI data…..

We would like to ask you if you were already able to include such mechanism in the rDCM code.
If not, have you tested the stability and reliability of the A matrix on rsfMRI data with rDCM in its current form?

2.    Following we provide details on our steps and procedure to apply rDCM to our rsfMRI dataset and the issues we have encountered so far:

-A DCM model was specified using SPM12 DCM (CSD). By specifying a CSD DCM, no u inputs are stored neither a dt variable as in the “timeseries” case.

-The DCM model was passed to the rDCM estimation script. At this step we set the p0 = 0, however the code would error out as unable to find a dt value.

-We were able to overcome this issue by hardcoding the DCM.U.dt variable (which was not generated during SPM's DCM model specification) into the model. We set it to the value 2, corresponding to our sequence's TR: this ultimately avoided up- and down-sample of the u array, as or model generates Y.y and U.u arrays of the same length.

-Adding this variable we were able to run the estimation. Nevertheless i) warnings were printed at each step at computation, as the script tries to invert an ill-conditioned matrix whose determinant is equal to zero. Further inspection showed that inversion occurred during ridge regression at line 124: sN_r = inv(t*W + l0_r); the matrix is W and the l0_r contains a column of zeroes as well.

-As a result, a NaN matrix was generated as final output.

We would kindly ask for your advice on how to overcome such hurdle as well as for any important caveat we should consider while using rDCM to analyze spectral connectivity in a resting state fMRI.

Kind regards,

Matteo Canini
Pasquale Anthony Della Rosa
Alessandro Petrini

StefanFraessle, 2020-04-21:
Hi Matteo, Pasquale, and Alessandro,

Thank you very much for your interest in the rDCM toolbox. Here are my thoughts on your questions:

(1) So far, we have not extended the model to include endogenous fluctuations in neuronal activity explicitly. Also, we have not thoroughly tested the stability and reliability of the current implementation of rDCM for resting-state data. This is mainly because our primary focus until now was on demonstrating the practical utility of the method in application to empirical task data, both in healthy individuals and clinical applications. Having said this, challenging the current implementation of rDCM with rsfMRI data is something I'm currently looking into. Unfortunately, I cannot make any reliable statements yet. We hope to be able to have something presentable soon.

(2) My suspicion is that something goes wrong in the specification of the model (e.g., when setting up the connectivity matrices). Note that rDCM does not fit cross-spectral densities (like spectral DCM) but the Fourier-transform of the dynamics of BOLD signal time series. Hence, when specifying your DCM structure in SPM12, you should **not** specify DCM (CSD) but a normal time-series DCM. If you enter such a time-series DCM to our toolbox, rDCM will take care of the transformation to the frequency domain. This also explains why your code gave an error regarding `DCM.U.dt`.

Regarding the error that you now end up with: I would probably need to take a look at the DCM structure that you are generating and that then enters `tapas_rdcm_estimate.m`. Having said this, I'm a bit hesitant to recommend to you any use of rDCM for resting-state data until we have not properly tested the model for this purpose. Would you think it might be sensible to pause with this project until we have finished our testing? As I mentioned, this is something I'm literally working on right now and I'm confident to be able to say more in a week or two.

I hope this is helpful.

With very best wishes,
Stefan

MatteoCanini, 2020-04-22:
Dear Stefan,	
we have just started working on the project and we’re currently processing data and testing rDCM implementation on some pilot subjects in order to verify if compliant to our data. Due to COVID emergency we are in a testing phase and we estimate that in a couple of weeks for sure we will move on to start batching the data analysis pipeline. It would be great to consider the eventual rDCM adjustments to resting-state data and to test them on our pilot dataset. In the meantime we will send you the specified DCM structure (i.e. TS specification) by email since the file is too big for direct uplaod in this post. However we had already tried to implement a TS DCM structure and feed it into tapas_rdcm_estimate.m and it returned the same NaN output matrix.	

Thank you for your availability. 

All the best,
Matteo and Pasquale.

StefanFraessle, 2020-04-22:
Dear Matteo and Pasquale,

Yes - feel free to send me the DCM structure by email. I think the NaNs are not related to the CSD vs. TS question, but is probably related to some small bug in specifying the model. I'll try my best to have a look at this as soon as possible - as you can probably imagine, things are incredibly busy here as well.

With very best wishes,
Stefan

MatteoCanini, 2020-04-24:
Dear Stefan,
yes we can perfectly imagine and thank you for your time and interest. We will wait to read news from you. In the meantime please contact us in case further detail on our implementation can be of any help.

All the best,
Matteo & Pasquale.