translationalneuromodeling/tapas #94: Query regarding relative_start_acquisition using Siemens VB files

trevorcsteward, 2020-05-01:
Dear Lars,

Thank you developing such a user-friendly toolbox. We are planning on incorporating it into our preprocessing pipeline for a series of projects using a Siemens 7T Mantegom system. 

Our team has run into some doubts as to whether the toolbox is properly synchronizing acquisition times from our DICOM file and our .puls and .resp files. I understand that this has been an issue for other and I downloaded the v7.2.2 release of PhysIO and essentially tried to imitate the Siemens_VB/PPU3T_Sync_Last.m example that was provided. However, if I leave the relative_start_acquisition field empty, I receive the following error: 

Index exceeds matrix dimensions.
In file "C:\spm12\toolbox\PhysIO\preproc\tapas_physio_simulate_pulse_samples.m" (???), function "tapas_physio_simulate_pulse_samples" at line 102.
In file "C:\spm12\toolbox\PhysIO\readin\tapas_physio_read_physlogfiles.m" (???), function "tapas_physio_read_physlogfiles" at line 104.
In file "C:\spm12\toolbox\PhysIO\tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 121.
In file "C:\spm12\toolbox\PhysIO\tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1596.

If you could provide some guidance surrounding this point, it would be greatly appreciated. I've attached the rest of the batch at the end of this message in case it might be useful. Please don't hesitate to let me know if you have any questions regarding our protocol. 

Best,
Trevor 

matlabbatch{1}.spm.tools.physio.save_dir = {'G:\7T_pilot\2020_test\scan_timing_log'};
matlabbatch{1}.spm.tools.physio.log_files.vendor = 'Siemens';
matlabbatch{1}.spm.tools.physio.log_files.cardiac = {'G:\7T_pilot\Physio_files\101_CNBT.puls'};
matlabbatch{1}.spm.tools.physio.log_files.respiration = {'G:\7T_pilot\Physio_files\101_CNBT.resp'};
matlabbatch{1}.spm.tools.physio.log_files.scan_timing = {'G:\7T_pilot\DICOMs\101_0603.dcm'};
matlabbatch{1}.spm.tools.physio.log_files.sampling_interval = [];
matlabbatch{1}.spm.tools.physio.log_files.relative_start_acquisition = [];
matlabbatch{1}.spm.tools.physio.log_files.align_scan = 'last';
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nslices = 84;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.NslicesPerBeat = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.TR = 0.8;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Ndummies = 0;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nscans = 603;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.onset_slice = 1;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.time_slice_to_slice = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nprep = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sync.scan_timing_log = struct([]);
matlabbatch{1}.spm.tools.physio.preproc.cardiac.modality = 'PPU';
matlabbatch{1}.spm.tools.physio.preproc.cardiac.filter.no = struct([]);
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.min = 0.4;
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.file = 'initial_cpulse_kRpeakfile.mat';
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.max_heart_rate_bpm = 90;
matlabbatch{1}.spm.tools.physio.preproc.cardiac.posthoc_cpulse_select.off = struct([]);
matlabbatch{1}.spm.tools.physio.model.output_multiple_regressors = 'multiple_regressors.txt';
matlabbatch{1}.spm.tools.physio.model.output_physio = 'physio.mat';
matlabbatch{1}.spm.tools.physio.model.orthogonalise = 'none';
matlabbatch{1}.spm.tools.physio.model.censor_unreliable_recording_intervals = false;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.c = 3;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.r = 4;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.cr = 1;
matlabbatch{1}.spm.tools.physio.model.rvt.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.hrv.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.noise_rois.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.movement.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.other.no = struct([]);

mrikasper, 2020-05-11:
Dear Trevor,

thank you so much for using PhysIO and diving into this forum before creating the issue.

The `simulate_pulse_samples` functionality is experimental and is invoked, if the timing info of the logfile and the DICOM don't match, so that the algorithm tries to fill up physiological data where none was acquired. So indeed, the first thing to make sure ist that the `.dcm` file you provide is the last one of the run, and that the `.resp` and `.puls` files relate to that exact run.

If this is all in order, the easiest thing would be to share these files with me to run the batch myself and look at the debugging plots. It's hard to say without looking at the data, if this is rather a code or data issue :-)

Let me know, if this is an option for you and how you could share the data in case!

All the best,
Lars

trevorcsteward, 2020-05-11:
Dear Lars,

Thank you so much for getting back to me. 

Since I last wrote you, I discovered that the v7.2.2 release of Phsyio hadn't been properly incorporated into my spm/toolbox folder and I was able to run the script that I had previously attached. 

However, the output generated from 'tapas_physio_report_contrasts' leads me to suspect that some syncing issues still persist and I've attached the data that was used for our batch. The .dcm attached does indeed belong to the last volume of our paradigm and it also belongs to the last sequence of the scanning session. I've also attached a copy of the first-level script that we used. Please let me know if also having a copy of the 4D .nii file would be helpful.

As you'll also see in the DICOM file, we used a sub-second multi-band sequence and I'm unsure on how to best set the sqpar parameters for this type of sequence. From what I understand from reading other threads on this forum, it doesn't make a substantial difference considering that we'll be running our GLM in SPM, however, I thought it would be best to seek clarification on this point before moving forward. 

Thanks again for all help.

Best,
Trevor 

[Phsyio_CNBT_101.zip](https://github.com/translationalneuromodeling/tapas/files/4612495/Phsyio_CNBT_101.zip)

mrikasper, 2020-05-11:
Dear Trevor,

thank you for the data, this is exactly what I need. I don't need the `.nii` files at the moment. 

I'll get back to you shortly.

All the best,
Lars

mrikasper, 2020-05-18:
Dear Trevor,

just to clarify something regarding the sampling: Do you have a specific subject cohort, e.g., athletes, because the heart rate PhysIO estimates in your example dataset is unusually low (s.th. at about 25 bpm...). For example, the peak detection works much better, if you limit the `max_heart_rate_bpm` to 60 (instead of 120). Just wanted to clarify before I look into an error regarding sampling rate :-)

With respect to your multiband question, I think it's usually fine to put the true `TR`, but `nSlicesTotal/multibandFactor` as the number of slices, i.e., the number of excitations that were performed during one TR. For a single band, where you have 1 excitation per slice, these numbers are identical, but if you have 84 slices and a multiband-factor of 4 (4 slices/excitation), you would put 21=84/4 here. It will not affect the end result, just speed up computation, but you don't need the finer resolution, because no slices were excited then. What is important, though, is to which of these 21 slice excitation events you want to align your regressors. Even though RETROICOR can compensate phase delays between slices via sine/cosine terms, you still end up with most robust estimates for the slices that you chose as `onset_slice`, because its timing is exact.

But I agree, looking at the F-Maps, that there is probably something odd about the timing. I'll look a bit deeper, but it might take some time.

All the best,
Lars

trevorcsteward, 2020-05-18:
Hello Lars,

Thank you for getting back to us and for resolving our doubts regarding how to best input information for multiband sequence. 

I am equally perplexed by the low BPM found in the example subject that I sent you. Although this participant was young and athletic, I agree that a BPM of 25 is unusually low. Moving forward, is there a way to read the raw .puls files for our participants to determine how to best set our max_heart_rate_bpm? 

My student and I have been working through different combinations of parameters in the Physio batch to try to identify the timing issues that are providing us with the odd results in the F-maps. Curiously, we were finally able to obtain results that strongly resembled the figures in the Physio publication by using nominal scan_timing.sync and manually setting our sampling_interval to 0.02

I've attached a copy of the script that was used and the output .ps file from running tapas physio report contrasts(). Do you believe that we can proceed with our 2nd level analysis using these parameters? 

Thank you again for your help.

Best,
Trevor 

[Lars_2.zip](https://github.com/translationalneuromodeling/tapas/files/4642168/Lars_2.zip)

mrikasper, 2020-07-10:
Dear Trevor,

my apologies for the late reply, COVID-19 is still limiting my availability.

Regarding the heart rate, I think the diagnostic plot for the raw physiological time series with the display of your beat-to-beat intervals is your best quality control tool, and it even gave a warning when the `max_heart_rate_bpm` threshold led to bad peak detection (see screenshots below with 120 and 60 pm, respectively)

[image]
[image]

Your contrast map results using the nominal timing look reasonable, I'm particularly happy that our RVT bugfix seems to have worked spells for those regressors. I think you could try the 2nd level analyses.

However, one cannot say whether the explained variance wouldn't be even better, if the correct timing was used. The sampling interval of 0.02 is also read correctly from the file automatically, so it has to do something with the timing onset that the nominal scan timing setting gets right, but the read-in from DICOM doesn't. But when I plot (via `tapas_physio_review`) the data in the `physio.mat` you provided in the last post to one created using the DICOM-timing, I get only a few seconds shift (nominal is shifted 2.3s earlier):

- *nominal timing*
[image]

- *DICOM timing*
[image]

This could explain some reduced effects, but the F-map you sent in your earlier post looks dramatically worse. 

So just to confirm, could you run subject 101 again with all settings identical and two cases: the scan timing sync switched to (1) nominal and (2) DICOM-scan-timing sync, please? And then send me the F-maps in both cases again?

Thank you so much for your help - it would be great, if we could get to the bottom of this!

All the best,
Lars

trevorcsteward, 2020-07-10:
Hi Lars,

Thank you for getting back to us and don't worry at all about the delay; it's a strange time for all of us. 

I've attached folders including the F-maps using both nominal and DICOM-scan-timing sync. The Physio and first-level SPM scripts are also located within each folder. You'll see that the only parameter that has changed is the syncing method used and the specification of the sampling interval (none vs. 0.02). GitHub only allows me to upload a maximum of 10 MB and I'd be happy to send you the complete 1st level results via email if that's preferable. 

After looking at the breathing belt histogram for subject 101 and considering the oddly low average BPM, I have the impression that the quality of her recorded physiological data isn't the highest. That being the case, I included another example subject whose recordings are much 'cleaner' and whose F-maps are much more similar to the results in the Physio Toolbox publication (the Card x Resp Interaction map is quite impressive). While comparing the physio.mat files using either nominal or DICOM-scan-timing sync, apart from the 2.3 second shift that you detected, we also noticed that the dt value within physio.ons_secs was .0201 when using DICOM-scan-timing sync as compared to .0200 when using nominal syncing. Could this be an explanation as to why the DICOM results are off?

Since my last message, we've begun to use the nominal timing parameters on our entire sample and our second-level results appear to have greatly benefited by using the toolbox. We're currently in the process of examining tSNR gains across all our subjects and I'll keep you posted on any issues that come up. 

Best,
Trevor 

[Send_Lars_July_2020.zip](https://github.com/translationalneuromodeling/tapas/files/4902004/Send_Lars_July_2020.zip)