translationalneuromodeling/tapas #96: Where to find sample PhysIO code not using SPM batch?

gjoue, 2020-06-18:
Hi,
I would like to run PhysIO without SPM with physiological files I have converted into a format following the BIDS convention, and for the life of me I cannot find the sample scripts to do it as described in the section "One-Page-Quickstart (Matlab only, no SPM)" on 
https://gitlab.ethz.ch/physio/physio-doc/-/wikis/QUICKSTART
and in the "PhysIO Toolbox Quick-Start Manual".  Where can I find them/download them (without using any automated scripts because I don't have administrative access to the Matlab install directory).
Thanks!
Gina

mrikasper, 2020-06-19:
Dear Gina,

thank you for trying out the toolbox. The Matlab-only examples should be downloaded once you run 
`tapas_download_example_data()`, as mentioned in the Requirements section of the quickstart. 

But I’m not sure from the top of my head whether the BIDS example is already included there. If not, please let me know and I’ll make it available.

All the best,
Lars

gjoue, 2020-06-21:
Hi Lars,

Really curious about the toolbox!
As I don't have administrative rights where I'm running Matlab, I can't
run tapas_download_example_data()...would you mind posting the BIDS
example somewhere?

Thanks!
Gina

mrikasper, 2020-06-21:
Dear Gina,

I wonder why you need admin rights to execute the Matlab function. Maybe it's saving it to a non-accessible location on your system? 

The function basically just downloads the right zip-file from our web location (as listed in [`log_tapas.txt`](https://github.com/translationalneuromodeling/tapas/blob/master/misc/log_tapas.txt)). 

If you have a look in there, it says that the current version of the example data can be found here: https://www.tapas.tnu-zurich.com/examples_v3.2.0.zip

Therein, you will also find a 3T example for pulse plethysmographic data in BIDS format. `bids_ppu3t_matlab_script.m` is the script that uses Matlab only.

I hope that helps!

All the best,
Lars

closed 2020-07-09 by mrikasper