translationalneuromodeling/tapas #97: what to do with low respiratory signal amplitude

gjoue, 2020-06-24:
Hi,

This is an absolute newbie question... 

We used a pulsoximeter and a breathing belt during our fMRI study with CED's Spike2 system and would like to use PhysIO's implementation of RETROICOR to create nuisance regressors to regress out these two physiological noise signals from our BOLD. I converted Spike2's proprietary format into BIDS convention and followed the PhysIO example script bids_ppu3t_matlab_script.m (sticking to the settings in that file for the most part) to import my BIDS tsv/json files into PhysIO. Testing on one subject who did several runs in one scanning session (no changes in equipment setup), I see from the PhysIO debug plots that sometimes the respiratory signal had very low amplitudes for certain runs, and for these runs (4 out of the 6 runs,  not consecutive), PhysIO created no/only a handful of regressors. 
Here is a side-by-side comparison of two runs from the test subject (left: where there was a problem in creating respiratory regressors and in the right where the regressors looked somehow reasonable?):
[image]

I see that the "lower threshold for selecting outliers" was higher than the respiratory signal -- and I'm not sure where to actually change this or is this an indication of a deeper problem? 

I saw that there was a previous post related to a similar question regarding preprocessing but for the cardiac signal
https://github.com/translationalneuromodeling/tapas/issues/10

and I tried applying a cardiac preprocessing filter, in case that would also impact the respiratory signal (I guess not?). In any case, I tried both Butterworth and the Chebyshev Type II filter with a lower minimum and lower bound on the filter passband (0.15 Hz following some studies that reported that as a useful cutoff for denoising), but that didn't do much.

I'm not sure what other info is needed to explain my problem, but I have also uploaded the entire set of PhysIO diagnostics plots and the mat file here
https://gigamove.rz.rwth-aachen.de/d/id/SfgkKLo3z3Czah

Thanks a ton in advance for any pointers/help/advice!
Gina

gjoue, 2020-06-29:
I am slowly stepping through the PhysIO code, and I see that the respiratory signal is preprocessed in `tapas_physio_filter_respiratory.m` with a 2nd-order Butterworth bandpass filter with a lower cut-off hard-coded at 0.1 Hz and upper cut-off at 5Hz, then normalized by the maximum amplitude magnitude, and outliers +/-3 std.dev. of the mean amplitude tagged. 

For my example problematic run, the pre-processed respiratory signal (red) looks pretty good (blue = raw, green = filtered, cyan = filtered+normalized):

[image]

Zooming in on the last few seconds:
[image]

I traced the problem for the runs where no nuisance regressors for the respiratory signal was calculated to `tapas_physio_detect_constants.m` where "suspicious constant values" are tagged for what seems like the entire signal. This is the diagnostic plot from that function (not sure if this is linked to the config debug level parameter in` tapas_physio_new.m`, but this plot will only print if the debug level is set to 4+, but `tapas_physio_new.m` only documents up to debug level 3):

[image]

Here is a close-up of a slice of time:

[image]

I guess the amplitude is so small that the signal is treated like it were flat?

Just to give an idea of the calculations that come out from the detect-constants function, for the respiratory data from which the plots are generated (`numel(y)` = 600601), `deltaMaxDiff` =  single   1.1921e-07. And after l.56,  `numel(idxIsConstant)` = 2387, and `numel(idxIsConstantFinal)` = 1456.

I don't think any part of the signal should've been tagged as "constant"?

Would someone guide me in how to best work around this problem?
Many thanks in advance!
Gina

mrikasper, 2020-07-10:
Dear Gina,

my apologies for the late reply, COVID-19 reorganization is still holding a firm grip on my time budget.

This is a very interesting dataset, and thank you for going through it in so much detail. The respiratory amplitudes for the failing run seem small, but the SNR looks actually good. I can only speculate that there must be a hard-coded threshold somewhere, eg. when censoring constant regimes of signal or within the scaling of the breathing signal.

What would be very helpful is to see both your matlab script (or batch) and the BIDS input file (if you can share it in an anonymized fashion) that created this weird behavior. Did you try to run the code with the censoring of constant time periods switched off, i.e, `physio.model.censor_unreliable_recording_intervals = 0`?

Having said this, the next release of PhysIO will include an improved version of `tapas_physio_filter_respiratory` thanks to Sam Harrison, that deals more gracefully with filter ringing artifacts etc. It should be available in the next couple of days.

All the best,
Lars

gjoue, 2020-07-10:
Thanks a ton for getting back to me, Lars, despite all that!

Yes, I did run` tapas_physio_main_create_regressors()` with censoring constant time periods turned off (`physio.model.censor_unreliable_recording_intervals = 0`), but the problem remains. I've also tried different filters (Butterworth vs. Cheby2)

How do I see that the SNR looks good or is good enough?

I've attached the Matlab script and the example BIDS input files, but here is the configuration I passed to PhysIO (which closely follows the example scripts you have made available), just for quick at-a-glance reference:

```
physio.model.orthogonalise                         = 'none';  % orthog mainly needed if acq triggered to heartbeat (set to 'cardiac') OR if session mean should be evaluated
physio.model.censor_unreliable_recording_intervals = 0;      % only for instantaneous effect of poor recordings, e.g., the phase estimates leading to RETROICOR regr. No censoring done for convolution models (HRV, RVT) since effects of unreliable recordings have long-term effects.
%% RETROICOR (Model): Glover et al. 2000
%% Retrospective image correction method, based on Fourier 
%% expansion of cardiac and respiratory phase, plus 
%% multiplicative interaction terms (Harvey et al, 2008)
physio.model.retroicor.include   = true;
physio.model.retroicor.order.c   = 3;     % natural number, order of cardiac phase Fourier expansion
physio.model.retroicor.order.r   = 4;     % natural number, order of respiratory phase Fourier expansion
physio.model.retroicor.order.cr  = 1;     % natural number, order of cardiac-respiratory-phase-interaction
%% RVT (Model): Respiratory Volume per time model , Birn et al, 2006/8
physio.model.rvt.include         = false;
physio.model.rvt.delays          = 0;
%% HRV (Model): Heart Rate variability, Chang et al, 2009
physio.model.hrv.include         = false;
physio.model.hrv.delays          = 0;
%% noise_rois (Model): Anatomical Component Correction, Behzadi et al, 2007
%% PC of time series of all vx in given ROI noise, e.g. CSF, vessels, white matter
physio.model.noise_rois.include = false;
physio.model.noise_rois.thresholds        = 0.9;
physio.model.noise_rois.n_voxel_crop      = 0;
physio.model.noise_rois.n_components      = 1;
physio.model.noise_rois.force_coregister  = 1;
%% movement (Model): Regressor model 6/12/24, Friston et al. 1996
physio.model.movement.include             = 0;
physio.model.movement.file_realignment_parameters = '';
physio.model.movement.order               = 6;
physio.model.movement.censoring_threshold = 0.5;
physio.model.movement.censoring_method    = 'FD';
%% other (Model): Additional, pre-computed nuisance regressors 
physio.model.other.include                = false;
%physio.model.other.input_multiple_regressors = '';
%% ________ OUTPUTS
physio.verbose.level        = 3;
physio.verbose.use_tabs     = false;
physio.ons_secs.c_scaling   = 1;  % scaling factor for cardiac data after normalization
physio.ons_secs.r_scaling   = 1;  
```
Thank you for helping!

[test_PhysIO.tar.gz](https://github.com/translationalneuromodeling/tapas/files/4903640/test_PhysIO.tar.gz)

gjoue, 2020-07-14:
Sorry, my error! I found a mistake in my call to PhysIO where I did not adjust the number of scans `physio.scan_timing.sqpar.Nscans` by the number of dummy scans (and my physiological log inputs were already adjusted, i.e. corresponded to data for a fewer number of scans). Fixing this results in the PhysIO diagnostic plots for the two runs I had originally posted plots for looking like this (left/right corresponds to the left/right of the plots as in my original post):
[image]

which seems to look much better? 
I wouldn't have thought specifying the incorrect number of scans would result in the problem I saw?
Sorry for that -- and just one last question/I would still interested how you tell that the SNR looks good or is good enough in these plots?

Many thanks,
Gina

mrikasper, 2020-07-14:
Dear Gina, 

thank you for following up on the issue. It's great that it works now, and indeed strange it should depend on the number of specified volumes. The only thing I could imagine is that the interval that was additionally included with more volumes had very different amplitude levels of the respiratory trace and the template creation for the peak detection got compromised by that.

I tried to run your code (there was no `.json` file, so I assumed `sampling_interval = 0.001` and `relative_start_acquisition = 0` as in your previous `physio.mat`), but couldn't reproduce the strange behavior easily.

With respect to telling whether SNR of physiological recordings is good enough, I suggest to look at the top left of your figures (cutout region for phys regressors), and zoom in a bit. If the traces look regular and smooth, with clearly identifiable peaks, you're probably fine, no matter what the absolute amplitude range was. Have a look at [Figure 5](https://www.sciencedirect.com/science/article/pii/S016502701630259X#fig0025) of our PhysIO paper to see with which kind of noise levels the peak detection algorithm can deal. It's actually quite robust, and your data is similar to the top panel there (low noise).

All the best,
Lars

gjoue, 2020-07-17:
Hi Lars,

I am coming up against a problem with respiratory signal phase calculation again (piggybacking on this thread as the observed behavior is similar). For cases when the physiological recording was started after the MRI, I see that when PhysIO reads in the physiological files (`readin/tapas_physio_read_physlogfiles.m`), it prepends the physiological data with simulated samples (`tapas_physio_simulate_pulse_samples()`), and in my test case where this happens, I  get a problem with the respiratory signal phase calculation.

I doublechecked the timing of my BIDS physiological files, and I think it is correct. Just to walk through my calculations, just in case -- I have 385 analyzed MRIs (5 of 390 were dummy vols), but I started my physiological recording too late and my logs show that only 370 MRI triggers were received. Following BIDS guidelines where "StartTime" is the start of the physiological logs relative to the first MRI volume analyzed, I then have

StartTime = 15 missing TRs * 1.54 sec/TR =  23.1 sec 

Here are the diagnostic plots, with the top 2 rightmost plots outputs from simulating samples  (which is not saved to the debug plot file, by the way), and the leftmost bottom plot a zoomed-in view of the top right plot:

[image]

I also have uploaded the BIDS physiological files (with .json this time!) and the physio.mat here:
https://gigamove.rz.rwth-aachen.de/d/id/8BVM7vHSqAUULX

Maybe I've done something wrong?

Thanks for your help!
Gina