translationalneuromodeling/tapas #104: HGF: Question on transformations in tapas_logrt_linear_binary

dsstolz, 2020-07-21:
Dear Dr. Mathys,

In the response model "tapas_logrt_linear_binary", irreducibel uncertainty, estimation uncertainty, and unexpected uncertainty are used to predict log-transformed reaction times. 

I was wondering why the estimation uncertainty and the unexpected uncertainty are computed by "transforming down to 1st level" (as indicated in the comments in the .m-file) before being used in the linear model. Precisely, the variance from level 2 (sa2) and the exponential mean from level 3 exp(mu3), respectively, are multiplied with tapas_sgm(mu2, 1).*(1-tapas_sgm(mu2, 1)).

Unfortunately, I could not figure out the rationale for doing so by myself. Would you be so kind to provide this rationale here? Or, put differently, is the transformation of sa2 or exp(mu3) when used as a predictor in linear models necessary/advantageous more generally, or is this just a particularity of this response model?

Thank you very much in advance,
David

chmathys, 2020-09-17:
Dear David,

We do this transformation so that all predictors in the linear model are on the same scale and can be interpreted in terms of outcome probability (instead of a nonlinear transformation of that). For example, without transforming to this space, the uncertainty sigma2 at the second level is associated not just with uncertainty about outcome probability, but also with the current value of mu2 because the sigmoid transformation leads to large sigma2 values (relative to uncertainty about outcomes at the first level) when mu2 is far from zero in either direction.

The mathematical rationale for this is given in the supplementary material of this paper:

Iglesias, S., Mathys, C., Brodersen, K. H., Kasper, L., Piccirelli, M., den Ouden, H. E. M., & Stephan, K. E. (2013). Hierarchical Prediction Errors in Midbrain and Basal Forebrain during Sensory Learning. Neuron, 80(2), 519–530. https://doi.org/10.1016/j.neuron.2013.09.009

Best wishes,
Christoph

dsstolz, 2021-03-04:
Dear Christoph,

thanks a lot for your response and the hint for the mathematical rationale!

Best,
David

closed 2021-09-09 by chmathys