translationalneuromodeling/tapas #105: PhysIO not in SPM toolboxes after installation

artturuohola, 2020-08-04:
Hello

I'm new to SPM and PhysIO

I follow the installation steps and after successful installation, start SPM by typing "spm" in matlab console.

SPM gui opens but there is no PhysIO toolbox in toolboxes drop menu. Am i doing something incorrectly?

mrikasper, 2020-08-06:
Hi,

thank you for trying out PhysIO!

The PhysIO Toolbox does not show up in the Toolbox Dropdown Menu of SPM. It can be found if you start the Batch Editor via the "Batch" Button on the SPM GUI and then select SPM->Tools->TAPAS Physio Toolbox from the Menu Bar. See the screenshot in the Wiki page [Quickstart](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/QUICKSTART).

I hope this helps!

All the best,
Lars

ttsengataeuroling, 2023-05-06:
Hello, 

I have the same problem that I have successfully install the PhysIO with the codes as:
"OK.
Success: PhysIO successfully installed, integration with Batch Editor possible.
Updating SPM batch editor information...done.
Finished!"

However, it does not appear in Batch Editor with SPM->Tools->TAPAS Physio Toolbox
even though I have also applied the Troubleshoot as
"If the PhysIO Toolbox does not show up in the SPM Batch Editor, the necessary matlab code-files cannot be
found by SPM. Manually copy the PhysIO Toolbox PhysIO/code folder to spm/toolbox/PhysIO (see Figure 1)."

Would it be possible that you would know how to fix this problematic situation please?
Or would you know how to add physio data in spm analysis please?

Thank you very much and have a good day.

charlotte-c-s, 2024-03-25:
I'm having a similar issue to @ttsengataeuroling - everything seems to be initialising happily, but the PhysIO toolbox is not coming up in the SPM batch editor, and I can't seem to copy the code folder to the toolbox manually as suggested in the troubleshoot. All of the integration tests are also failing, as if the folders/codes don't exist. I've uninstalled and re-installed the entire package, but I'm still having the same issue.
Any suggestions would be much appreciated!
Thank you,
Charlotte