translationalneuromodeling/tapas #109: tapas variational approximation invalid

eoberwelland, 2020-09-21:
Hi there, 

thank you very much for providing the tapas toolbox and the help function here. 
We've used the tapas toolbox (version 2.0.0.1) for a while to do some RW and HGF modeling and now want to update to the newest tapas toolbox version. While trying to do so, we always get the following error: "tapas variational approximation invalid". 
We tried to use various Matlab versions, but it didn't held. We also followed your advice in the documentation and lowered the omega prior values, but this also didn't help. We don't have this error with the older tapas version. 

Maybe this is relevant: we added the parameter theta to tapas_hgf_binary_config.m file. This was no problem with the older tapas version. 

I hope I provided sufficient information and thank you very much in advance. 
Eileen

chmathys, 2020-09-21:
Hi Eileen,

Please post your config file, the command you use to call tapas_fitModel(), and the exact output you receive.

Thanks,
Christoph

eoberwelland, 2020-09-21:
Thanks, 

we use the following command: 
bopars_hgf = tapas_fitModel(choice_input, cont_input,'tapas_hgf_binary_config_custom', 'tapas_softmax_binary_config','tapas_quasinewton_optim_config');

I attached the config file as a .txt file. 

This is the error we receive: 

Error using tapas_hgf_binary (line 215)
Variational approximation invalid. Parameters are in a region where model assumptions are violated.

Error in tapas_fitModel>negLogJoint (line 436)
    [dummy, infStates] = prc_fun(r, ptrans_prc, 'trans');

Error in tapas_fitModel>@(p)[negLogJoint(r,prc_fun,obs_fun,p(1:n_prcpars),p(n_prcpars+1:n_prcpars+n_obspars))]
(line 349)
nlj = @(p) [negLogJoint(r, prc_fun, obs_fun, p(1:n_prcpars), p(n_prcpars+1:n_prcpars+n_obspars))];

Error in tapas_fitModel>optim (line 356)
[dummy1, dummy2, rval, err] = nlj(init);

Error in tapas_fitModel (line 199)
r = optim(r, r.c_prc.prc_fun, r.c_obs.obs_fun, r.c_opt.opt_algo);

Error in run_analysis/model_data (line 71)
        bopars_hgf = tapas_fitModel(choice_input, cont_input,'tapas_hgf_binary_config_custom',
        'tapas_softmax_binary_config','tapas_quasinewton_optim_config');

Error in run_analysis (line 46)
    sh = model_data(sh,i);
 
Thank you very much. 
Best, Eileen 

[tapas_hgf_binary_config_custom.txt](https://github.com/translationalneuromodeling/tapas/files/5254429/tapas_hgf_binary_config_custom.txt)

chmathys, 2020-09-21:
Hi Eileen,

Thanks. Your custom config file is incompatible with the current toolbox version. This is causing the errors.

However, you may not really need a toolbox upgrade if you were happy with the setup you had. If you prefer to have theta with an upper bound instead of omega3=log(theta) without  an upper bound, that’s fine. Just keep using the older version you had.

If you have a particular reason to upgrade, then you will have to adapt your config and main model files accordingly. Note the following points:

— In current versions, kappa1 exists and its hyperpriors cannot be NaN in the config file
— If you use theta instead of omega3, you need to use a corresponding version of tapas_hgf_binary.m (where the omega3 entry in the parameter vector is *not* exponentiated since it’s now a theta entry)
— Same goes for tapas_hgf_binary_transp.m, where you need to apply tapas_sgm() to the theta entry

(Even if you stay with your old setup, please make sure the last two points are ok there).

I hope this helps.

Best wishes,
Christoph

eoberwelland, 2020-09-22:
Thank you very much for the quick and detailed response!

We might then just stay with the older version, but I am not sure anymore if we have done the adjustments (correct), you've mentioned. So that I have two more questions: 

1. in the "tapas_hgf_binary.m" file we have the following line: 
    th   = exp(p(5*l-1));
   does your aforementioned comment apply to this line? and the "exp" here is wrong?

2. in the "tapas_hgf_binary_transp.m" file we don't use the tapas_sgm() function yet. This is what the  "tapas_hgf_binary_transp.m" looks like right now. Does that mean we have to add theta here as well an use the tapas_sgm() just for theta? 

    pvec    = NaN(1,length(ptrans));
    pstruct = struct;

    l = r.c_prc.n_levels;

    pvec(1:l)         = ptrans(1:l);                           % mu_0
    pstruct.mu_0      = pvec(1:l);
    pvec(l+1:2*l)     = exp(ptrans(l+1:2*l));                  % sa_0
    pstruct.sa_0      = pvec(l+1:2*l);
    pvec(2*l+1:3*l)   = ptrans(2*l+1:3*l);                     % rho
    pstruct.rho       = pvec(2*l+1:3*l);
    pvec(3*l+1:4*l-1) = exp(ptrans(3*l+1:4*l-1));              % ka
    pstruct.ka        = pvec(3*l+1:4*l-1);
    pvec(4*l:5*l-1)   = ptrans(4*l:5*l-1);                     % om
    pstruct.om        = pvec(4*l:5*l-1);

Hope my questions are clear! Thank you very much! Best, Eileen

chmathys, 2020-11-16:
Dear Eileen,

This is just a quick message to ask whether you’ve been able to get this to work. Please tell me if you still have questions.

Best wishes,
Christoph