translationalneuromodeling/tapas #110: no respiratoty regression created

adomagalik, 2020-10-12:
Dear PhysIO creators and users,
I would like to use the toolbox for physiological artefact removal with the RETROICOR method for fMRI data that were acquired with CMRR sequence (MB 2; physiological data saved in dicom). I tried to use it via GUI and it did not work properly - no regressors for respiratory signal were created. I thought that I did something wrong with the settings, so I decided to see the examples and I run the "siemens_vd_ppu3t_matlab_script.m". The same thing happened - no regressors for respiratory signal. I then tried with "bids_ppu3t_matlab_script.m" and it seems that it works fine. Attached the output figures (BIDS example on the left; SIEMENS VD on the right).
Any idea what is wrong?I have the newest version of tapas. I run it on Matlab 2018b
Thanks!
Aleksandra
[image]

mrikasper, 2020-10-14:
Dear Aleksandra,

thank you for trying out the PhysIO Toolbox and for your detailed bug description. If you are willing and able to share the data, it would be great if you could send me the script or matlabbatch files you used, as well as the dicom or BIDS-files.

Did you use the dicom-file directly or did you convert them to the VD format with separate files by using `extractCMRRPhysio.m` from https://github.com/CMRR-C2P/MB ?

And how did you convert the data to BIDS for the left-hand-side example?

All the best,
Lars

adomagalik, 2020-10-14:
Dear Lars,

Thank you for quick response!

The data you see are from the PhysIO 'examples' folder and I run the scripts that were there with no changes.
The BIDS data (left example) are exactly: '_../tapas-master/examples/4.0.0/PhysIO/BIDS/PPU3T/sub-s082_task-fnclearning_run-01_physio.tsv_' and script is: '_../tapas-master/examples/4.0.0/PhysIO/BIDS/PPU3T/bids_ppu3t_matlab_script.m_'. The Siemens VD (right example) are:  '_tapas-master/examples/4.0.0/PhysIO/Siemens_VD/PPU3T/Physio_20180101_120001_pppppppp-hhhh-yyyy-ssss-ioioioioioio_****.log_' run with script:  '_../tapas-master/examples/4.0.0/PhysIO/Siemens_VD/PPU3T/siemens_vd_ppu3t_matlab_script.m_' .

I suspect that the problem with my data (attached) is the same as with those in Siemens VD example. Yes, I used extractCMRRPhysio.m script to get '**.log'  files

Thanks!
Aleksandra

[PhysioLog.zip](https://github.com/translationalneuromodeling/tapas/files/5377180/PhysioLog.zip)

JoeZiminski, 2020-10-19:
Funnily enough I have just had a similar issue, which may or may not be related - are you using linux? 

I have found that no physiological regressors are generated while using PhysioIO v7.3 on linux, but v7.2 works okay. Also, both physio v7.2 and v7.3 are working on windows (please see below for more details).

OS: Ubuntu 16.04.7 LTS (Xenial Xerus)
matlab version: R2019a
respiration regressors are not generated in physIO version v7.3 but are working correctly in v7.2.

OS: Windows 10
matlab version: R2019a.
respiration regressors are generated in both physIO version v7.2 and v7.3.

adomagalik, 2020-10-19:
Hi,

Yes, I am working on Linux (Ubuntu 16.04); Matlab 2018b. Thanks for your post - I will check the toolbox on Windows and/or check previous version.

Aleksandra

mrikasper, 2020-10-19:
Dear Aleksandra and Joe,

thank you for providing more detail. I just downloaded TAPAS and the example data freshly and tried the `siemens_vd_ppu3t_matlab_script` on a Linux server (CentOS7, Matlab R2018a):

The results look fine to me:
[image]
[image]

I do, however, get a warning
```
Warning: Invalid trend type '3'.. assuming 'linear'. 
> In detrend (line 52)
  In tapas_physio_filter_respiratory (line 38)
  In tapas_physio_main_create_regressors (line 209)
  In siemens_vd_ppu3t_matlab_script (line 66) 
```

that's related to a change in Matlab's `detrend` function between R2018a and R2019b.

So my suspicion is that it's not about Linux and Windows per se, but about differences in `tapas_physio_filter_respiratory` that might be related to this detrending. What you can see, for example, in Aleksandra's right hand side top plot example is an overshoot of the filter at the end of the respiratory time series that makes the rest look flat. We overhauled the filtering of respiratory data for v7.3, so it should actually be more robust than before, so it's important to find out what's going on here.

1. Could you confirm whether you get this `detrend` warning on either of your Win/Linux systems, @adomagalik and @JoeZiminski ?
2. @adomagalik - It could still very well be that your own data works fine with the toolbox v7.3, because the example VD dataset we use is rather short (35s) and might create some extra problems. Maybe worthwhile trying by replacing the input data and parameter info from the example script - I couldn't try it myself, because I am lacking the scan info (TR, nVolumes etc.) for your files.

All the best, and thank you for your help,
Lars

mrikasper, 2020-10-19:
Dear Aleksandra,

I just looked into your `_Info.log` file and retrieved the relevant scan info from there. I find that on my Windows system (Win 10, R2019b) the attached matlab script (adapted from the Siemens VD PPU example, just rename it to `.m`) runs through with reasonable regressor output, also for the respiratory data:

[siemens_vd_ppu3t_matlab_script_adapted.txt](https://github.com/translationalneuromodeling/tapas/files/5404693/siemens_vd_ppu3t_matlab_script_adapted.txt)

[image]

[image]

Maybe you could try the same script on your Linux machine with PhysIO v7.3 and see whether it works?

Of course, we also still have to understand what's going on with the example dataset.

Thank you again for your feedback and all the best,
Lars

JoeZiminski, 2020-10-19:
Hi Lars,

I did a bit more digging, I don't see any detrend warning when running through the example dataset (please see pictures of not-working version below). I also tried switching between version  7.2 and 7.3 on the same matlab version (2019b) and it worked okay on 7.2 but not 7.3.

Through incremetally swapping scripts from 7.3 into the 7.2 package to see when it breaks, the problem does not seem to be in any of the main scripts in PhysIO dir, nor in the model or readin dir. However, when swapping the folder 'preproc' 7.3 into 7.2, the problem occurs. Some further digging shows that the fields:

ons_secs.fr
ons_secs.r_sample_phase
ons_secs.acq_code

used as input into tapas_physio_create_retroicor_regressors() script are different between the working (7.2) and not working (7.3) version. 

Please find an plot below of the working and broken .fr field. The r_sample_phase is all nan in the broken version. Please also find the full saved ons_secs variables for the working and broken version attached.

I am happy to look more into this as I appreciate you can't reproduce the error, however I couldn't find what script / where these subfields are generated in the prepro package.

Best,
Joe

[ons_secs.zip](https://github.com/translationalneuromodeling/tapas/files/5404886/ons_secs.zip)
[tapas_issues.pdf](https://github.com/translationalneuromodeling/tapas/files/5404900/tapas_issues.pdf)

mrikasper, 2020-10-19:
Dear Joe,

this is indeed very helpful, thank you for investing the time! From your comment and the pdf-plots, it seems that indeed `tapas_physio_filter_respiratory` is the culprit, since this is creating `ons_secs.fr` in line 209f. of `tapas_physio_main_create_regressors`:

```
                ons_secs.fr = tapas_physio_filter_respiratory(ons_secs.r, ...
                    ons_secs.dt, doNormalize);
```

`r_sample_phase` is derived from that during the RETROICOR regressor creation. `acq_code` is odd, but maybe the issue is timing-related.

Do you have a longer example time series at hand (e.g., Aleksandra's with the script I just posted above) to check whether the error persists for longer files in v7.3 and in comparison to v7.2? The reason I am asking is that our new filtering algorithm uses some padding to avoid ringing artifacts at start/end of the time series.

Thank you again for your help and all the best,
Lars

JoeZiminski, 2020-10-21:
No problem! the least I could do after all the use I have got from the toolbox.

It turns out the problem in my case was the 'Fieldtrip' toolbox that I am not familiar with that was hidden away on the cluster in the SPM toolbox. Fieldtrip has a whole folder of functions that shadow matlab in-built functions, including filtfilt :@@@. After removing this folder from the path the toolbox worked without issue (and I am now going through the cluster pathdef with a fine tooth comb!). I wonder if this is also the case for you Aleksandra?

Thanks for your support Lars on using the excellent physio toolbox!

Cheers,
Joe

adomagalik, 2020-10-21:
Hi Lars and Joe,

I was able to run scripts on Windows 10, however the newer Matlab version
(2020a). It worked both for example and my data.

Going back to Linux, yes I have a 'detrend' warning. In the case of Siemens
VD example and my data I also have additional warnings about ignoring
sampling intervals (also on Windows). See matlab output below.

Aleksandra

*>>
run('/.../tapas-master/examples/4.0.0/PhysIO/BIDS/PPU3T/bids_ppu3t_matlab_script.m')*
TR = 1.450 +/- 0.010 s (Estimated mean +/- std time of repetition for one
volume; nTriggers = 409)
No cardiac R-peak (heartbeat) events provided
Warning: Invalid trend type '3'.. assuming 'linear'.
> In detrend (line 52)
  In tapas_physio_filter_respiratory (line 38)
  In tapas_physio_main_create_regressors (line 209)
  In bids_ppu3t_matlab_script (line 64)
  In run (line 91)
    maxscan (incl. dummies) = 408
    tmin (1st scan start (1st dummy))=  29.82 s
    tmin (1st scan start (after dummies))=  29.82 s
    tmax = 621.32 s
    mean TR =   1.45 s

*>>
run('/../tapas-master/examples/4.0.0/PhysIO/Siemens_VD/PPU3T/siemens_vd_ppu3t_matlab_script.m')*
Warning: Ignoring given sampling intervals, using dt = 2.5 ms (tics) instead
> In tapas_physio_log (line 58)
  In tapas_physio_read_physlogfiles_siemens_tics (line 85)
  In tapas_physio_read_physlogfiles (line 92)
  In tapas_physio_main_create_regressors (line 121)
  In siemens_vd_ppu3t_matlab_script (line 66)
  In run (line 91)
Warning: Invalid trend type '3'.. assuming 'linear'.
> In detrend (line 52)
  In tapas_physio_filter_respiratory (line 38)
  In tapas_physio_main_create_regressors (line 209)
  In siemens_vd_ppu3t_matlab_script (line 66)
  In run (line 91)
    maxscan (incl. dummies) = 30
    tmin (1st scan start (1st dummy))=  17.25 s
    tmin (1st scan start (after dummies))=  17.25 s
    tmax =  46.99 s
    mean TR =   0.99 s

Note: Guessed 1 additional cardiac pulse(s) at time series end for phase
estimation

*>> run('/../../siemens_vd_ppu3t_matlab_script_adapted.m')*
Warning: Ignoring given sampling intervals, using dt = 2.5 ms (tics) instead
> In tapas_physio_log (line 58)
  In tapas_physio_read_physlogfiles_siemens_tics (line 85)
  In tapas_physio_read_physlogfiles (line 92)
  In tapas_physio_main_create_regressors (line 121)
  In siemens_vd_ppu3t_matlab_script_adapted (line 66)
  In run (line 91)
Warning: Invalid trend type '3'.. assuming 'linear'.
> In detrend (line 52)
  In tapas_physio_filter_respiratory (line 38)
  In tapas_physio_main_create_regressors (line 209)
  In siemens_vd_ppu3t_matlab_script_adapted (line 66)
  In run (line 91)
    maxscan (incl. dummies) = 189
    tmin (1st scan start (1st dummy))=  15.48 s
    tmin (1st scan start (after dummies))=  15.48 s
    tmax = 279.99 s
    mean TR =   1.40 s

Note: Guessed 1 additional cardiac pulse(s) at time series end for phase
estimation

adomagalik, 2020-10-21:
update on Linux:

After removing SPM and initializing TAPAS again, analysis on both example and my data worked. However, warnings are still there:

Warning: Ignoring given sampling intervals, using dt = 2.5 ms (tics) instead 
> In tapas_physio_log (line 58)
  In tapas_physio_read_physlogfiles_siemens_tics (line 85)
  In tapas_physio_read_physlogfiles (line 92)
  In tapas_physio_main_create_regressors (line 121)
  In siemens_vd_ppu3t_matlab_script (line 66)
  In run (line 91) 
Warning: Invalid trend type '3'.. assuming 'linear'. 
> In detrend (line 52)
  In tapas_physio_filter_respiratory (line 38)
  In tapas_physio_main_create_regressors (line 209)
  In siemens_vd_ppu3t_matlab_script (line 66)
  In run (line 91)

mrikasper, 2020-10-21:
Dear Aleksandra and Joe,

That's excellent news! 

1. @JoeZiminski The SPM path issue is a tricky (but unfortunately common) one, and it's hard to reproduce on a different system, therefore thank you so much for debugging it! 
    - From my understanding, SPM should *never* be added with subfolders to the path, just the spm main folder itself. SPM will take care of what it needs when you call it. 
    - The shadowing of inbuilt functions by fieldtrip can lead to really obscure error messages (e.g., "function not implemented for this data type"). I will try to catch such errors for PhysIO in the next release by checking whether some utilized functions are the inbuilt versions from Matlab toolboxes or fieldtrip. 
2. @adomagalik Great that removing SPM (from the path?) and initializing TAPAS helped. As I mentioned above, adding the spm path only without the subfolders should be fine and enables you to use the batch editor etc. The other warnings you see are benign:
    - `detrend` fails gracefully (linear instead of cubic detrending) for older Matlab versions
    - `ignoring given sampling intervals` means that for SIEMENS Tics there is a convention that a time sample is always 2.5 milliseconds, and that convention overrides any manual settings for sampling interval
    -  `guessed 1 additional cardiac pulse` is perfectly normal (the scan ended, but to estimate the cardiac phase of the last volume, we need to know when the first cardiac pulse after the scan happened), it would only be suspicious if the number was much higher (e.g., 10)

I hope that helps!

All the best,
Lars

closed 2021-01-26 by mrikasper