translationalneuromodeling/tapas #112: DVARS algorithm to censor motion outliers does not seem to work

juufsoares, 2020-11-16:
Dear Lars, 

I am trying to use DVARS censoring method to correct motion outliers effects on my fMRI data, however the number of motion outliers detected is always 0 which is strange since using the Framewise Displacement (FD) algorithm the number of detected motion outliers for the same participants is not always 0. Below are the physio batches for the same participant using FD and DVARS. When using FD the number of detected motion outliers is 14 but when using DVARS is 0.  I used the recommended threshold (1.4%) and I already tried with different threshold values and with different participants but the result is always the same, 0 motion outliers. Could you provide some help on this? . 

Thank you in advance. 

Kind Regards, 

Júlia Soares
[physio.zip](https://github.com/translationalneuromodeling/tapas/files/5547488/physio.zip)

mrikasper, 2020-11-30:
Dear Júlia,

my apologies for the late reply! 

While DVARS is one of the feature requests for PhysIO, it's still pending at the moment and hasn't been implemented.

I didn't catch this parameter selection in the GUI, I am afraid, that's why you didn't see a proper error message that this feature is missing.

I am sorry that this might have stalled your work - is there any particular reason you would like to use DVARS? FD is probably suitable for most applications, and is much faster, since it does not need access to the imaging data itself. I can only see very few cases (e.g., partial coverage where motion estimation is difficult) that would require it.

All the best,
Lars

juufsoares, 2020-12-05:
Dear Lars, 

Thank you for the quick answer. 
I am performing a study in which I compare different strategies to deal with motion outliers in fMRI data, namely censoring using FD and DVARS algorithms and volume interpolation, hence my interest in using DVARS. 

Kind Regards, 

Júlia Soares

mrikasper, 2021-01-26:
Dear Júlia,

Thank your for the clarification!
I will close this issue for now, but increase the priority of the pending feature request for DVARS.

All the best,
Lars

closed 2021-01-26 by mrikasper