translationalneuromodeling/tapas #113: HGF: weird omega3 model fit result & choosing between unitsq_softmax and softmax

haoxue-fan, 2020-11-16:
Dear Prof. Mathys,

I hope you are doing well, and hope everything is great in the new institution! My name is Haoxue and I am a psychology grad student. I am really grateful towards the tapas package which enables me to implement the HGF for a study I have run earlier this year. I have two questions regarding the model fitting, and I would appreciate it if you could give me some guidance on them:

Some information about the task: I am using a task very similar to Behrens et al. (2007, Nature Neuroscience), with a stable period(outcome prob: 0.2/0.8) and a volatile period(outcome prob: 0.1/0.9, switch every 20 trials). The outcome is binary. I am using tapas_hgf, tapas_unitsq_sgm and tapas_softmax functions.

<u>Quetsion1:</u> Sometimes the fitted results have a very large omega3 and the model didn't throw any error, such as the two examples below (different participants)
[image]
[image] 

I wonder whether this is due to a relatively high correlation between the parameters in the posterior. The output from model.optim.corr is:
(first pic)corr = 
    1.0000    0.9236   -0.8961
    0.9236    1.0000   -0.8683
   -0.8961   -0.8683    1.0000
(second pic)corr =   
   1.0000   -0.6786   -0.5425
   -0.6786    1.0000    0.1438
   -0.5425    0.1438    1.0000

Is there anything I can do to mitigate this problem? Or should I simply throw this participant's behavior away, and focus on the rest 'normal' parameter results?

<u>Quetsion2:</u> I would appreciate it if you can speak a little more about the difference between unitsq_softmax and softmax functions. In my understanding, they are different ways of mapping a number (-inf, inf) to [0,1], and unitsq_softmax has the advantage of not only being able to produce a curve that can be higher than y=x in [0, 0.5]. But because I haven't seen unitsq_softmax so often outside of tapas, I am wondering which one I should choose when it comes to modeling.

Thank you so much and I am looking forward to hearing back from you.

Best,
Haoxue

chmathys, 2020-11-17:
Dear Haoxue,

My responses are below.

Best wishes,
Christoph

> Dear Prof. Mathys,
> I hope you are doing well, and hope everything is great in the new institution! My name is Haoxue and I am a psychology grad student. I am really grateful towards the tapas [...]

tapas_hgf_binary, I assume?
>  tapas_unitsq_sgm and tapas_softmax functions.
> Quetsion1: Sometimes the fitted results have a very large omega3 and the model didn't throw any error, such as the two examples below (different participants)
> I wonder whether this is due to a relatively high correlation between the parameters in the posterior. The output from model.optim.corr is:
> (first pic)corr =
> 1.0000 0.9236 -0.8961
> 0.9236 1.0000 -0.8683
> -0.8961 -0.8683 1.0000
> (second pic)corr =
> 1.0000 -0.6786 -0.5425
> -0.6786 1.0000 0.1438
> -0.5425 0.1438 1.0000
> Is there anything I can do to mitigate this problem? Or should I simply throw this participant's behavior away, and focus on the rest 'normal' parameter results?

There is a number of things you could try here:

• Place tighter priors on one or several of the parameters you are estimation (i.e., reduce the prior variance). In particular, keeping omega three closer to its prior mean could avoid the jumps in the trajectories
• In addition, or alone, it might help to reduce the prior mean of omega3
• You could use tapas_ehgf_binary instead (see the interactive demo). This reverses the order of mean and precision updates, which leads to much greater robustness to exactly the kind of jumps you’re seeing
• If there are only very few participants whose trajectories look like this, you could consider excluding them. However, their behaviour in the task looks quite reasonable, so I would try to avoid excluding them.

> Quetsion2: I would appreciate it if you can speak a little more about the difference between unitsq_softmax and softmax functions. In my understanding, they are different ways of mapping a number (-inf, inf) to [0,1], and unitsq_softmax has the advantage of not only being able to produce a curve that can be higher than y=x in [0, 0.5]. But because I haven't seen unitsq_softmax so often outside of tapas, I am wondering which one I should choose when it comes to modeling.

You’re right that the unit-square sigmoid isn’t all that common, but it’s not unheard of either. It has the advantage you mention, but it also has another property that’s important to keep in mind: in addition to the fixed point at (0.5, 0.5), it also has (0, 0) and (1, 1) as fixed points. In other words, it enforces a zero probability of choosing an option that is believed to be impossible, and a certainty of choosing option that is believed to be certain. While this is logically consistent, it may not always be what you want: the logistic sigmoid is somewhat more flexible in accommodating decision noise because it has only one fixed point.

haoxue-fan, 2020-11-29:
Dear Prof. Mathys,

Thank you so much for your detailed reply on my data and explanation for the unit-square sigmoid, really helpful! I deeply appreciate it. I am still in the process of trying different hyperparameters and see how the results look like, and I will close the issue now. Thank you again and hope everything goes on well!

closed 2020-11-29 by haoxue-fan