translationalneuromodeling/tapas #114: rDCM for resting-state data--No predicted BOLD signal?

peterjbedford, 2020-11-26:
Hello there,

Thank you for sharing the rDCM tools!
We wish to apply the rDCM tools to find connectivity parameter estimates for participants of a resting-state fMRI experiment. By following the special instructions for resting-state data (as found in the rDCM Manual) we are able to successfully run a script which produces the parameter estimates. However, we get a predicted BOLD signal output which is entirely made up of zeros. My question is:

_Are we meant to ignore the predicted BOLD signal in the case of rDCM for rs-fMRI?_

I can see that, though the instructions indicate we should leave empty the input structure **U**, in this case **tapas_rdcm_estimate** generates a placeholder array **U.u** containing only zeros. Through a chain of function calls, starting with **tapas_rdcm_compute_statistics**, and ending with the following excerpt from **tapas_dcm_euler_gen**

```
[x,~,~,v,q] = dcm_euler_integration(A,C,U,mArrayB,mArrayD,...
                       oxygenExtractionFraction,alphainv,tau,gamma,kappa,paramList);
```

where **x** is the synthetic neuronal signal, it is clear the final predicted BOLD signal depends on **U**. I didn't see any special treatment for the case of a null input array, so that's why I'm wondering if we're just meant to ignore the predicted BOLD signal.

Thanks for your help,
Peter

StefanFraessle, 2020-11-27:
Hi Peter,

Thank you very much for your interest in rDCM -- we hope you find the tool useful.

Yes, this is indeed a bit confusing and probably something I should optimize at some point in the future. In general, rDCM is a model that fits the data in the frequency domain, not the time domain. This is not entirely dissimilar to other variants of DCM that are suitable for rs-fMRI data, like spectral DCM. Because of that, the "right" way to look at the fit of the model (i.e., observed vs. predicted signal), is to look at data in the frequency domain. These are stored in `output.signal.yd_source_fft` (observed signal) and `output.signal.yd_pred_rdcm_fft` (predicted signal). For instance, you could compute the power spectral densities for these data (as is done in `tapas_rdcm_visualize.m`). This is what I would suggest in order to assess the fit of the model both for resting-state and task-based fMRI data.

Now, for task-based fMRI data (where one has driving inputs **U**), there is the possibility to also generate time series. However, because (as mentioned above) the model does not by itself have a representation of the time domain, a detour is necessary. Specifically, the toolbox takes the rDCM estimates and essentially plugs them into a classical (linear) DCM to generate time series. This works well most of the times, however, its important to note that this is not really an accurate representation because rDCM and classical DCM are essentially different generative models (with different underlying assumptions).

To make a long story short: It's best to look at observed and predicted data in the frequency domain (as highlighted above).

I hope this is helpful - let me know if you have further questions :)

All the best,
Stefan

peterjbedford, 2020-11-28:
Hi Stefan,

Thank you for your speedy and considered response. This was enormously helpful! Thank you also for offering your help with further questions.

Best,
Peter

StefanFraessle, 2020-11-28:
HI Peter,

Excellent - glad you found this helpful. Good luck with your study,

All the best,
Stefan

closed 2020-11-28 by TNU-yaoy