translationalneuromodeling/tapas #117: tapas_physio_create_noise_rois_regressors.m Error, line 186

Ashrmis, 2021-01-15:
I was trying to run tapas_physio_create_noise_rois_regressors.m as tapas_physio_create_noise_rois_regressors(physio.model.noise_rois,physio.verbose) where I had set the fmri file name and the anatomical CSF mask in the physio.model sub-structure. It seems to run most of it, I even get an output SPM graphics window with some plots, but then I get the following error (I have copied and pasted my matlab command window)

```
>>tapas_physio_create_noise_rois_regressors(physio.model.noise_rois,physio.verbose)
SPM12: spm_check_registration (v7759)              15:51:45 - 15/01/2021
========================================================================
Display sub-hc01_label-CSF_probseg.nii,1
------------------------------------------------------------------------
15-Jan-2021 15:51:50 - Running job #2
------------------------------------------------------------------------
15-Jan-2021 15:51:50 - Running 'Coregister: Estimate & Reslice'
SPM12: spm_coreg (v7320)                           15:51:50 - 15/01/2021
========================================================================
Printing 'Graphics' figure to:
  /usr/local/MATLAB/R2019a/toolbox/tapas-master/PhysIO/spm_2021Jan15.ps
Completed                               :          15:52:41 - 15/01/2021
SPM12: spm_reslice (v7141)                15:52:41 - 15/01/2021
========================================================================
Completed                               :          15:52:44 - 15/01/2021
15-Jan-2021 15:52:44 - Done    'Coregister: Estimate & Reslice'
15-Jan-2021 15:52:44 - Done

Reference to non-existent field 'cm'.
Error in tapas_physio_create_noise_rois_regressors (line 186)
        c_handle    = findobj(findobj(st.vols{r}.ax{1}.cm,'label','Overlay'),'Label','Remove coloured blobs');
```
Any help would be appreciated, pretty new to this toolbox. I thought it be due to by input arguments for this function, but it does successfully go through majority of the function, just not this part. 

Thanks,

Singerjohannes, 2021-01-21:
I had the same problem. Try to set verbose to 1 than the part where the problem comes from will be skipped. That solved it for me

Ashrmis, 2021-01-22:
> I had the same problem. Try to set verbose to 1 than the part where the problem comes from will be skipped. That solved it for me

 

what do you mean by set verbose to 1? so like replace physio.verbose in the function call? if so that just gives me an error. Because when I look into the physio.verbose structure I get:

physio.verbose

 

              level: 2

        process_log: {3×1 cell}

        fig_handles: [2 3 4 5 6 7]

    fig_output_file: '/usr/local/MATLAB/R2019a/toolbox/tapas-master/PhysIO/physio_log'

           use_tabs: 0

          show_figs: 1

          save_figs: 0

         close_figs: 0

 So not really sure what you mean by set verbose to 1

mrikasper, 2021-01-22:
Dear Ash,

thank you for trying out the toolbox! This error is related to an SPM visualization we are using and for some reason the interactive plotting using SPM's checkreg does not always work on all systems (What is your system configuration, @Ashrmis?; Did you try opening SPM via `spm fmri` before running the PhysIO code?) 

Thank you, Johannes @Singerjohannes , for providing a work-around solution, it's great to see PhysIO users support each other!

Just to clarify: The parameter that defines the verbosity level (and suppresses the call to the SPM-related visualization), is `level` within the `verbose` structure, i.e., 

`physio.verbose.level = 1;` 

before calling the noise creation should circumvent the problem.

All the best,
Lars

Ashrmis, 2021-01-22:
Hi Lars,
I am on centos 6. Yes `spm fmri` opens up fine. I tried the `physio.verbose.level=1` and that seems to do the trick. 

Thanks!

closed 2021-01-22 by Ashrmis